corundum/corundum #3: ExaBlaze10G requires .fw to be downloaded onto FPGA

Winters123, 2020-05-19:
Hi,

I used corundum to generate the bitstream (fpga.bit) for exablaze10G. But when I was going to download it onto the board, I found the instruction of Exa10G said that:
>The `exanic-fwupdate` utility can be used to update the firmware image stored on the ExaNIC. This >utility can use the compressed .fw.gz file format that firmware images are released as, or >uncompressed .fw files.
So it looks like the exaNIC needs something like xxx.fw instead of typical xxx.bit to program the FPGA. How should I proceed with this?

Thanks,

alexforencich, 2020-05-19:
So, the firmware update side of things needs work at the moment.  Even if you had an image in the proper format for the exablaze firmware updater to load it onto the card, you wouldn't be able to change it afterwards.  There is support for flash access in many of the corundum designs, but the userspace tool mqnic-fw is incomplete and there is currently no support in the FPGA design to reset the FPGA to load the new image.  These things are on the to-do list.  Currently, the recommended way to load corundum onto the card is via JTAG (and you'll need to have JTAG access anyway if you want to use tools like the ILA).  

For the ExaNIC X10, they have an edge connector for JTAG (J3).  Mating part is Sullins RBB10DHHN, and the pinout is as follows:

          [FPGA side]           [Back side]
    ____________________        _________________________
        TDI --######### \      / #########== GND   TP11 o
        TDO --######### |      | #########== GND  2 o
              ######### |      | #########== GND  1
        TMS --######### |      | #########== GND  P   o
        TCK --######### |      | #########== GND  T   TP9
              ######### |      | #########== GND    o
     J3       ######### |      | #########== GND    TP10
              ######### |      | #########== 12V
              ######### |      | #########== 12V
        1V8 ==######### /      \ #########== 12V
          (_____________       ______________)
                        \     / Patents: www.exablaze.com/...
     L7    [inductor]   |     |

**Warning: the pinout is poorly designed and and the connector is not keyed!  Connecting the connector backwards will connect 12V to the JTAG TDI and TDO pins.  So triple check before plugging in and/or powering on.**

Winters123, 2020-05-19:
So.. looks like I will need to plug a JTAG after the lockdown is removed.
Thanks again!

closed 2020-05-19 by Winters123

alexforencich, 2020-05-21:
Actually, is firmware updates a feature you might be interested in working on? I can look in to setting up some sort of remote access to an X10 with a JTAG cable connected and get my partial mqnic-fw code into the repo, as well as the icap code for triggering an FPGA reload. A lot of it is already written including flash access for both BPI and QSPI flash devices, it just needs some polish, and the ability to parse bit files and hex files. And I have not looked at the ExaNIC fw format at all, that might also be a good thing to implement (both generating and parsing) to ease switching between ExaNIC and corundum firmware.

Winters123, 2020-05-25:
Yes. I think this is a good idea. But if I understand it correctly, whether we have this `mqnic-fw` or not we still a JTAG to be plugged.
I think your second idea that providing a tool for switching between ExaNIC and corundum firmware is more attractive since ExaNIC users can simply follow the instructions from ExaNIC to implement corundum on the NIC.