corundum/corundum #5: Implementation errors for U50 - Vivado 2019.2

ELHorta, 2020-07-20:
Hi,

I've just tried to implement the design for U50 and I got some errors.
I'm attaching the runme.log.

Could you please check what is missing?

Thanks

Edson
[runme.log](https://github.com/ucsdsysnet/corundum/files/4948852/runme.log)

alexforencich, 2020-07-20:
What version of Vivado are you using? It looks like it's complaining about a change in the CMAC IP core pins that happened recently. You might need to use 2020.1 to build the design. Alternatively, you can edit those pins to match the older version of the core.

ELHorta, 2020-07-20:
I am using 2019.2. Do you know what files I need to edit in order to solve the IP core pins compatibility?

alexforencich, 2020-07-20:
Apparently in 2020.1, the IP core interface was changed from using a bunch of 1 bit ports to ports that accept arrays.  So, if you want to build on 2019.2 or earlier, what you need to do is edit fpga.v and replace

    .gt_rxp_in({qsfp_rx4_p, qsfp_rx3_p, qsfp_rx2_p, qsfp_rx1_p}), // input
    .gt_rxn_in({qsfp_rx4_n, qsfp_rx3_n, qsfp_rx2_n, qsfp_rx1_n}), // input
    .gt_txp_out({qsfp_tx4_p, qsfp_tx3_p, qsfp_tx2_p, qsfp_tx1_p}), // output
    .gt_txn_out({qsfp_tx4_n, qsfp_tx3_n, qsfp_tx2_n, qsfp_tx1_n}), // output

with

    .gt0_rxp_in(qsfp_rx1_p), // input
    .gt0_rxn_in(qsfp_rx1_n), // input
    .gt1_rxp_in(qsfp_rx2_p), // input
    .gt1_rxn_in(qsfp_rx2_n), // input
    .gt2_rxp_in(qsfp_rx3_p), // input
    .gt2_rxn_in(qsfp_rx3_n), // input
    .gt3_rxp_in(qsfp_rx4_p), // input
    .gt3_rxn_in(qsfp_rx4_n), // input
    .gt0_txn_out(qsfp_tx1_n), // output
    .gt0_txp_out(qsfp_tx1_p), // output
    .gt1_txn_out(qsfp_tx2_n), // output
    .gt1_txp_out(qsfp_tx2_p), // output
    .gt2_txn_out(qsfp_tx3_n), // output
    .gt2_txp_out(qsfp_tx3_p), // output
    .gt3_txn_out(qsfp_tx4_n), // output
    .gt3_txp_out(qsfp_tx4_p), // output

or whatever the equivalent is for the design in question.  That should be the only change you need to make.

ELHorta, 2020-07-20:
I was able to implement it. Thanks.

closed 2020-07-20 by ELHorta

alexforencich, 2020-07-20:
Good to hear!

hl4080, 2020-12-02:
hello, alexforencich, I am using vivado 2019.2 and implement the design in AU280
I followed your suggestion to replace port instantiation in fpga.v and it worked well in synthesis. But during implementation,  errors emerged in DRC task.
Can you help me figure out what's going on?
Thanks!

Here's the error logs
```
Starting DRC Task
INFO: [DRC 23-27] Running DRC with 8 threads
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtytxn_out[0] has multiple drivers: qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[0].GTYE4_CHANNEL_PRIM_INST/GTYTXN, and qsfp0_cmac_inst/inst/cmac_usplus_0_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_0_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[10].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[0].GTYE4_CHANNEL_PRIM_INST/GTYTXN.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtytxn_out[1] has multiple drivers: qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[1].GTYE4_CHANNEL_PRIM_INST/GTYTXN, and qsfp0_cmac_inst/inst/cmac_usplus_0_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_0_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[10].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[1].GTYE4_CHANNEL_PRIM_INST/GTYTXN.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtytxn_out[2] has multiple drivers: qsfp0_cmac_inst/inst/cmac_usplus_0_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_0_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[10].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[2].GTYE4_CHANNEL_PRIM_INST/GTYTXN, and qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[2].GTYE4_CHANNEL_PRIM_INST/GTYTXN.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtytxn_out[3] has multiple drivers: qsfp0_cmac_inst/inst/cmac_usplus_0_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_0_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[10].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[3].GTYE4_CHANNEL_PRIM_INST/GTYTXN, and qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[3].GTYE4_CHANNEL_PRIM_INST/GTYTXN.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtytxp_out[0] has multiple drivers: qsfp0_cmac_inst/inst/cmac_usplus_0_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_0_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[10].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[0].GTYE4_CHANNEL_PRIM_INST/GTYTXP, and qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[0].GTYE4_CHANNEL_PRIM_INST/GTYTXP.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtytxp_out[1] has multiple drivers: qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[1].GTYE4_CHANNEL_PRIM_INST/GTYTXP, and qsfp0_cmac_inst/inst/cmac_usplus_0_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_0_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[10].gen_enabled_channel.gtye4_channel_wrapper_inst/ch [... cut ...] d qsfp1_cmac_inst/inst/cmac_usplus_1_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_1_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[11].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[3].GTYE4_CHANNEL_PRIM_INST/GTYTXP.
INFO: [Project 1-461] DRC finished with 8 Errors
INFO: [Project 1-462] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.
```

alexforencich, 2020-12-02:
What do the gt pins in the CMAC instance look like?

hl4080, 2020-12-02:
changed as you suggested
```
      .gt0_rxp_in(qsfp0_rx1_p),
      .gt0_rxn_in(qsfp0_rx1_n),
      .gt1_rxp_in(qsfp0_rx2_p),
      .gt1_rxn_in(qsfp0_rx2_n),
      .gt2_rxp_in(qsfp0_rx3_p),
      .gt2_rxn_in(qsfp0_rx3_n),
      .gt3_rxp_in(qsfp0_rx4_p),
      .gt3_rxn_in(qsfp0_rx4_n),
      .gt0_txn_out(qsfp0_tx1_n),
      .gt0_txp_out(qsfp0_tx1_p),
      .gt1_txn_out(qsfp0_tx2_n),
      .gt1_txp_out(qsfp0_tx2_p),
      .gt2_txn_out(qsfp0_tx3_n),
      .gt2_txp_out(qsfp0_tx3_p),
      .gt3_txn_out(qsfp0_tx4_n),
      .gt3_txp_out(qsfp0_tx4_p),
```

alexforencich, 2020-12-02:
And the other CMAC instance?

hl4080, 2020-12-02:
other ports have no modified in both two CAMCs
```
cmac_usplus_0
qsfp0_cmac_inst (
    //.gt_rxp_in({qsfp0_rx4_p, qsfp0_rx3_p, qsfp0_rx2_p, qsfp0_rx1_p}), // input
    //.gt_rxn_in({qsfp0_rx4_n, qsfp0_rx3_n, qsfp0_rx2_n, qsfp0_rx1_n}), // input
    //.gt_txp_out({qsfp0_tx4_p, qsfp0_tx3_p, qsfp0_tx2_p, qsfp0_tx1_p}), // output
    //.gt_txn_out({qsfp0_tx4_n, qsfp0_tx3_n, qsfp0_tx2_n, qsfp0_tx1_n}), // output
      .gt0_rxp_in(qsfp0_rx1_p),
      .gt0_rxn_in(qsfp0_rx1_n),
      .gt1_rxp_in(qsfp0_rx2_p),
      .gt1_rxn_in(qsfp0_rx2_n),
      .gt2_rxp_in(qsfp0_rx3_p),
      .gt2_rxn_in(qsfp0_rx3_n),
      .gt3_rxp_in(qsfp0_rx4_p),
      .gt3_rxn_in(qsfp0_rx4_n),
      .gt0_txn_out(qsfp0_tx1_n),
      .gt0_txp_out(qsfp0_tx1_p),
      .gt1_txn_out(qsfp0_tx2_n),
      .gt1_txp_out(qsfp0_tx2_p),
      .gt2_txn_out(qsfp0_tx3_n),
      .gt2_txp_out(qsfp0_tx3_p),
      .gt3_txn_out(qsfp0_tx4_n),
      .gt3_txp_out(qsfp0_tx4_p),
    .gt_txusrclk2(qsfp0_txuserclk2), // output
    .gt_loopback_in(12'd0), // input [11:0]
    .gt_rxrecclkout(), // output [3:0]
    .gt_powergoodout(), // output [3:0]
    .gt_ref_clk_out(clk_161mhz_ref_int), // output
    .gtwiz_reset_tx_datapath(1'b0), // input
    .gtwiz_reset_rx_datapath(1'b0), // input
    .sys_reset(rst_125mhz_int), // input
    .gt_ref_clk_p(qsfp0_mgt_refclk_1_p), // input
    .gt_ref_clk_n(qsfp0_mgt_refclk_1_n), // input
    .init_clk(clk_125mhz_int), // input

    .rx_axis_tvalid(qsfp0_rx_axis_tvalid_int), // output
    .rx_axis_tdata(qsfp0_rx_axis_tdata_int), // output [511:0]
    .rx_axis_tlast(qsfp0_rx_axis_tlast_int), // output
    .rx_axis_tkeep(qsfp0_rx_axis_tkeep_int), // output [63:0]
    .rx_axis_tuser(qsfp0_rx_axis_tuser_int), // output

    .rx_otn_bip8_0(), // output [7:0]
    .rx_otn_bip8_1(), // output [7:0]
    .rx_otn_bip8_2(), // output [7:0]
    .rx_otn_bip8_3(), // output [7:0]
    .rx_otn_bip8_4(), // output [7:0]
    .rx_otn_data_0(), // output [65:0]
    .rx_otn_data_1(), // output [65:0]
    .rx_otn_data_2(), // output [65:0]
    .rx_otn_data_3(), // output [65:0]
    .rx_otn_data_4(), // output [65:0]
    .rx_otn_ena(), // output
    .rx_otn_lane0(), // output
    .rx_otn_vlmarker(), // output
    .rx_preambleout(), // output [55:0]
    .usr_rx_reset(qsfp0_rx_rst_int), // output
    .gt_rxusrclk2(), // output

    .stat_rx_aligned(), // output
    .stat_rx_aligned_err(), // output
    .stat_rx_bad_code(), // output [2:0]
    .stat_rx_bad_fcs(), // output [2:0]
    .stat_rx_bad_preamble(), // output
    .stat_rx_bad_sfd(), // output
    .stat_rx_bip_err_0(), // output
    .stat_rx_bip_err_1(), // output
    .stat_rx_bip_err_10(), // output
    .stat_rx_bip_err_11(), // output
    .stat_rx_bip_err_12(), // output
[... 183 lines cut ...]
    .ctl_tx_send_lfi(1'b0), // input
    .core_tx_reset(1'b0), // input

    .tx_axis_tready(qsfp0_mac_tx_axis_tready), // output
    .tx_axis_tvalid(qsfp0_mac_tx_axis_tvalid), // input
    .tx_axis_tdata(qsfp0_mac_tx_axis_tdata), // input [511:0]
    .tx_axis_tlast(qsfp0_mac_tx_axis_tlast), // input
    .tx_axis_tkeep(qsfp0_mac_tx_axis_tkeep), // input [63:0]
    .tx_axis_tuser(qsfp0_mac_tx_axis_tuser), // input

    .tx_ovfout(), // output
    .tx_unfout(), // output
    .tx_preamblein(56'd0), // input [55:0]
    .usr_tx_reset(qsfp0_tx_rst_int), // output

    .core_drp_reset(1'b0), // input
    .drp_clk(1'b0), // input
    .drp_addr(10'd0), // input [9:0]
    .drp_di(16'd0), // input [15:0]
    .drp_en(1'b0), // input
    .drp_do(), // output [15:0]
    .drp_rdy(), // output
    .drp_we(1'b0) // input
);
```

alexforencich, 2020-12-02:
You're showing me `qsfp0_cmac_inst`, but the errors are on `qsfp1_cmac_inst`.....

alexforencich, 2020-12-02:
It looks like maybe the pins on `qsfp1_cmac_inst` are connected to the `qsfp0_...` signals, instead of the `qsfp1_...` signals

hl4080, 2020-12-02:
Oh, I got it, I just simply copy the ports intantiated in qsfp0_cmac_inst without modifying its signals!
What a careless mistake!
Thanks a lot, alexforencich~

alexforencich, 2020-12-02:
Heh, no worries, I have made that same mistake many times before, and now it's high on my list of things to suspect when I see things like multi-driven nets.

articice, 2021-12-27:
> if you want to build on 2019.2 or earlier

Note, it can't be built on 2018.2-2018.3, because those include version 2.5 of [cmac_usplus](https://www.xilinx.com/support/documentation/ip_documentation/cmac_usplus/v2_5/pg203-cmac-usplus.pdf) which doesn't export an AXIS interface.

There will be warnings at product output generation:
```
WARNING: [IP_Flow 19-3374] An attempt to modify the value of disabled parameter 'USER_INTERFACE' from 'LBUS' to 'AXIS' has been ignored for IP 'cmac_usplus_0'
```
```
WARNING: [IP_Flow 19-3374] An attempt to modify the value of disabled parameter 'USER_INTERFACE' from 'LBUS' to 'AXIS' has been ignored for IP 'cmac_usplus_1'
```

And then errors at synthesis:
```
ERROR: [Synth 8-448] named port connection 'rx_axis_tvalid' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1051]
ERROR: [Synth 8-448] named port connection 'rx_axis_tdata' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1052]
ERROR: [Synth 8-448] named port connection 'rx_axis_tlast' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1053]
ERROR: [Synth 8-448] named port connection 'rx_axis_tkeep' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1054]
ERROR: [Synth 8-448] named port connection 'rx_axis_tuser' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1055]
ERROR: [Synth 8-448] named port connection 'tx_axis_tready' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1307]
ERROR: [Synth 8-448] named port connection 'tx_axis_tvalid' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1308]
ERROR: [Synth 8-448] named port connection 'tx_axis_tdata' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1309]
ERROR: [Synth 8-448] named port connection 'tx_axis_tlast' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1310]
ERROR: [Synth 8-448] named port connection 'tx_axis_tkeep' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1311]
ERROR: [Synth 8-448] named port connection 'tx_axis_tuser' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0' [/home/user/corundum/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v:1312]
```

alexforencich, 2021-12-27:
Ah yeah, that's a good point, if you want to go back even earlier beyond when the CMACs natively support AXI stream (well, more like include soft logic for conversion), you'll have to edit the TCL scripts and steal the LBUS to AXIS code from a later version of Vivado.  Or you can just use the latest version of Vivado and save yourself the headache.