corundum/corundum #9: Support DPDK

alexforencich, 2020-07-24:
Supporting DPDK would permit the use of userspace network stacks, enabling development of features across the entire stack.  Implementation of a DPDK PMD should be investigated.

YangZhou1997, 2020-10-11:
Hi Alex, 

I just tried your corundum on Xilinx Alevo U250, which works great! Thanks for open-sourcing it! 

Just wonder how the DPDK driver development goes now? Really want to use fast user-space networking. 

Best,
Yang

alexforencich, 2020-10-12:
There is nobody working on it right now.  Currently I am the only one working on corundum, and my current priorities are to implement variable-length descriptors and metadata and then to support Stratix 10 DX.  Take a look at the roadmap here: https://github.com/corundum/corundum/wiki/Corundum-Roadmap

I want to at least get variable-length descriptors finished before taking a crack at DPDK as this will change the descriptor format and the PMD would have to be reworked for the new descriptor format.  

Now, if there are folks who are using corundum that would like to help write a DPDK PMD sooner, then I would be happy to facilitate that.

Basseuph, 2022-01-21:
Hi Yang,

are you still interested in DPDK support? We are working on it and I am wondering whether we should have a quick discussion about this topic.

Ulrich

kimns516, 2022-08-01:
I am also interested in using DPDK. What's the status? Is [1] stable to use at this point?

alexforencich, 2022-08-01:
No, that one is based on a very old version of Corundum, and it is missing some vital sanity checks to prevent it from attaching to more recent versions.  However, it seems like MLE has picked it up again and has made quite a few updates, so hopefully they will have an updated version available soon which will work with a more recent version of Corundum (and I think a more recent version of DPDK as well).

AdamI75, 2023-08-07:
@alexforencich is there any more status on the MLE version of DPDK for Corundum, thanks?

alexforencich, 2023-08-11:
Take a look at the branches on the linked repo.  Looks like maybe https://github.com/missinglinkelectronics/dpdk-stable/tree/22.11-add-corundum-mqnic-pmd is the latest.

AdamI75, 2023-08-14:
@alexforencich thanks very much. I will investigate this shortly.

AdamI75, 2023-08-16:
@alexforencich I am able to compile and build the drivers using meson and ninja. I have configured the Alveo U50 and U280 with Corundum and when I load the kernel I can see the Corundum NIC interfaces. However, when I run the DPDK command: "./dpdk-devbind.py --status" from the MLE DPDK repo "usertools" directory, I do not see the Corundum NICs whether I load the kernel or not. How should I enable the PMD? It says it was tested with a slightly older version of Corundum and only for 10GbE. Therefore, does this mean that it won't work for the Alveo U50 and U280 as they are 100GbE? I am not sure. I am still reading up and will continue to try on my end. I would like to create a test where I can produce data from one Alveo configured with Corundum on my one server and then send that to another Alveo configured with Corundum on another server using iperf3 and then load the mqnic kernel driver and compare with the DPDK MLE mqnic driver. I am concerned that the script is not seeing the Corundum configured Alveos. It may need further porting. What do you think?

AdamI75, 2023-08-17:
@alexforencich Okay, the Alveo is seen as a memory controller and not a network device, so the class needs to be a 5 and not a 2. I now see it and so will play around with the binding and then test what I see. I am not sure why this file was not updated when this was released. I have let MLE know as well.

alexforencich, 2023-08-17:
That's odd.  The class code should be set from the config.tcl script for all of the Alveo boards.  Can you show me an `lspci -vvv -nn` output for the board?

alexforencich, 2023-08-17:
Or are you saying that the DPDK PMD is looking for a memory controller class code?

AdamI75, 2023-08-17:
@alexforencich I have attached two files for you. The one you wanted and a screen shot of the DPDK drivers. I am able to bind the DPDK driver to the Alveo U50 using vfio-pci driver. However, as soon as I do this then your mqnic kernel driver is removed and when I do ifconfig then I don't see the Corundum NIC anymore. However, it is listed as a DPDK compatible driver, so maybe I am misunderstanding something?

Yes, I am talking about the DPDK PMD. Should that be run with mqnic driver or just the vfio-pci driver. How do I test this vfio-pci driver if I can't see the NIC using ifconfig. If I rebind to your mqnic driver then it is fine. I can see the NIC, but then I lose the DPDK driver. 

[image]
[image]

alexforencich, 2023-08-18:
The drivers are mutually exclusive, you can use either the kernel driver or the DPDK PMD, but not both.  Seems like at least that part is behaving as expected, but I'm not sure why the DPDK applications are not working.  It's possible something in the DPDK PMD needs to be adjusted due to a change in the HW, but I'm not familiar with how MLE has implemented the DPDK PMD.  Presumably if it has trouble talking to the hardware, it would print out some sort of a warning message.

AdamI75, 2023-08-18:
@alexforencich Yes, at least I can bind the PMD driver to the Corundum NIC. It seems the "Failed to create interface.." print out is from the drivers/net/mqnic/rte_eth_mqnic.c file, so I think it might need a tweak due to different infrastructure or something like that. I have asked MLE for support as well. I am still waiting to hear back. I will put this on the back burner for now. I am going to do some test with your kernel driver.

AdamI75, 2023-08-18:
@alexforencich MLE got back to me with some useful information. I am recompiling Corundum using their tested repo version just to see if there is a difference. Refer to our correspondence here: https://github.com/missinglinkelectronics/dpdk-stable/issues/4

alexforencich, 2023-08-19:
I thought they had updated their PMD code for more recent versions of Corundum.  Looks like they didn't, so their code only works with very old versions of Corundum, and also doesn't include some checks to prevent it from trying to talk to newer versions of Corundum.

AdamI75, 2023-08-19:
@alexforencich thanks for the input. This is all part of integration. I don't really need the DPDK drivers for now, but when it comes to testing the DPDK improvements from your original driver then I will need it. I am busy documenting the ease of use of Corundum over other frameworks. I don't mind contributing towards the Corundum framework at a later stage, but it really would be nice to see the difference if I load your mqnic driver with iperf3 vs the DPDK driver. I am not sure that iperf3 will work with the DPDK driver. I may need to use the pktgen utility. Anyway, I have only spent a couple of days on DPDK, so I am a newbie with lots to learn. I am going to do a cold boot and then just load the MLE Corundum build and try again. I don't think I will have like since I can't even build your driver on that version. I will keep you posted. Thanks for the great work that you are doing :).

vaniaprkl, 2023-08-25:
Hi @alexforencich and @AdamI75 
So excited to suddenly see movement on this front and something to try out. I had switched to testing with kernel drivers having not been successful with the creation of interfaces. I just tried the older corundum fw and had gotten to the point were I tried with both dpdk drivers, ```vfio-pci``` and ```uio_pci_generic```... no dice with getting interfaces up with either driver. Don't know how mle validated these without an interface being seen. Maybe they didn't get that far?

AdamI75, 2023-08-25:
@vaniaprkl MLE have done some good work and so has @alexforencich. The documentation has been a bit lacking for testing the DPDK with Corundum, but check this out - added by MLE yesterday: https://github.com/missinglinkelectronics/dpdk-stable/tree/22.11-add-corundum-mqnic-pmd/drivers/net/mqnic and check this issue I created: https://github.com/missinglinkelectronics/dpdk-stable/issues/4. The issue is that MLE incorrectly referred to the incorrect Corundum githash version in their documentation, which didn't work. I am now using the corrected githash and getting success. I hope this helps!

vaniaprkl, 2023-08-25:
Just as an update, I was able to get testpmd running: 

```
don@machine:/disk2/dpdk_test3/dpdk-stable# sudo ./build/app/dpdk-testpmd 
EAL: Detected CPU lcores: 56
EAL: Detected NUMA nodes: 2
EAL: Detected static linkage of DPDK
EAL: Multi-process socket /var/run/dpdk/rte/mp_socket
EAL: Selected IOVA mode 'VA'
EAL: VFIO support initialized
EAL: Using IOMMU type 1 (Type 1)
EAL: Ignore mapping IO port bar(1)
EAL: Probe PCI driver: net_mqnic (1234:1001) device: 0000:81:00.0 (socket 1)
eth_mqnic_pci_probe():  Vendor: 0x1234
eth_mqnic_pci_probe():  Device: 0x1001
eth_mqnic_pci_probe():  Subsystem vendor: 0x10ee
eth_mqnic_pci_probe():  Subsystem device: 0x9032
eth_mqnic_pci_probe():  Class: 0x020000
mqnic_dev_pci_specific_init(): Control BAR size: 33554432
mqnic_common_probe():  type 0xffffffff (v 0.0.1.0)
mqnic_common_probe():  type 0x0000c007 (v 0.0.1.0)
mqnic_common_probe():  type 0x0000c000 (v 0.0.1.0)
mqnic_common_probe():  type 0x00000000 (v 0.0.0.0)
mqnic_common_probe():  type 0x0000c006 (v 0.0.1.0)
mqnic_common_probe():  type 0x0000c080 (v 0.0.1.0)
mqnic_common_probe():  type 0x0000c120 (v 0.0.2.0)
mqnic_common_probe():  type 0x0000c140 (v 0.0.1.0)
mqnic_common_probe():  type 0x0000c150 (v 0.0.1.0)
mqnic_common_probe(): Find Register Block
mqnic_common_probe(): FPGA ID: 0x04b77093
mqnic_common_probe(): FW ID: 0x00000000
mqnic_common_probe(): FW version: 0.0.1.0
mqnic_common_probe(): Board ID: 0x10ee9032
mqnic_common_probe(): Board version: 1.0.0.0
mqnic_common_probe(): Git hash: 56fe10f2
mqnic_common_probe(): Release info: 00000000
mqnic_common_probe(): IF offset: 0x00000000
mqnic_common_probe(): IF count: 4
mqnic_common_probe(): IF stride: 0x00800000
mqnic_common_probe(): IF CSR offset: 0x00040000
mqnic_create_interface(): Interface-level register blocks:
mqnic_create_interface():  type 0x0000c001 (v 0.0.4.0)
mqnic_create_interface():  type 0x0000c010 (v 0.0.1.0)
mqnic_create_interface():  type 0x0000c020 (v 0.0.1.0)
mqnic_create_interface():  type 0x0000c030 (v 0.0.1.0)
mqnic_create_interface():  type 0x0000c021 (v 0.0.1.0)
mqnic_create_interface():  type 0x0000c031 (v 0.0.1.0)
mqnic_create_interface():  type 0x0000c090 (v 0.0.1.0)
mqnic_create_interface():  type 0x0000c002 (v 0.0.2.0)
mqnic_create_interface():  type 0x0000c004 (v 0.0.3.0)
mqnic_create_interface(): IF features: 0x00000711
mqnic_create_interface(): Port count: 1
mqnic_create_interface(): Scheduler block count: 1
mqnic_create_interface(): Max TX MTU: 9214
mqnic_create_interface(): Max RX MTU: 9214
mqnic_create_interface(): TX queue offset: 0x00100000
mqnic_create_interface(): TX queue count: 8192
mqnic_create_interface(): TX queue stride: 0x00000020
mqnic_create_interface(): TX completion queue offset: 0x00200000
mqnic_create_interface(): TX completion queue count: 8192
mqnic_create_interface(): TX completion queue stride: 0x00000020
mqnic_create_interface(): RX queue offset: 0x00300000
mqnic_create_interface(): RX queue count: 256
mqnic_create_interface(): RX queue stride: 0x00000020
mqnic_create_interface(): RX completion queue offset: 0x00380000
mqnic_create_interface(): RX completion queue count: 256
mqnic_create_interface(): RX completion queue stride: 0x00000020
mqnic_create_interface(): Max desc block size: 8
mqnic_create_port(): Port-level register blocks:
mqnic_create_port():  type 0x0000c003 (v 0.0.2.0)
[... 290 lines cut ...]

Shutting down port 2...
Closing ports...
mqnic_dev_close(): 
mqnic_dev_rx_queue_release(): RX Completion Queue 0 released
mqnic_dev_rx_queue_release(): RX Queue 0 released
mqnic_dev_tx_queue_release(): TX Completion Queue 0 released
mqnic_dev_tx_queue_release(): TX Queue 0 released
Port 2 is closed
Done

Shutting down port 3...
Closing ports...
mqnic_dev_close(): 
mqnic_dev_rx_queue_release(): RX Completion Queue 0 released
mqnic_dev_rx_queue_release(): RX Queue 0 released
mqnic_dev_tx_queue_release(): TX Completion Queue 0 released
mqnic_dev_tx_queue_release(): TX Queue 0 released
Port 3 is closed
Done

Bye...
don@machine:/disk2/dpdk_test3/dpdk-stable# 

```

vaniaprkl, 2023-08-25:
Hi @AdamI75 thanks for the quick feedback. I had stumbled on that update from mle and that's how I got the above output which I couldn't before. Yeah its pretty cool seeing all of it work. 

I was playing around with Corundum for a few weeks now and I really like the framework Alex has built out. It's a lot more complete than OpenNic that I briefly went over. I am not familiar with DPDK, but just starting to read up on it because of Alex mentioning it to be the way to go to get better performance in his paper.

AdamI75, 2023-08-29:
@alexforencich I agree that it should be linked to the Corundum repo to keep it up to date - maybe a submodule with an appropriate githash. It would also be useful if the kernel driver and DPDK PMD didn't need to be mutually exclusive like the Mellanox drivers. If we decide to use the Corundum framework, then I would like to get the MLE DPDK driver to work with the latest Corundum version and then possibly to explore having both the kernel driver and  DPDK PMD driver working together, so it can look like a NIC to the OS using both drivers - not possible at this point.  I will definitely keep you posted. We should decide by the end of the week.

liyangyang-linux, 2023-12-08:
@vaniaprkl Hi, have you ever encountered a situation where there is no data in io mode, but tx is normal in txonly mode