corundum/corundum #13: PTP slave?

alexisfrjp, 2020-08-12:
I was interested in the PTP slave I thought corundum is providing.

The `ptp_clock` module's period/offset/drift adjustments seem to come from the PCIe rather than an Ethernet interface.

Do you confirm Corundum doesn't parse and process the PTP messages via UDP to update the timestamp?

Do we need a different NIC that is connected to a PTP master and then forwarding the PTP adjustment messages via PCIe?

Thanks

alexforencich, 2020-08-12:
That's partially correct; corundum implements the minimum hardware features required to implement PTP (hardware clock + hardware timestamps) as well as the necessary driver support for these features.  It does not implement the actual protocol; this is provided by linux-ptp.  No additional NIC is required if you're using the standard mqnic kernel module; all you should have to do is run ptp4l to sync the hardware clock.

alexisfrjp, 2020-08-12:
Understood, thank you for the confirmation Alex :)

closed 2020-08-12 by alexisfrjp

alexforencich, 2020-08-12:
To clarify: from looking at several different NIC drivers, there are a few variations on how PTP support is implemented.  Any NIC that implements PTP has to implement some sort of clock and has to capture high-resolution hardware timestamps in some form.  

Usually the hardware clock is a simple cycle counter running on some fixed internal frequency, in which case the correction is not applied to the clock directly but it is instead applied to the mapping between the cycle count and wall clock time.  This is done when the NIC does not need to use the timestamps for anything internally.  Corundum does not do this; corundum implements a proper PTP clock in hardware so that PTP time is available on the FPGA.

Some NICs capture hardware timestamps for PTP packets only and expose them to the driver through register space.  In this case, the NIC has to recognize PTP packets in order to determine which packets to store a timestamp for, but it does not implement any more of the PTP protocol.  In this case, timestamps are only available for PTP packets, and the driver must read them out after each timestamp is captured.  Another strategy is to return timestamps in the completion data unconditionally for all packets, as high-resolution timestamps can have many other uses aside from time synchronization.  This is what corundum does, along with mellanox connectx-4 and connectx-5 NICs and others.  

So corundum provides exactly what you get on many other NICs by providing hardware timestamps for all packets, except it also makes the hardware clock available to user logic on the FPGA.

alexisfrjp, 2020-08-12:
Thanks for the clarification.
Indeed there are seceral ways to do it. Nevertheless, the way corundum implements the timestamp adjustments doesn't provide the best accuracy. To make the adjustments, you do a PCIe round-trip and that time isn't deterministic nor constant.
That still accurate though.

alexforencich, 2020-08-12:
I should also mention that I am working on supporting corundum on Zynq, connected over AXI to the ARM cores instead of PCIe. With this setup, you can get a single-chip solution for PTP time sync by running ptp4l inside of petalinux on the Zynq.

If you want to do that on a non-Zynq part, you would probably have to implement a good portion of ptp4l and a network stack on some sort of soft core.

alexforencich, 2020-08-12:
Why would there be a PCIe round trip in the loop? The timestamps are captured as the packet comes off the wire, then ptp4l processes the packet and timestamp at its convenience, and adjusts the clock step size. Nowhere does the PCIe round trip delay have any significant impact on the sync performance.

alexforencich, 2020-08-12:
Or are you saying that the time sync performance is highly dependent on the delay of the control loop, from packet arrival to clock adjustment? If you really need to lock that down, then it seems like maybe you'll want to implement the main part of the time servo itself in hardware so the delay is guaranteed to be constant, with the management logic implemented in a soft core or similar. However, it seems to me that when things are synchronized, the frequency adjustments will be small (handful of ppb), and a small delay in applying the adjustment (ms) will not have a significant impact on sync performance.

alexisfrjp, 2020-08-12:
Exactly, `PTP msg -> PCIe (t1) -> ptp4l -> PCIe (t2, != t1) -> ptp_clock(adjusts the TS clock inside the fabric)`

Comparing the accuracy of a PTP slave soft core and ptp4l (or sfptpd from Solarflare/Xilinx), we see hundreds of ns of difference.

That might not matter and most of the NICs are implemented that way anyway.

Thanks again.

alexforencich, 2020-08-12:
Well, the PCIe delay is likely going to be far less than the delay through the Linux kernel network stack and ptp4l, I would think. At any rate, corundum can consistently lock to better than 100 ns, at least as reported by ptp4l. I have not done a detailed analysis of the performance because I only need time sync to a precision of around 100 ns at the moment. I have also not tried adjusting the ptp4l control loop parameters, which I suspect may improve time sync performance as a it is right now there is quite a bit of variation in the frequency adjustment that ptp4l applies when it is locked.