corundum/corundum #14: Support For Intel FPGA?

piotrraczynski, 2020-08-20:
Do you have any plans to support one of the Intel FPGAs, like Arria 10 devices?

alexforencich, 2020-08-20:
We're looking at what it would take to support Intel P-tile (Stratix 10 DX, Agilex). But it will be a while as I don't have any hardware yet. Arria 10 is not planned, but I may be open to considering it if someone can provide hardware.

piotrraczynski, 2020-08-20:
Sure, Arria was just an example, thanks for quick info!

alexforencich, 2020-08-20:
The main thing is to port https://github.com/alexforencich/verilog-pcie to Intel devices.  Once that's working, then corundum should follow along as that repo provides both the DMA components and simulation framework.

shanefly1989, 2020-09-09:
@alexforencich Hi, i am porting this to Intel device. My target is Arria 10 GX 1150. Unfortunately, I am in china, so maybe i can't provide hardware for you... And i was hindered by a few confusion generate by the complexity of verilog-pcie... Could you give me some help? I will be very appreciate for that...

alexforencich, 2020-09-09:
The technical details of porting the PCIe DMA components would be better served on the verilog-pcie repo. Porting to other devices requires extensive addition of the hard IP core models to the simulation framework, creation of new variants of the DMA interface components, modules for splitting and merging TLPs, and possibly more. However, without a hardware donation, I will not add support to either repo as I cannot maintain the code.

alexforencich, 2020-10-12:
Take a look at the roadmap for more information concerning future device support: https://github.com/corundum/corundum/wiki/Corundum-Roadmap

Also, I may use Arria 10 as a stepping stone to supporting Stratix 10 DX.

piotrraczynski, 2020-10-13:
Thanks, is there any porting guide available?

alexforencich, 2020-10-13:
See: https://github.com/corundum/corundum/wiki/Porting-Guide.

Note that corundum cannot be ported until both verilog-pcie and verilog-ethernet support the device in question, and porting verilog-pcie to support the Intel device PCIe IP cores is non-trivial - I have to write new DMA interface libraries as well as TLP muxing and buffering logic, likely one set for Stratix 5/Arria 10/Cyclone 10, maybe one for Stratix 10 if that can't be supported via parameters, and one for Stratix 10 DX.  So, it will be some time before you'll be able to run it on your board.  

Out of curiosity, do you have a specific target board in mind?

alexforencich, 2022-08-19:
Corundum now supports the Intel H-Tile (Stratix 10 GX/SX/TX/MX) and P-Tile (Stratix 10 DX, Agilex).  Support for other Intel devices is TBD.