corundum/corundum #15: specify xci generation output directory?

TripRichert, 2020-09-05:
Currently, running the xci generation tcl scripts from the ip/ directory of a card puts the generated xci file in

<prjname>.srcs/sources_1/ip/<ipname>/<ipname>.xci

I would like more control of where the xci file is generated (to better support vivado nonproject flow).

I can do that by checking to see if a tcl variable is defined, and if so, specifying a directory with -dir argument on the create_ip.  If the variable isn't defined, it uses the default location.

This is what I've got in mind https://github.com/TripRichert/corundum/commit/210016e27baa7ff324bd1c2776c6e2d0f744f6bd?branch=210016e27baa7ff324bd1c2776c6e2d0f744f6bd&diff=unified

I recognize that this doesn't help your workflow in the slightest, and it pollutes your namespace, so I fully understand if you don't want this kind of change to your repo.  If you're open to the change, I'll submit a pull request.  If not, I'll work around it.

Thanks for building this great project!

alexforencich, 2020-09-14:
In principle I'm not opposed to doing something like this, but I think we'll have to find a way to make it cleaner.

TripRichert, 2020-09-15:
Ok, would putting it in a function that runs by default but could be deferred (with the caller able to pass in more arguments) be cleaner?

proc generate_the_ip { args } {
    create_ip -name gtwizard_ultrascale -vendor xilinx.com -library ip -module_name gtwizard_ultrascale_0
    create_ip -name gtwizard_ultrascale -vendor xilinx.com -library ip -module_name gtwizard_ultrascale_0 {*}$args
}

if { [info exists defer_ip_gen]  && $defer_ip_gen } {
} else {
    generate_the_ip("")
}