corundum/corundum #18: build VCU118 with vivado 2018.3

Winters123, 2020-10-10:
Hi Alex,

I followed the instructions in VCP118's 100g makefile to build Corundum on `vivado 2018.3` (win platform). But while I was looking into the options of 100g mac core, I found its not possbile to change the user interface from localbus to axi stream. Does that mean Corundum on VCU118 cannot be built with vavado 2018.3? Is there any heck that Corundum provided to replace localbuswith  AXIS as the interface?

Thanks,

alexforencich, 2020-10-10:
At some point AXI stream support was added to the CMAC cores in Vivado.  Must have been after 2018.3.  I'm currently using 2020.1, but it works fine on 2019.1 with a small modification to the CMAC interface to the transceivers.  Simplest solution would be to use a newer version of Vivado, but the AXIS-LBUS wrappers are pure verilog, so presumably it would be possible to back-port those separately.

Winters123, 2020-10-10:
I see. Thanks for the tip. Let me try to use a wrapper for that purpose. otherwise, a newer version would be a time-saving method to build the project. 

Thanks.

closed 2020-10-12 by Winters123