corundum/corundum #19: PTP mapping of timestamp and eth message

RenGraef, 2020-10-12:
Hi Alex, thanks for your nice repo! :)

I have a question regarding PTP and interaction with software driver and linux-ptp.
I understand that the PTP modules will create a timestamp for each and every eth packet and will be saved either in the RX or TX fifo. But how is the split done afterwards in the Software because i never saw an ID for the timestamps or etc? Because for Sync follow up message (master side) it's needed in the sent data. And for slave it's needed in general for all packets I saw.

And another question: Is mqnic kernel Software the only Software which is running (beside of linux-ptp) to fulfill the PTP tasks on a linux basis? And do you see a realistic chance to run the ptp stack without the help of linux as operating system? Or will this take lot's of hours  (100 and more). I just want to know what would come up to me because i am sure you have more knowledge and you can judge it well.

Thanks

alexforencich, 2020-10-12:
The timestamps are included in the completion records, which are tied to the original packets based on the queue number and index of the packet in the queue.  The driver then extracts the timestamps and inserts them into the SKBs during completion handling.  For RX, this is done for every packet.  For TX, this is only done when software has marked an SKB for hardware timestamping.  This is because the timestamped SKB must be returned to the SKB, and this is only done when it is requested in advance as it adds software overhead.  

Currently, you only need the kernel module and ptp4l (from linuxptp) for time synchronization to work.  

For running PTP without linux, are you talking about entirely on the FPGA, or with some other host OS?  

For running PTP completely on the FPGA, you would need some sort of a soft core, and then you would probably want to port PTP to run on that soft core, modified to interface with the hardware.  The PTP protocol is quite complicated with best master clock selection algorithms, etc.  And corundum is a NIC, so TBH it doesn't make much sense to consider how to use it without a host OS.  

With a different host OS, once you have a driver working and handling the timestamps as expected by the OS, then presumably you can port linuxptp or use a similar ptp implementation that already targets the OS in question.

RenGraef, 2020-10-12:
What do you exactly mean with "completion records"? Are we talking about the complete ethernet packet saved in the eth axis buffer, the forwared one to the udp complete or am i completely wrong? Is this step still in HW or in the SW kernel module?

Sorry for the missunderstanding. Yes i meant running on a simple soft core microcontroller like microBlaze or etc because i am not using it as a NIC. Therefore it's also required for me to get rid of all pcie communication and just use the axi stream or a simelar one.

alexforencich, 2020-10-12:
The completion record is part of the driver interface.  The driver writes out descriptors that describe the packet to be sent or buffer space to store a received packet (pointers and lengths), then corundum reads these, performs the operation, and writes out completion records to tell the driver what was done so the driver can either free up the buffer space or hand off packets to the OS.  

Well, if you're not using it as a NIC, then presumably you should just use the verilog-ethernet code instead.

alexforencich, 2020-10-12:
Actually, I guess I should mention that I am working on a version of corundum that can be used on a Zynq MPSoC part, with the ARM cores as the host.  Basically, an AXI host interface instead of PCIe.  So perhaps that could be used with a microblaze once I get some more of the kinks worked out.

RenGraef, 2020-10-12:
Ok understood. Yes I also saw the verilog ethernet repo and will also take this as a starting point. I just started to look in corundum because there is also the PTP clock module inserted by default and you added the kernel code etc. 

Ok nice to know, i will keep looking at your commints. And thanks again for the quick response and wish you a nice day!

closed 2020-10-12 by RenGraef