corundum/corundum #22: PCIe CORE Parameter AXIS_DATAWIDTH = 64 breaks the design.

kaoruzhu1, 2020-10-18:
The folowing PCIe CORE configuration broke the design:  

AXIS_DATAWIDTH = 64bit 
CORE_FREQ = 250MHz

compiler complained:
../lib/pcie/rtl/dma_client_axis_sink.v:239: error: Concatenation repeat may not be zero in this context.
../lib/pcie/rtl/dma_client_axis_sink.v:240: error: Concatenation repeat may not be zero in this context.
../lib/pcie/rtl/dma_client_axis_sink.v:319: error: Concatenation repeat may not be zero in this context.

It's  **iface[0].interface_inst.cpl_write_inst.dma_client_axis_sink_inst**   had errors.

plz see the flowing code:

dma_client_axis_sink #(
    .SEG_COUNT(SEG_COUNT),
    .SEG_DATA_WIDTH(SEG_DATA_WIDTH),
    .SEG_ADDR_WIDTH(SEG_ADDR_WIDTH),
    .SEG_BE_WIDTH(SEG_BE_WIDTH),
    .RAM_ADDR_WIDTH(RAM_ADDR_WIDTH),
    .AXIS_DATA_WIDTH(CPL_SIZE*8),                            ----------------> NO good for PCIe AXIS_DATAWIDTH=64
    .AXIS_KEEP_ENABLE(CPL_SIZE > 1),
    .AXIS_KEEP_WIDTH(CPL_SIZE),
    .AXIS_LAST_ENABLE(1),
    .AXIS_ID_ENABLE(0),
    .AXIS_DEST_ENABLE(0),
    .AXIS_USER_ENABLE(1),
    .AXIS_USER_WIDTH(1),
    .LEN_WIDTH(8),
    .TAG_WIDTH(CL_DESC_TABLE_SIZE)
)
dma_client_axis_sink_inst (
    .clk(clk),
    .rst(rst),
.....

alexforencich, 2020-10-18:
Presumably you mean AXIS_PCIE_DATA_WIDTH?  Why would you want to set that to 64?

Currently, CPL_SIZE is 32, so SEG_DATA_WIDTH*SEG_COUNT must be at least CPL_SIZE*8 (256), so AXIS_PCIE_DATA_WIDTH must be at least half of that (128).  Now, there should be an assert that triggers, but perhaps it is failing earlier than that due to the parameter values.

kaoruzhu1, 2020-10-19:
yes, it's not urgent.

I think G2x4 is a reasonable PCIe setting for 10g ETH.

It's expected that a design with 64bX250MHz bus setting has better performance than that with 128bX125MHz.

alexforencich, 2020-10-19:
64 bit at 250 MHz is probably also going to be lower resource consumption than 128 bit at 125 MHz.  What is your use case for that?  

The issue is really on the completion handling side of things; I'm probably going to revisit that relatively soon and I will keep this issue in mind when working on it to see if I can come up with something more reasonable that will work for a 64 bit PCIe interface.