corundum/corundum #23: eth_mac_10g_fifo.v minor bug

adi8v, 2020-10-20:
Minor typo when synchronising Ethernet MAC rx errors I believe:
corundum/fpga/lib/eth/rtl/eth_mac_10g_fifo.v
line 230
rx_sync_reg_1 <= rx_sync_reg_1 ^ {**rx_error_bad_frame_int**, rx_error_bad_frame_int};
should be rx_error_bad_fcs_int I think.

alexforencich, 2020-10-20:
Yes, I think you are correct about that, I will fix that in the verilog-ethernet codebase.

adi8v, 2020-10-21:
Ok, great. I didn't know whether to do a pull request or leave it to you. Do you periodically merge the verilog-ethernet codebase across to Corrundum?

alexforencich, 2020-10-21:
For simple changes like this, it's best to just let me know and then I will deal with it correctly. Some of these simple fixes - especially fixes in example designs - can also affect multiple files due to code duplication, so if your PR doesn't fix all of them in one shot then I would just do it myself anyway for consistency.

And yes, I will update the Ethernet code in here when it makes sense to do so, after verifying that those changes don't break anything.

alexforencich, 2021-09-14:
I forgot to mention that I merged the fix for this in back in February.  But, the latest set of changes move the async FIFOs into the new core logic modules, so the `eth_mac_10g_fifo` module is actually no longer used in Corundum; the `eth_mac_10g` module is used instead with separate FIFOs.

closed 2021-09-14 by alexforencich