corundum/corundum #24: Cannot connect to switch

Gaojiaqi, 2020-10-21:
I successfully set up the image, driver. When I connect the two ports on my U250 card directly, the os can send and receive packets. But when we connect the U250 to our switch, the switch cannot receive the packet. I tried to turn off the auto-negotiation and it did not work either. Do you have any hint that might solve this problem?

alexforencich, 2020-10-21:
What switch are you using? What kind of cable? And what link rate?

Gaojiaqi, 2020-10-21:
I'm using a Wedge 100B from EdgeCore: https://www.edge-core.com/productsInfo.php?cls=1&cls2=5&cls3=181&id=335.
The cable is 1 meter 100G QSFP28 DAC Twinax cable.
The link speed is 100G, I've tried both RS and None FEC options.

alexforencich, 2020-10-22:
Well, a DAC cable is less likely to have issues than an AOC, so that's good.  What does the switch say for the port status?

dmsspb, 2020-10-22:
> When I connect the two ports on my U250 card directly, the os can send and receive packets.

I guess your OS is Linux. If you looped two ports on a U250 without the Linux routing table trick, Linux loop the packet inside the kernel, no data will be sent to the U250.
If your IP of the link 'eth0' is X.X.X.X, you may try to send ARP packets to the interface: arping -c 100 -I eth0 X.X.X.Y. ARP will be sent to the physical link through U250.

alexforencich, 2020-10-22:
Ah yes, that's a good point. Simple sanity check: unplug the cable and see if that stops the flow of traffic.

Gaojiaqi, 2020-10-22:
The port status is down and no change at all after I tried to bring the interface up, down, add/remove the kernel module, etc.
Are you looking for something? Perhaps I can dig it out in the switch.

For the loopback testing, I'm sending packets directly to the interface via scapy and capturing packets on the other interface and it works.

I think it might be a cable issue, may I know what are the QSFP modules supported by the current CMAC?

dmsspb, 2020-10-22:
Corundum CMAC 100G "out of the box" has RS-FEC(528/514) and no AutoNeg(AN), i.e. 100GBASE-SR4. IMHO, Alveo should work with any QSFP28 AOC with multimode fiber (MMF) or 100GBASE-SR4 transceiver with MTP/MTO MMF.
Or 100GBASE-LR4 (single mode fiber) when AN, RS-FEC OFF.
Or 100GBASE-CR4 (DAC cables) when AN, RS-FEC ON.
In practice, short DAC cables work when the AN is disabled, but may not work.

Sorry for my question, but can you try this:
> Simple sanity check: unplug the cable and see if that stops the flow of traffic.

If traffic stops - DAC cable seems to be good

alexforencich, 2020-10-22:
I have had no issues with DACs the way corundum is configured, but I usually test with a Mellanox NIC on the other end instead of a switch.  

The annoying thing about AN is Xilinx charges a lot of money for the privilege of using AN, so I have not turned it on.

Gaojiaqi, 2020-11-03:
I connected it to another NIC and this works fine now. I think we can close this issue for now because it seems more likely to be a switch side issue.

closed 2020-11-03 by Gaojiaqi

TimmonSha, 2020-11-03:
hello, I have some trouble on loopback testing, could I ask you some questions?
I flashed the corundum bitstream into my U200 card and installed the driver. It seemed worked because it printed some information when I run dmesg commend. 
Then I connected two interfaces on my board, use scapy to send packets though one interface, and use tcpdump to capture packets on the other interface. It seems that the interface send packets successfully because I can see the number of packets sent when I run ifconfig, but tcpdump captured nothing on the other interface. 
So I want to ask what tools do you use to capture packets, and did you do any other configuration on the two interfaces when you test loopback, like configuring ip address? thanks a lot.

alexforencich, 2020-11-03:
It might make more sense to open a new issue or post on the mailing list instead of hijacking this one.