corundum/corundum #28: Can you share your Vivado version?

YangWang92, 2020-11-27:
Hi all, 
Thanks for sharing the interesting work and it helps me a lot in building an FPGA based NIC. 

When I compiled the project for vcu1525 on Vivado 2019.10, I got an error on connecting qsfp modules like this.

> ERROR: [Synth 8-448] named port connection 'gt_rxp_in' does not exist for instance 'qsfp0_cmac_inst' of module 'cmac_usplus_0'

I fixed it on my own and replace gt_rxp_in with gt0_rxp_in. 

It seems that different version of Vivado generates different ports name of qsfp modules.  

Can you share your Vivado version and update it to readme? That would help others.

If possible, can I pull a request to update cmac_usplus modules configurations?

Thanks!
Yang

alexforencich, 2020-11-27:
I'm on 2020.1 and I will be looking in to switching to 2020.2 in the next couple of weeks probably.  I think in 2020.1 they switched the CMAC GTY pin configuration from a bunch of 1 bit pins to a smaller set of 4 bit pins.  Sure, I should probably add a note to the readme for how to adjust the CMAC instances if you're using an old version of Vivado.

alexforencich, 2020-11-27:
See https://github.com/corundum/corundum/issues/5

YangWang92, 2020-11-27:
thanks a lot!

closed 2020-11-27 by YangWang92