corundum/corundum #29: Is it possible to support bump-in-wire?

YangWang92, 2020-11-27:
Hi all, 
For dual-port FPGA board, is it possible to support bump-in-wire based on the project? 

Thanks, 
Yang

alexforencich, 2020-11-27:
Sure.  The "interface" module just exposes an AXI stream interface to the MAC, you can insert whatever logic you like on that path.  However, I recommend disabling PTP timestamping when you do so.

YangWang92, 2020-12-02:
Thanks a lot! I'm trying to implement this based on simulation at first.

BTW, do you need any help supporting DPDK?

alexforencich, 2020-12-02:
No problem.  I'm actually in the process of moving the whole simulation framework over to cocotb, just FYI.  

Yeah, I want to start work on a DPDK PMD once I get variable-length descriptors working, as this will change the descriptor formats and a few other things about the driver-NIC communication.  There is definitely a lot of interest in using corundum with DPDK, but unfortunately I don't personally have much experience working with DPDK.  If you are interested in helping out with that, hop on the mailing list.

YangWang92, 2020-12-02:
BTW, DPDK supports IFPGA Rawdev Driver on Intel OPAE(because DPDK is maintained by intel now). Is it possible to bridge corundum with opae-xilinx and connect with DPDK. 

https://github.com/RSPwFPGAs/opae-xilinx

Are you planing to connect DPDK with corundum as a "normal" NIC like Intel 82599?

alexforencich, 2020-12-02:
I'm not familiar with OPAE.  Most likely it would be more work to learn the framework and port to that instead of porting to DPDK directly, unless there is some other reason to support the framework.  And with more layers, there is more opportunity for things to break in strange ways.  IME, acceleration frameworks are usually incompatible with networking applications as they are usually designed primarily for compute offload and not packet processing.  It's mainly a question of who is in charge - the FPGA or the host machine.  With things like the XDMA core, RIFFA, etc., the host machine is in charge and coordinates all of the operations.  But this is terrible for receiving packets that vary in size and arrive at random times.  And many of these frameworks seriously restrict what you can do with the card, for instance the Alveo framework doesn't provide direct access to PCIe, only a small selection of QDMA ports, which is not sufficient for high-performance networking.  Additionally you can't change the PCIe IDs, so it's not really feasible to bind a custom device driver.  So corundum must run "bare metal" on Alveo cards.  

Anyway, the idea would be to make it look like a normal NIC, but perhaps have some form of extension and/or metadata capability.  (although how else would something interface with DPDK?)

YangWang92, 2020-12-02:
You may take a look at https://xilinx.github.io/dma_ip_drivers/2019.1/DPDK/html/build.html

alexforencich, 2020-12-02:
Corundum does not use QDMA, so I'm not sure if that article is going to be very useful