corundum/corundum #30: Is there anything I need to modify when using U250/fpga_100g?

ParkGyuhwan, 2020-12-03:
Hello
I am trying to execute the codum with U250 now.
I'm following the AU250/fpga_100g execution method after watching the YouTube video.
(FPGA Dev Live Stream: More Corundum Porting : https://www.youtube.com/watch?v=oD6qlJXQMkg&t=2316s)

[Timing 38-282] The design failed to meet the timing requirements. 

These critical error messages keep appearing.
I wonder if there's anything else I need to touch that's not in the video. (e.g. modifying the clocks in the Transceivers Wizard, ...) 

I would appreciate if you reply. Thanks :)

alexforencich, 2020-12-03:
I think Vivado has been doing a really bad job of placement on the U250.  I am working on adding some placement constraints and some pipeline registers on the connection to the CMACs, but I have not committed those changes yet.  Unfortunately, it can be a bit of a crapshoot whether or not vivado picks a good placement or a terrible one, and placement constraints only help so much.

ParkGyuhwan, 2020-12-03:
Oh, I see. I'll look for a solution to my partial problems. Thank you for your reply and for your hard working on the project. If I have any questions, I will leave you an issue again next time. Thank you very much :)

alexforencich, 2020-12-03:
Try https://github.com/alexforencich/corundum and see if that closes timing

ParkGyuhwan, 2020-12-04:
I just finished executing in Vivado and the timing error disappeared. Thank you very much.

closed 2020-12-04 by ParkGyuhwan