corundum/corundum #31: Migration to customized FPGA board

xtfan, 2020-12-08:
Hi Alex, we have a customized fpga board with the same fpga chip as ZCU106.  Our board contains one SFP+ and a X8 PCIE interface. The optical module and EEPROM have dedicated I2C inteface. That are the major differences compared to ZCU106. To run  corundum on our board, we have modified two related verilog files: fpga.v and fpga_core.v. Results turn out that we can see the network interface eth0 when we install mqnic kernel module.  However, when we connect two boards directly via cable, we failed to ping from one board to another. We also noticed that there is no light sending out from the optical interface.  
Could you please give me some advices on how to debug and project migration? Thanks.

alexforencich, 2020-12-08:
If there is no light, then presumably the txdisable pin is pulled the wrong way, or you're not providing power to the SFP+ module.  If you have a custom board, I can't really say much more than that without looking at the schematic.

xtfan, 2020-12-08:
I just run another bitstream that setting sfp0_tx_disable_b to 0 in fpga.v. Now we can observe light from optical module. But still failed to ping.  Should we do some modification to the driver  accordingly?

alexforencich, 2020-12-08:
No driver modifications should be necessary.  However, there are some transient issues with the transceivers.  Specifically, GTH and GTY transceiver DFE can get stuck if there is no signal at the input for a while, requiring a transceiver reset to fix.  Implementing a proper fix for this is unfortunately a bit non-trivial, it's on my to-do list, but I have not gotten around to fixing it yet.  The workaround is to hot reset the whole card while there is a valid input signal.  So, load the design on both cards, connect them together, hot reset both cards, and then try again.

xtfan, 2020-12-11:
My card can work now. But there still remains one problem. When I configure IP address for eth0, for example, sudo ifconfig eth0 192.2.2.2,   and ping another IP, such as 192.2.2.3. Sometimes it can work, but sometimes not.  When it fails to ping, ifconfig shows that the assigned IP address for eth0 disappears. It seems that eth0 is not stable and will loss its IP address occasionally.

alexforencich, 2020-12-11:
That's a network manager problem.  Either disable network manager, configure network manager to ignore the interface, or use network manager to manage the interface and configure the IP.