corundum/corundum #33: Optimize iperf Performance of External Loopback in Dual-port Design

nightseas, 2020-12-09:
Hi Alex,

In your YouTube video I saw you isolated one CX5 card with network namespace and tested the traffic to/from corundum on the same server. While I don't have 100G NIC at hand, I'm thinking if it's feasible to test the performance of a dual-port corundum design with external loopback by connecting the two port with a DAC.

[image]

The iperf is functional with network namespace isolation on one port but the performance is not good. With one server/client process the throughput is only ~1.3Gbps, and with 10 server/client processes in parallel the total throughput only can reach ~10Gbps.

I also tried testing traffic of two FPGA cards with corundum 100G design on the same server and got similar results.

[image]

The server has two 14C/28T CPUs and runs Ubuntu 18.04 LTS (5.4.0-56 Kernel), which should not be the bottleneck, or should it?

So the question is what's your test environment to get throughput higher than 90Gbps? Two servers, one with FPGA the other with CX5?
Any specific software optimization needed? e.g. manually set CPU affinity for queues and interrupts?

Thanks & BR,
Xiaohai Li

beer-belly, 2020-12-16:
Hello, I have the same problem with you. The iperf's performance is very poor.  Have you solved it now

alexforencich, 2020-12-16:
If you have multiple NUMA nodes, you'll need to use `numactl` or similar to run everything that touches the NIC on the same NUMA node.  You can see the NUMA node associated with the PCIe device in `/sys/class/net/<dev>/device/numa_node`.  For iperf, you'll have to run both the client and the server under numactl.  So if sysfs reports NUMA node 3, then you'll want to run `numactl -N 3 iperf3 -s`.  It's always a good idea to compare the performance against a commercial 100G NIC as well.  

Strange that you're not getting anywhere near 10 Gbps even with one process; on the server I tested corundum on for the paper (dual socket Intel Xeon 6138), a single iperf process would run at 20-30 Gbps with 1500 byte MTU, or 40-50 Gbps with 9000 byte MTU.

alexforencich, 2020-12-16:
I'm putting together some notes on performance tuning here: https://github.com/corundum/corundum/wiki/Performance-Tuning

alexforencich, 2020-12-16:
I'm also going to take a look on my end and see if I missed any settings between what was used for the test in the paper and what is in the repo.   I know I added some instrumentation to the driver for measuring a few things, but this would not improve performance.  However, I may have adjusted a couple of other settings in the FPGA config that perhaps didn't make it into the repo.  Now that the cocotb migration is complete, I have some time to do some performance measurements, and I will be sure to put together some notes on that.

nightseas, 2021-02-13:
I've tested two corundum FPGAs (ADM9v3 and VCU118) on two separated servers. Although I haven't got 100Gbps line rate, single thread performance was increased to 30Gbps.
So I assume that namespace isolation has certain limitation comparing to two real physical nodes.
Thanks for your suggestions :)

From optimization perspective, the application (e.g. iperf), queue handling, and interrupt handling shall all be attached to the correct NUMA node to get stable performance.
Here's an example of optimizing eth0 with queue and interrupt CPU affinity to NUMA node 1. My server has two 14C28T E5v4 processors.

```bash
for i in {0..255}; do echo 0003ff,f0003fff > /sys/class/net/eth0/queues/rx-$i/rps_cpus ; done
for i in {0..8191}; do echo 0003ff,f0003fff > /sys/class/net/eth0/queues/tx-$i/xps_cpus ; done

for i in {83..114}; do echo 0-13,28-41 > /proc/irq/eth0/smp_affinity_list ; done
```

closed 2021-02-13 by nightseas

alexforencich, 2021-02-13:
Hmm, that's a good point, my main use case for namespace isolation was testing the UDP stack without having to screw around with IP addresses if my local network conflicts.  But for corundum, it's probably not sufficient for performance isolation, although I have been using it quite a bit recently as it's at least a decent tool for sanity checking.  

Now, interrupt handling: if you want to figure out how to do that automatically in the driver, I would gladly accept a pull request for that.  Although I would think linux *should* be smart enough to figure out that interrupts for a device should be bound to the same NUMA node.

nightseas, 2021-11-30:
> Hmm, that's a good point, my main use case for namespace isolation was testing the UDP stack without having to screw around with IP addresses if my local network conflicts. But for corundum, it's [...]

Recently I've switched my test platform from Broadwell to Skylake and have MLX5 installed to replace the VCU118. And most of the issues are disappeared :) 
With network namespace isolation the performance between an ADM9V3 and a MLX5 can reach 95~97Gbps on a single server (MTU=8K, have a problem on 9K MTU). The bottleneck appears to be on the CPU side (old ark and low frequency). Even so, on servers from different vendors, e.g. Supermicro and Dell, the performance are not same, which may be relative to the BIOS implementation.
And you are right about the interrupt handling. In up-to-date versions of Kernel, disabling irqbalance and having hard/soft IRQ manually assigned would not help on overall throughput.
I've tried all kinds of ways to optimize the performance on BIOS and OS level to the old Broadwell server and I have to laugh at myself that most of them are actually negative optimization. However this is still a really good lesson to learn ;)

Thanks & Best regards,
Xiaohai

canchen0720, 2025-02-25:
Hi , Xiaohai Li
I encountered the same issue as you did. While testing two 100G loops on a single board and running multiple iperf3 processes, the maximum performance I achieved was only 46Gbps（tried many different quantities of client/server parallel processes）. Could you share what modifications you made to improve the performance from 10Gbps?
some information about my test： 
namespace， numa node 0， MTU 9000，TCP buffer 2GB，using pcie gen3 x16，

（1）Base Board Information
	Manufacturer: Dell Inc.
	Product Name: 0KJCC5
	Version: A00
	Serial Number: /GVJZKF2/CN7220067100A2/
	Asset Tag: Not Specified
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0
（2）CPU
Architecture:                       x86_64
CPU op-mode(s):                     32-bit, 64-bit
Byte Order:                         Little Endian
Address sizes:                      46 bits physical, 48 bits virtual
CPU(s):                             64
On-line CPU(s) list:                0-63
Thread(s) per core:                 2
Core(s) per socket:                 16
Socket(s):                          2
NUMA node(s):                       2
Vendor ID:                          GenuineIntel
CPU family:                         6
Model:                              79
Model name:                         Intel(R) Xeon(R) CPU E5-2683 v4 @ 2.10GHz
Stepping:                           1
CPU MHz:                            1200.000
CPU max MHz:                        3000.0000
CPU min MHz:                        1200.0000
BogoMIPS:                           4190.17
Virtualization:                     VT-x
L1d cache:                          1 MiB
L1i cache:                          1 MiB
L2 cache:                           8 MiB
L3 cache:                           80 MiB
NUMA node0 CPU(s):                  0-15,32-47
NUMA node1 CPU(s):                  16-31,48-63
Vulnerability Gather data sampling: Not affected
Vulnerability Itlb multihit:        KVM: Mitigation: VMX disabled
Vulnerability L1tf:                 Mitigation; PTE Inversion; VMX conditional cache flushes, SMT vulnerable
Vulnerability Mds:                  Mitigation; Clear CPU buffers; SMT vulnerable
Vulnerability Meltdown:             Mitigation; PTI
Vulnerability Mmio stale data:      Mitigation; Clear CPU buffers; SMT vulnerable
Vulnerability Retbleed:             Not affected
Vulnerability Spec rstack overflow: Not affected
Vulnerability Spec store bypass:    Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:           Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:           Mitigation; Retpolines; IBPB conditional; IBRS_FW; STIBP conditional; RSB filling; PBRSB-eIBRS Not affected; BHI Not affected
Vulnerability Srbds:                Not affected
Vulnerability Tsx async abort:      Mitigation; Clear CPU buffers; SMT vulnerable
Flags:                              fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_
                                    perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1
                                     sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l3 cdp_l3 invpcid_single pti intel_ppin ssbd ibr
                                    s ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm rdt_a rdseed adx smap intel_pt xsaveopt cqm_
                                    llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts md_clear flush_l1d
（3）numa
available: 2 nodes (0-1)
node 0 cpus: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47
node 0 size: 32103 MB
node 0 free: 28783 MB
node 1 cpus: 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63
node 1 size: 32197 MB
node 1 free: 30783 MB
node distances:
node   0   1 
  0:  10  21 
  1:  21  10