corundum/corundum #36: Not working on Alveo U50DD(Alveo U50-ES3)

Perlmint, 2020-12-14:
Hi, I'm trying to use corundum on Alveo U50DD, but it seems not working.

U50DD has 2 SFP-DD instead of QSFP. This is the only difference with AU50, as I know. So, I programmed my U50DD with original corundum design for AU50 10g. After warm-reset or hot-reset and loading the mqnic module - MAC address is hardcoded -, It showed up.

I connected the first port of U50DD to other 10g NIC with DAC, assigned static IPs, run ping. Sending and receiving are failed.

Additionally, I tried examples in [verilog-ethernet](https://github.com/alexforencich/verilog-ethernet), [verilog-pcie](https://github.com/alexforencich/verilog-pcie). And both of them are working on the same setup.

Is there anything I missed or wrong?

alexforencich, 2020-12-14:
If the verilog-ethernet and verilog-pcie reference designs work, then corundum should also work as it uses the same interfaces in exactly the same way.  So that points to either a transient issue of some sort, or a software issue.  What is the link partner?  Do you see any inbound traffic in tcpdump (`tcpdump -i <iface> -Q in`) on either end of the link?

alexforencich, 2020-12-14:
Also, I have been slowly piecing together https://github.com/corundum/corundum/wiki/Frequently-Asked-Questions, have you tried everything on there?

Perlmint, 2020-12-15:
Thanks for your fast response.
I feel sorry about it was my network setup issue. Corundum works nicely on AU50DD.

Below is my wrong network setup.
I installed 10g NIC and AU50DD on the same machine. I assigned static IPs with 192.168.254.x to them. In this setup, tcpdump shows incoming packets on each interface. But, Failed to receive via socket and No response to ping.

I moved 10g NIC to another machine. After moving, It works.

closed 2020-12-15 by Perlmint