corundum/corundum #37: Is the Xilinx UniSim library under Apache 2.0 license useful to you?

mithro, 2020-12-14:
Hi!

I don't know if you saw, but [Xilinx did a release of their UniSim libraries under the Apache 2.0 license](https://github.com/Xilinx/XilinxUnisimLibrary). The SymbiFlow project is slowly working on making these libraries compatible with open source simulators @ https://github.com/SymbiFlow/XilinxUnisimLibrary like Icarus and Verilator.

Are these of any use to you?

Hope that helps,

Tim @mithro Ansell

alexforencich, 2020-12-15:
First glance: I see CMAC, PCIE, etc. Looks very interesting.  Second glance: CMAC pulls in SIP_* modules, which I'm not seeing?  Is the library incomplete?

alexforencich, 2020-12-15:
Same thing with the GTY stuff, GTYE4_COMMON pulls in SIP_GTYE4_COMMON, which is not present in the repo.  If all of the nontrivial stuff is missing, then the library is useless.

closed 2020-12-16 by alexforencich