corundum/corundum #38: U200 packet receiving performance test

beer-belly, 2020-12-16:
Hello, doctor
about U200 board performance test, I want to ask you some questions.

test environment:
We use Ixia device to do the packet test for U200 board. Ixia sets pseudo-random IP address and port number to ensure that packets can be evenly distributed to each queue, and the packet size is 9600 bytes
The main parameters of U200 board are as follows: Rx_ MTU=16384, rx_ fifo_ depth=131072, number_ Rx_ queue=4096, rx_ pkt_ table_ size=16

Test results:
Through PF_ Ring count instance packet receiving test, only supports 23Gbps no packet loss reception
Set the count in engine.v . The counter is set after the queue response is completed and the data is written to ram, it can reach 37Gbps and 40Gbps respectively by reading the register
In Xilinx's CMAC IP core, RX data is returned to TX for direct forwarding, and the loopback rate can reach 96Gbps

So what other configurations do I need to do to achieve the high-speed rate of 95.5Gbps mentioned in your paper

the result of 35Gbps flow
[image]
[image]

the result of 40Gbps flow
[image]
[image]

the result of 50Gbps flow
[image]
[image]

alexforencich, 2020-12-16:
Can you provide any more information about the host system?

beer-belly, 2020-12-17:
Thank you very much for your answer.
CPU:Intel(R) Xeon(R) CPU E5-2609 v3 @ 1.90GHz     physical id is : 2    processor is : 12
mem: 128G  DDR3 
[image]

I added counters in rx_engine.v
[image]
[image]

alexforencich, 2020-12-17:
How in the hell is QDMA running at 100G when the PCIe link is only 64 Gbps?

alexforencich, 2020-12-17:
And those are very old CPUs. I have had issues with some of the older Xeon CPUs not going over 50 Gbps, even with a x16 link.  I have not been able to figure out why, but it's not a Corundum problem as the Mellanox NICs I tested had the same problem.  Only thing I can think of is that it's some limitation of the CPU uncore.