corundum/corundum #39: Support MSI-X

alexforencich, 2020-12-16:
MSI-X supports more interrupt channels than MSI.  Additionally, some motherboards limit MSI to 1 channel, but do not limit MSI-X.

beer-belly, 2022-03-01:
[image]
The support of msix requires a certain amount of space to be allocated for the msix table in the bar space. So want to know how much register space is currently used, where should I start to set it?

alexforencich, 2022-06-03:
MSI-X support will be merged into this repo soon, pending testing on all target hardware: https://github.com/alexforencich/corundum/commit/21b0f014a506c2233177e4c5943389e3a2c39db1