corundum/corundum #44: U200 cannot receive packets

xiuwen8686, 2020-12-29:
I followed the step in the Readme, generated bitstream file and mqnic.ko then programed 100G design on U200, also inserted module after reboot host, configured MAC address by programming eeprom . Message from the dmesg looks fine. U200 host is running on Ubuntu18.04.

I connect U200 and CX-5 nic with 100g DAC and cable, then I just ping CX-5 nic from U200 , I can see arp packet send out in the U200 host wireshark, but received nothing. In the CX-5 nic host wireshark, it received arp packet form  U200 and did a response to that arp packet, the ping test failed.

Do you have any hint that might solve this problem?
Please give me some advises.
Thank you

alexforencich, 2020-12-30:
Have you tried a hot reset of the card before loading the driver?  There may be some reset-related issues that need to be rooted out.  Usually 100G is pretty reliable, it's usually 10G/25G that needs to be reset a couple of times, and improving that situation is on my to-do list.

xiuwen8686, 2021-06-26:
Sorry, I didn't notice that this reply was not sent successfully.
Sometimes it can work successfully after hot reset.
I appreciate your help very much.

WSEmma, 2022-03-30:
Hi xiuwen8686,
I noticed you successfully programed u200. Could I ask you how to program the u200? It semms that there is no physical interface to connect with JTAG. And there is no Corundum already in my u200 board, so I can't program it through pcie. In my case ,Vivado can't  detect the hardware xcu200……after I open hardware manager
Thanks for your help!
Best