corundum/corundum #46: Test Issue about Constant User Function

happyPisces, 2021-02-02:
Hi Alex,

When I ran the test using python3 test_fpga_core.py, I met the issue as follows,

Running test...
../rtl/fpga_core.v:1075: sorry: constant user functions are not currently supported: w_32().
../lib/axi/rtl/axil_interconnect.v:134: sorry: constant user functions are not currently supported: calcBaseAddrs().
../rtl/fpga_core.v:1128: sorry: constant user functions are not currently supported: w_32().
../lib/axi/rtl/axil_interconnect.v:134: sorry: constant user functions are not currently supported: calcBaseAddrs().
../rtl/fpga_core.v:1128: error: Unable to evaluate parameter M_ADDR_WIDTH value: {}
../rtl/fpga_core.v:1075: error: Unable to evaluate parameter M_ADDR_WIDTH value: 2{}
../rtl/common/interface.v:879: sorry: constant user functions are not currently supported: w_32().
../lib/axi/rtl/axil_interconnect.v:134: sorry: constant user functions are not currently supported: calcBaseAddrs().
../rtl/common/interface.v:879: sorry: constant user functions are not currently supported: w_32().
../lib/axi/rtl/axil_interconnect.v:134: sorry: constant user functions are not currently supported: calcBaseAddrs().
../rtl/common/interface.v:879: error: Unable to evaluate parameter M_ADDR_WIDTH value: {{}, , , , , , , }
../rtl/common/interface.v:879: error: Unable to evaluate parameter M_ADDR_WIDTH value: {{}, , , , , , , }
../rtl/common/port.v:746: sorry: constant user functions are not currently supported: w_32().
../lib/axi/rtl/axil_interconnect.v:134: sorry: constant user functions are not currently supported: calcBaseAddrs().
../rtl/common/port.v:746: sorry: constant user functions are not currently supported: w_32().
../lib/axi/rtl/axil_interconnect.v:134: sorry: constant user functions are not currently supported: calcBaseAddrs().
../rtl/common/port.v:746: error: Unable to evaluate parameter M_ADDR_WIDTH value: 2{}
...

Do you have any hint that might solve this problem?
Please give me some advises.
Thank you.

alexforencich, 2021-02-02:
What simulator are you using?

alexforencich, 2021-02-02:
And exactly what file are you running?  cocotb doesn't print out "Running test..." (at least, I didn't see that in the output when I ran it myself), and you can't run the script directly from python at the moment, only via makefiles or via pytest.  The old myhdl tests are deprecated and have been removed, and AFAIK those only supported icarus verilog, which is fine with constant functions.

happyPisces, 2021-02-02:
I am running the old myhdl tests, It is strange that some constant functions are supported, such as hash_toep.

alexforencich, 2021-02-02:
What is the output of `iverilog -v`?

happyPisces, 2021-02-02:
Icarus Verilog version 0.9.7  (v0_9_7)

alexforencich, 2021-02-02:
Well.  There's your problem, Icarus Verilog 10.0 was released in 2015.  Current stable version is 11.0.  Your simulator is more than 6 years old, might be time to upgrade.

happyPisces, 2021-02-02:
Thank you. By the way, for the new version using cocotb, would you please give the minimum version requirements of the relevant tools, such as python, pytest, cocotb and so on?

alexforencich, 2021-02-02:
Currently several of the cocotb extensions that I have written - along with cocotb itself - are evolving along with corundum.  So you'll definitely want to run the latest versions of those.  But python has a package manager called pip, so as long as you install them from pip, you should be fine.  I won't push anything to this repo that doesn't work with the versions of the tools on pip, as tox runs the tests on github actions using the pip versions of everything.  However, there is at least one major breaking change that I will be making at some point, just FYI.  But again, I will make sure the pip packages and the corundum testbenches are updated together.  Not sure what python version would be best, latest will definitely work, but I think it will probably work all the way back to python 3.6.