corundum/corundum #48: Interfacing CMAC to PCIe x8

nightseas, 2021-02-16:
Hi Alex,

I'm porting Corundum to a customized VU9P FPGA board with x8 PCIe and QSFP28 interfaces.
The board works fine with 10G/25G designs. But with 100G design there is issue sending/receiving packages.

The purpose is not to fully utilize the 100G bandwidth (limited by PCIe x8) but to save logic resources with CMAC hard IP.

### Issue Description

Almost all packages are dropped (checked in ifconfig) and a few captured by tcpdump show incomplete data received.
I guess there's something wrong after changing PCIe AXIS parameters. Half of the data are missing in the captured package.

Ping package sent from other side with pktsize = 256:
```
        0x0000:  4500 011c 86f2 0000 4001 719b c0a8 0001  E.......@.q.....
        0x0010:  c0a8 0002 0000 4529 11cd 0003 ddbe 1660  ......E).......`
        0x0020:  0000 0000 23e8 0900 0000 0000 1011 1213  ....#...........
        0x0030:  1415 1617 1819 1a1b 1c1d 1e1f 2021 2223  .............!"#
        0x0040:  2425 2627 2829 2a2b 2c2d 2e2f 3031 3233  $%&'()*+,-./0123
        0x0050:  3435 3637 3839 3a3b 3c3d 3e3f 4041 4243  456789:;<=>?@ABC
        0x0060:  4445 4647 4849 4a4b 4c4d 4e4f 5051 5253  DEFGHIJKLMNOPQRS
        0x0070:  5455 5657 5859 5a5b 5c5d 5e5f 6061 6263  TUVWXYZ[\]^_`abc
        0x0080:  6465 6667 6869 6a6b 6c6d 6e6f 7071 7273  defghijklmnopqrs
        0x0090:  7475 7677 7879 7a7b 7c7d 7e7f 8081 8283  tuvwxyz{|}~.....
        0x00a0:  8485 8687 8889 8a8b 8c8d 8e8f 9091 9293  ................
        0x00b0:  9495 9697 9899 9a9b 9c9d 9e9f a0a1 a2a3  ................
        0x00c0:  a4a5 a6a7 a8a9 aaab acad aeaf b0b1 b2b3  ................
        0x00d0:  b4b5 b6b7 b8b9 babb bcbd bebf c0c1 c2c3  ................
        0x00e0:  c4c5 c6c7 c8c9 cacb cccd cecf d0d1 d2d3  ................
        0x00f0:  d4d5 d6d7 d8d9 dadb dcdd dedf e0e1 e2e3  ................
        0x0100:  e4e5 e6e7 e8e9 eaeb eced eeef f0f1 f2f3  ................
        0x0110:  f4f5 f6f7 f8f9 fafb fcfd feff            ............
```

Received ping package:
```
        0x0000:  4500 011c 86f2 0000 4001 719b c0a8 0001  E.......@.q.....
        0x0010:  c0a8 0002 0000 4529 11cd 0003 ddbe 1660  ......E).......`
        0x0020:  0000 0000 23e8 0900 0000 0000 1011 1213  ....#...........
        0x0030:  1415 0000 0000 0000 0000 0000 0000 0000  ................
        0x0040:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0050:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0060:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0070:  0000 5657 5859 5a5b 5c5d 5e5f 6061 6263  ..VWXYZ[\]^_`abc
        0x0080:  6465 6667 6869 6a6b 6c6d 6e6f 7071 7273  defghijklmnopqrs
        0x0090:  7475 7677 7879 7a7b 7c7d 7e7f 8081 8283  tuvwxyz{|}~.....
        0x00a0:  8485 8687 8889 8a8b 8c8d 8e8f 9091 9293  ................
        0x00b0:  9495 0000 0000 0000 0000 0000 0000 0000  ................
        0x00c0:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x00d0:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x00e0:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x00f0:  0000 d6d7 d8d9 dadb dcdd dedf e0e1 e2e3  ................
        0x0100:  e4e5 e6e7 e8e9 eaeb eced eeef f0f1 f2f3  ................
        0x0110:  f4f5 f6f7 f8f9 fafb fcfd feff            ............
```

### Design Modifications

The design baseline is VCU118, with PCIe config referred to ExaNIC 25G.
Beside the changes below, anything else shall be take care of?
Or maybe should I ask if it's possible to interface PCIe x8 with CMAC without hacking into interface modules or DMA engine?

PCIe IP changes:
```diff
@@ -3,10 +3,11 @@ 
 set_property -dict [list \
     CONFIG.PL_LINK_CAP_MAX_LINK_SPEED {8.0_GT/s} \
-    CONFIG.PL_LINK_CAP_MAX_LINK_WIDTH {X16} \
-    CONFIG.AXISTEN_IF_EXT_512_RQ_STRADDLE {false} \
+    CONFIG.PL_LINK_CAP_MAX_LINK_WIDTH {X8} \
+    CONFIG.AXISTEN_IF_RC_STRADDLE {false} \
     CONFIG.axisten_if_enable_client_tag {true} \
-    CONFIG.axisten_if_width {512_bit} \
+    CONFIG.axisten_if_width {256_bit} \
+    CONFIG.extended_tag_field {true} \
     CONFIG.axisten_freq {250} \
     CONFIG.PF0_CLASS_CODE {020000} \
     CONFIG.PF0_DEVICE_ID {1001} \
@@ -20,7 +21,13 @@
     CONFIG.pf0_bar0_64bit {true} \
     CONFIG.pf0_bar0_prefetchable {true} \
     CONFIG.pf0_bar0_scale {Megabytes} \
-    CONFIG.pf0_bar0_size {16} \
+    CONFIG.pf0_bar0_size {64} \
     CONFIG.vendor_id {1234} \
```

fpga.v changes:
```diff
     /*
      * PCI express
      */
-    input  wire [15:0]  pcie_rx_p,
-    input  wire [15:0]  pcie_rx_n,
-    output wire [15:0]  pcie_tx_p,
-    output wire [15:0]  pcie_tx_n,
+    input  wire [7:0]   pcie_rx_p,
+    input  wire [7:0]   pcie_rx_n,
+    output wire [7:0]   pcie_tx_p,
+    output wire [7:0]   pcie_tx_n,
+    input  wire         pcie_refclk_0_p,
+    input  wire         pcie_refclk_0_n,
     // input  wire         pcie_refclk_1_p,
     // input  wire         pcie_refclk_1_n,
-    input  wire         pcie_refclk_2_p,
-    input  wire         pcie_refclk_2_n,
     input  wire         pcie_reset_n,

...

-parameter AXIS_PCIE_DATA_WIDTH = 512;
+// PCIe bus config refered to ExaNIC_X25
+parameter AXIS_PCIE_DATA_WIDTH = 256;
 parameter AXIS_PCIE_KEEP_WIDTH = (AXIS_PCIE_DATA_WIDTH/32);
-parameter AXIS_PCIE_RC_USER_WIDTH = 161;
-parameter AXIS_PCIE_RQ_USER_WIDTH = 137;
-parameter AXIS_PCIE_CQ_USER_WIDTH = 183;
-parameter AXIS_PCIE_CC_USER_WIDTH = 81;
+parameter AXIS_PCIE_RC_USER_WIDTH = 75;
+parameter AXIS_PCIE_RQ_USER_WIDTH = 62;
+parameter AXIS_PCIE_CQ_USER_WIDTH = 88;
+parameter AXIS_PCIE_CC_USER_WIDTH = 33;
 parameter RQ_SEQ_NUM_WIDTH = 6;
-parameter BAR0_APERTURE = 24;
+parameter BAR0_APERTURE = 26;
 
 parameter AXIS_ETH_DATA_WIDTH = 512;
 parameter AXIS_ETH_KEEP_WIDTH = AXIS_ETH_DATA_WIDTH/8;
```

Thanks & BR.

nightseas, 2021-02-16:
One way I'm thinking is to replace the mac_tx_fifo/mac_rx_fifo with axis_async_fifo_adapter to adapt 512-bit AXI stream to 256-bit.

alexforencich, 2021-02-16:
The only restriction is that the AXI stream interface width on the DMA clients must be no wider than the segmented interface, which is 2x the PCIe interface width.  There was a bug wrt. that actually that I ran in to with the ZCU106 PCIe design as that one is only PCIe gen 3 x4 (128 bits), and the DMA client for writing completions is 256 bits wide.  So using a 512 bit AXI stream interface with a 256 bit PCIe interface *should* work, but I have never tested it in sim, so it's entirely possible there is some other bug.  I'll run some tests and see if I find something obvious.  

Interesting that you're seeing exactly 64 bytes of all 0s, that definitely narrows down what might be going on.  Looks like it's exactly every other cycle on the 512 bit AXI stream interface.  Are you seeing this in the receive direction only, the transmit direction only, or both?  And can you show the full packet hex dump, not just the Etherney payload?

alexforencich, 2021-02-16:
Replicated in sim:

    118403.56ns INFO     cocotb.tb                             test_fpga_core.py:413  in run_test_nic                    Packet: Packet(data=b'\xda\xd1\xd2\xd3\xd4\xd5ZQRSTU\x08\x00E\x00\x01\x1c\x00\x01\x00\x00@\x11\xf5\xb6\xc0\xa8\x01d\xc0\xa8\x01e\x00\x01\x00\x02\x01\x08\x17\x1f\x00\x01\x02\x03\x04\x05\x06\x07\x08\t\n\x0b\x0c\r\x0e\x0f\x10\x11\x12\x13\x14\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00VWXYZ[\\]^_`abcdefghijklmnopqrstuvwxyz{|}~\x7f\x80\x81\x82\x83\x84\x85\x86\x87\x88\x89\x8a\x8b\x8c\x8d\x8e\x8f\x90\x91\x92\x93\x94\x95\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd6\xd7\xd8\xd9\xda\xdb\xdc\xdd\xde\xdf\xe0\xe1\xe2\xe3\xe4\xe5\xe6\xe7\xe8\xe9\xea\xeb\xec\xed\xee\xef\xf0\xf1\xf2\xf3\xf4\xf5\xf6\xf7\xf8\xf9\xfa\xfb\xfc\xfd\xfe\xff', timestamp_s=0, timestamp_ns=0, rx_checksum=0x7acc)

I'll see if I can sort that out in the next couple of days.

alexforencich, 2021-02-16:
Well, I just added another test case to the DMA client source/sink modules where the AXI stream width and RAM interface widths are the same...sink was fine, as I fixed the issue there last time around, but the source is bugged.  Fix is simple, just replace

    m_axis_read_data_tdata_int = ram_rd_resp_data >> (((addr_reg >> PART_OFFSET_WIDTH) & {PART_COUNT_WIDTH{1'b1}}) * AXIS_DATA_WIDTH);

with

    if (PART_COUNT > 1) begin
        m_axis_read_data_tdata_int = ram_rd_resp_data >> (((addr_reg >> PART_OFFSET_WIDTH) & {PART_COUNT_WIDTH{1'b1}}) * AXIS_DATA_WIDTH);
    end else begin
        m_axis_read_data_tdata_int = ram_rd_resp_data;
    end

in dma_client_axis_source.v.

nightseas, 2021-02-16:
Good good! 
My test environment is down now during holiday. I'll try to verify it later and let you know the results.

Thanks a lot.

nightseas, 2021-02-18:
Fix verified with iperf traffic on my board. It works now!

closed 2021-02-18 by nightseas