corundum/corundum #50: Add support for adm-pcie-8v3

Applepi, 2021-03-02:
https://www.alpha-data.com/dcp/products.php?product=adm-pcie-8v3

I have a couple of these FPGA cards and I would like to have support for these to test out 100gbps networks. Also will MAC splitting work with this firmware? If I were to take the QSFP28 port and split to 4 SFP28 connectors would I be able to have 4 seperate 25gbps links?

Please let me know if there is anything I could assist with in adding support for these devices.

Thanks,

alexforencich, 2021-03-02:
UltraScale is not so great for 100G.  The PCIe hard IP only supports gen 3 x8, which is only 64 Gbps, and Corundum doesn't currently support combining two PCIe interfaces.  Also, the hard CMACs on UltraScale do not include RS-FEC, which is required by a lot of 100G devices.  On these devices, RS-FEC must be implemented in soft logic, which likely means purchasing a license from Xilinx ($$$).  This is in contrast to UltraScale+, which supports gen 3 x16 and gen 4 x8 (on HBM parts) and has newer CMACs that include hard RS-FEC, so that series will work just fine at 100 Gbps without any paid IP licenses.  

The CMACs only support operation at 100G (annoyingly, they don't even support 40G...).  However, I usually make 25 Gbps variants that have 4x 25G MACs instantiated instead of 1x CMAC per QSFP28, and then each of those can be used as a separate link.  So you have to swap out the FPGA config to switch configurations, it's not a run time switch.  Eventually I want to make the 25G designs 10G/25G switchable.  

If you want to experiment with the card, take a look at the VCU108 design, presumably you'll be able to drop that on the card without too much trouble as it's also an XCVU095 part.  For upstream support for that card in this repo, you'll have to donate a card somehow so I can maintain the design on that card.

Applepi, 2021-03-03:
Okay great, I will attempt to establish 4x10gbps links using a QSFP+ module and a Connect-X 3 Card I have. 

I also have some Mellanox Innova 2 cards, that have UltraScale+ FPGAs on them, coming in but I don't think this would work without taking the signals from the OPENCAPI interface and wiring to transceivers. The SFP28 connectors are wired to the connect-x 5 NIC instead of the FPGA directly. They would likely need a lot of work to get something hacked together, but for internally connected networks you could still have all the features of the physical NIC but for a virtual interface. I.E. to a RISC soft cores or accelerator logic.