corundum/corundum #52: Understanding the queue management and DMA modules in Corundum

Winters123, 2021-03-31:
Hi Alex,

I'm interested in understanding the detailed design of the DMA engine and queue mechanisms of Corundum. Is there any technical document or other stuff that could help to understand the code?

Thanks.

alexforencich, 2021-04-02:
Unfortunately, I don't have much documentation at this time aside from the FCCM paper, my thesis, and what's on the github wiki.  Corundum has been a bit of a moving target; I don't want to spend hours and hours writing documentation for a module that ends up being totally rewritten.  For example, I'm in the middle of an overhaul of the descriptor management logic that includes a rewrite of all of the queue management logic, so it makes little sense for me to document how all that works right now as it's all going to get tweaked.  If you do have any questions, feel free to ask, either here or on the mailing list.

happyPisces, 2021-08-06:
Hi Alex,

I  would like to get the information about DMA registers, such as name, address, function of each bit and so on. If you could help list some, I would appreciate that.

Thanks.