corundum/corundum #53: mqnic module compile error

kimanha, 2021-04-01:
make -C /lib/modules/4.18.0-193.28.1.el8_2.x86_64/build M=/home/kha/fpga/corundum/modules/mqnic modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-193.28.1.el8_2.x86_64'
  CC [M]  /home/kha/fpga/corundum/modules/mqnic/mqnic_tx.o
In file included from ./include/linux/kernel.h:10,
                 from /home/kha/fpga/corundum/modules/mqnic/mqnic.h:37,
                 from /home/kha/fpga/corundum/modules/mqnic/mqnic_tx.c:35:
/home/kha/fpga/corundum/modules/mqnic/mqnic_tx.c: In function ‘mqnic_start_xmit’:
/home/kha/fpga/corundum/modules/mqnic/mqnic_tx.c:526:22: error: ‘struct sk_buff’ has no member named ‘xmit_more’
     if (unlikely(!skb->xmit_more || stop_queue))
                      ^~
./include/linux/compiler.h:77:42: note: in definition of macro ‘unlikely’
 # define unlikely(x) __builtin_expect(!!(x), 0)
                                          ^
make[2]: *** [scripts/Makefile.build:313: /home/kha/fpga/corundum/modules/mqnic/mqnic_tx.o] Error 1
make[1]: *** [Makefile:1545: _module_/home/kha/fpga/corundum/modules/mqnic] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-193.28.1.el8_2.x86_64'
make: *** [Makefile:18: all] Error 2

alexforencich, 2021-04-01:
RHEL/centos is not currently supported.  This is in large part because the RHEL kernel version is a total lie due to all of their patching and back-porting.  Most likely corundum will not support RHEL unless someone decides to sponsor the maintenance of the driver.

kimanha, 2021-04-01:
I see, I am using the RHEL 8.2, which means I can’t run the RHEL environment?
What Linux OS recommended for the AU50 or AU280 target board?

Thanks,

alexforencich, 2021-04-01:
Ubuntu, debian, arch, etc.  Basically any distribution other than redhat/centos should work fine.

kimanha, 2021-04-01:
Ok, thank you. I am available Debian 9.

Can you please give some basic information about the test procedure?
For example, corundum for the AU50 10 G tests instead of 100G.

  1.  Connect DAC cable with Ethernet cards.
  2.  FPGA bitstream download using HW-DMB-1<https://www.digikey.com/en/products/detail/xilinx-inc/HW-DMB-1-G/11588469> ?
Do you have download tools throw PCIe?

  1.  Server hard reboot or soft reboot?
  2.  Module/ mqnic driver module loading?
  3.  How to setup MAC address and IP address?
  4.  Ping test from another INC and check Wireshark?
  5.  Standard UDP or TCP/IP server/client test?

Any particular procedure for starting?

Thanks,

alexforencich, 2021-04-01:
I guess I don't have a getting started guide on the wiki yet; I should put something together.  At any rate, here are the basics.  You should be able to use pretty much any QSFP or QSFP28 solution with the design (DAC, AOC, optical transceivers of various types), though if you want to use a QSFP28 AOC or optical module at 10G on the U50 or U280 you may run in to problems as you can't adjust the transceiver configuration via I2C on those boards due to lack of support in the BMC firmware.  You can also use SFP+ and SFP28 DACs, AOCs, and optical modules with the appropriate adapter.  

To load the design onto a fresh card, you'll likely need to use JTAG.  There are a few different ways to do this.  One is to write to the flash directly from Vivado.  This is a very slow process, I don't recommend it if you have to flash a bunch of cards.  The faster way is to load the design into the FPGA directly, then do a warm reboot so you can talk to the card from linux, and use mqnic-fw to flash the design over PCIe.  It may even be possible to do this without rebooting.  Once the design is loaded into flash, you can use mqnic-fw to update it over PCIe.  The mqnic-fw utility does not require the kernel module; it can either interface with the card via the kernel module or via sysfs if the kernel module is not loaded.  

Once the card is installed and the driver is loaded, the driver will attempt attempt to read persistent MAC addresses from the card.  See https://github.com/corundum/corundum/wiki/Persistent-MAC-Addresses for more information about this.  For the U50, it should automatically read the MAC addresses out of the BMC.  

After that, it should behave like a normal network card in linux, so you can use iproute2 (ip link, ip addr, etc.) or ifconfig to configure the IP addresses and what not.  For simple testing, you can use ping, iperf, netperf, etc.

alexforencich, 2021-04-02:
I just added: https://github.com/corundum/corundum/wiki/Getting-Started

kimanha, 2021-04-02:
It looks perfect for starting point.
I did build AU50 100g and 10g bitstream files.
Next step I will test the hardware, I think it takes time.
Thank you so much.

muzzammilashraf, 2023-05-09:
@alexforencich Hi, if it is required then I can add support for RHEL/CentOS in the driver.

alexforencich, 2023-05-09:
The issue is not so much simply "adding support," but continuous maintenance and testing going forward.  RHEL lies about the kernel version number and back-ports all sorts of stuff from the mainline kernel, so there would need to be a complete set of RHEL-specific kernel version ifdefs that's parallel to the existing ones for the mainline Linux kernel.  If you're interested in helping to implement and maintain all of the RHEL kernel release specific ifdefs in the driver going forward, then we can certainly discuss that.

muzzammilashraf, 2023-05-10:
Yes, I am interested in adding support for RHEL/CentOS in the driver. We can discuss.

alexforencich, 2023-05-10:
Got it.  Please make sure you're on the Corundum zulip (https://corundum.zulipchat.com/). I know the folks from MLE are also potentially looking in to doing some stuff with CentOS as well, so there is probably going to be some opportunity for cooperation along those lines.

muzzammilashraf, 2023-05-10:
I have joined Corundum zulip.