corundum/corundum #58: Synth Error

bizhaan47, 2021-05-29:
While synthesizing the code, I get the following error:
Synth Error: Error: AXI lite address width too narrow (instance) [tx_scheduler_rr.v ": 150]
While line 149 equals:
 if (AXIL_ADDR_WIDTH <QUEUE_INDEX_WIDTH+5) where AXIL_ADDR_WIDTH is 16 and QUEUE_INDEX_WIDTH is 6!
also:
part-select [15: 2] out of range of prefix 'axil_ctrl_awaddr' [mqnic_port.v ": 622]
And:
conditional expression could not be resolved to a constant [mqnic_interface.v ": 2025]

alexforencich, 2021-05-29:
That means the PCIe BAR size is too small for the settings that you selected

bizhaan47, 2021-05-30:
i am a beginner and work with vivado on windows. how can i fix it?

alexforencich, 2021-05-30:
Did you change any of the parameters in fpga_core.v? Or are you trying to build the "stock" design straight out of the repo?

bizhaan47, 2021-05-30:
No, I have not changed anything. Can you guide me step by step to solve my problems? Note that I do not work with Linux.

alexforencich, 2021-05-30:
Hmmmm, that's very odd. All of the designs in the repo will build without issues like this. I test build everything before anything gets pushed to the main corundum repo.

Which design are you trying to build? Walk me through exactly what you've done to build the design.

Also, you'll need to use Linux at some point as the repo only has drivers for Linux. And the design is pretty useless without the drivers.

bizhaan47, 2021-05-30:
I have a board just like the NetFPGA SUME board. At this point in my work, I just need to synthesize and implementation the code of a NIC on the board, and for the next steps, I will definitely go to Linux. So, I downloaded the code from "https://github.com/corundum/corundum". I'm confused; So I added most of the files to my project. So i added the files in the "corundum-master\fpga\common\rtl" path to my new project. My project does not already have any code. I also added other files from the "corundum-master\fpga\lib\eth\rtl" and "corundum-master\fpga\lib\pcie\rtl" folders to the code.
I emphasize that my goal is just to synthesize and implement corundum code to build a NIC bit file. I just did the same thing and encountered the problems I mentioned.

alexforencich, 2021-05-30:
Are you trying to build the NetFPGA SUME design as-is?  (That would be a good place to start, before you start adjusting it to work on your board)  Open up the makefile in fpga/mqnic/NetFPGA_SUME/fpga/fpga.  That has all of the files you need listed.  The top-level module is fpga.v.  It's possible you have the wrong top-level file selected.  Also note that you'll have to source all of the IP TCL files from the TCL console in Vivado to create the IP cores for PCIe and for the serializers.

bizhaan47, 2021-06-14:
Hello. I collected all the said files from Makefile from the relevant folders and put them together but I do not know how to add tcl files to the project! you can help me? I have 3 tcl files. Is there a need for another job after this?

alexforencich, 2021-06-14:
You should be able to add the tcl files the same way you add other timing constraints.

bizhaan47, 2021-06-15:
Hi, I tried to run NetFPGA SUME code on the Virtex 7 690T 1927 -3, but I encountered the following problem while place_design. you can help me?
"[Place 30-510] Unroutable Placement! A GTHE_COMMON / GTHE_CHANNEL clock component pair is not placed in a routable site pair. The GTHE_COMMON component can use the dedicated path between the GTHE_COMMON and the GTHE_CHANNEL if both are placed in the same clock region. If this sub optimal condition is acceptable for this design, you may use the CLOCK_DEDICATED_ROUTE constraint in the .xdc file to demote this message to a WARNING. However, the use of this override is highly discouraged. These examples can be used directly in the .xdc file to override this clock rule.
< set_property CLOCK_DEDICATED_ROUTE FALSE [get_nets sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_gt_common_block/qplloutclk_out] >

sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_gt_common_block/gthe2_common_0_i (GTHE2_COMMON.QPLLOUTCLK) is provisionally placed by clockplacer on GTHE2_COMMON_X1Y9
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.QPLLCLK) is locked to GTHE2_CHANNEL_X1Y39
 sfp_2_pcs_pma_inst/inst/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.QPLLCLK) is locked to GTHE2_CHANNEL_X1Y38
 sfp_3_pcs_pma_inst/inst/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.QPLLCLK) is locked to GTHE2_CHANNEL_X1Y37
 sfp_4_pcs_pma_inst/inst/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.QPLLCLK) is locked to GTHE2_CHANNEL_X1Y35

The above error could possibly be related to other connected instances. Following is a list of 
all the related clock rules and their respective instances.

Clock Rule: rule_gt_bufg
Status: PASS 
Rule Description: A GT driving a BUFG must be placed on the same half side (top/bottom) of the device
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.TXOUTCLK) is locked to GTHE2_CHANNEL_X1Y39
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_shared_clock_reset_block/txoutclk_bufg_i (BUFG.I) is provisionally placed by clockplacer on BUFGCTRL_X0Y31

Clock Rule: rule_gt_bufhce
Status: PASS 
Rule Description: A GT driving a BUFH must both be in the same horizontal row (clockregion-wise)
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.RXOUTCLK) is locked to GTHE2_CHANNEL_X1Y39
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/ten_gig_eth_pcs_pma_0_local_clock_reset_block/rxoutclk_bufh_i (BUFH.I) is provisionally placed by clockplacer on BUFHCE_X1Y119

Clock Rule: rule_bufh_bufr_ramb
Status: PASS 
Rule Description: Reginal buffers in the same clock region must drive a total number of brams less
than the capacity of the region
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/ten_gig_eth_pcs_pma_0_local_clock_reset_block/rxoutclk_bufh_i (BUFH.O) is provisionally placed by clockplacer on BUFHCE_X1Y119

Clock Rule: rule_bufh_gth
Status: PASS 
Rule Description: A BUFH driving a GT must both be in the same clock region
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/ten_gig_eth_pcs_pma_0_local_clock_reset_block/rxoutclk_bufh_i (BUFH.O) is provisionally placed by clockplacer on BUFHCE_X1Y119
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.RXUSRCLK) is locked to GTHE2_CHANNEL_X1Y39

Clock Rule: rule_clk_locked_loads
Status: PASS 
Rule Description NOT AVAILABLE
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/ten_gig_eth_pcs_pma_0_local_clock_reset_block/rxoutclk_bufh_i (BUFH.O) is provisionally placed by clockplacer on BUFHCE_X1Y119
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.RXUSRCLK2) is locked to GTHE2_CHANNEL_X1Y39
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_block_i/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.RXUSRCLK) is locked to GTHE2_CHANNEL_X1Y39

Clock Rule: rule_bufds_bufg
Status: PASS 
Rule Description: A BUFDS driving a BUFG must be placed on the same half side (top/bottom) of the device
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_shared_clock_reset_block/ibufds_inst (IBUFDS_GTE2.O) is locked to IBUFDS_GTE2_X1Y16
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_shared_clock_reset_block/coreclk_bufg_inst (BUFG.I) is provisionally placed by clockplacer on BUFGCTRL_X0Y30

Clock Rule: rule_bufds_gthcommon_intelligent_pin
Status: PASS 
Rule Description: A BUFDS driving a GTHCommon must both be placed in the same or adjacent clock region
(top/bottom)
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_shared_clock_reset_block/ibufds_inst (IBUFDS_GTE2.O) is locked to IBUFDS_GTE2_X1Y16
 sfp_1_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_gt_common_block/gthe2_common_0_i (GTHE2_COMMON.GTREFCLK0) is provisionally placed by clockplacer on GTHE2_COMMON_X1Y9

Clock Rule: rule_gt_bufhce
Status: PASS 
Rule Description: A GT driving a BUFH must both be in the same horizontal row (clockregion-wise)
 sfp_2_pcs_pma_inst/inst/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.RXOUTCLK) is locked to GTHE2_CHANNEL_X1Y38
[... cut ...]
Status: PASS 
Rule Description NOT AVAILABLE
 sfp_4_pcs_pma_inst/inst/ten_gig_eth_pcs_pma_1_local_clock_reset_block/rxoutclk_bufh_i (BUFH.O) is provisionally placed by clockplacer on BUFHCE_X1Y107
 sfp_4_pcs_pma_inst/inst/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.RXUSRCLK) is locked to GTHE2_CHANNEL_X1Y35
 and sfp_4_pcs_pma_inst/inst/gt0_gtwizard_10gbaser_multi_gt_i/gt0_gtwizard_gth_10gbaser_i/gthe2_i (GTHE2_CHANNEL.RXUSRCLK2) is locked to GTHE2_CHANNEL_X1Y35"

alexforencich, 2021-06-15:
Well, that makes no sense.  You're sure you haven't changed any of the XDC files?

Anyway, I went ahead and made a clean clone of the repo and built the SUME design, maybe starting with that would solve some of these issues: https://drive.google.com/file/d/1cNEk1HwUTVeiQziJrntYOpOA_uzwgeBQ/view?usp=sharing

closed 2023-03-09 by bizhaan47