corundum/corundum #64: Is it possible to port the project to 1Gbe?

Quenii, 2021-07-27:
Is it possible to port the project to 1Gbe?

alexforencich, 2021-07-27:
Probably.  Do you have a particular board in mind?

ohault, 2023-08-26:
In follow-up to https://github.com/corundum/corundum/issues/88, this ticket could be renamed to also support 2.5G and 5G (IEEE 802.3bz, NBASE-T and MGBASE-T)

alexforencich, 2023-08-28:
I suppose a related question here: would this be instead of 10G, or in addition to 10G?  I am looking at supporting down-configuration of 10GBASE-R ports to 1000BASE-X and potentially 2.5GBASE-X and 5GBASE-R.  I think supporting SGMII might also be reasonable as SGMII and 1000BASE-X are pretty similar, and USXGMII and 10GBASE-R are also related in a similar way.  All of the BASE-T stuff is dependent on the external PHY.

At any rate I don't have access to any 2.5G or 5G hardware, either FPGA boards with the appropriate PHYs or NICs/switches to test with.