corundum/corundum #67: Driver Error

BearRiding, 2021-08-24:
When I installed the driver, dmesg kept reporting  “mqnic_process_rx_cq ring 161 null page at index 0”.  

[   59.746452] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   59.746493] mqnic 0000:02:00.0: mqnic_process_rx_cq ring 161 null page at index 0
[   59.746494] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   59.746494] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   59.746535] mqnic 0000:02:00.0: mqnic_process_rx_cq ring 193 null page at index 0
[   59.746536] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   59.746536] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

alexforencich, 2021-08-24:
Hmmm, that's very strange.  That is a sanity check that's failing in the driver that checks to see if there is a memory allocation associated with the receive buffer.  I'm not sure offhand what could lead up to that failing.  Are there any other messages printed?  Also, it's strange that the offending completion is all zeros.  At minimum it should have the RXQ number in the completion record, and since this is not on queue 0 then it should not be all zeros.  

Can you provide any more details about your setup - what board, what design, did you make any modifications, did it close timing, what are the specs on the host machine?

BearRiding, 2021-08-24:
[syslog.log](https://github.com/corundum/corundum/files/7037638/syslog.log)
This is my syslog. I didn't make any modification. The board is xc7vx690tffg1761-3.  DDR4 memory is 8GB. The system kernel is 5.4.

alexforencich, 2021-08-24:
So that's the FPGA, on what board?  NetFPGA SUME presumably?  Did you modify the design for that board?  What does vivado report for WNS/TNS?  What OS?  And what CPU? (don't need the whole lscpu or cpuinfo, just the model)

8 GB is not a whole heck of a lot of RAM, but I should be checking all of the allocations to catch if anything fails.  But, I could have missed something, I'll have to see if I can do a low memory test somehow.

BearRiding, 2021-08-24:
Board is NetFPGA SUME. I didn't modify the design. OS is ubuntu 18.04 and kernel is 5.4. CPU is Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz. And where is the WNS/TNS log file? I just follow the README and get the .bit file.

alexforencich, 2021-08-24:
Check fpga_timing_summary_routed.rpt in fpga.runs/impl_1.  There is a section near the top "design timing summary" that lists the WNS and TNS.  If these are negative, then the design may not run correctly on the FPGA.

alexforencich, 2021-08-24:
Oh, one other thing to collect: the output of mqnic-dump (probably need to run `mqnic-dump -d /dev/mqnic0`) after you get this message.  I need to see what the queue pointers are doing.

BearRiding, 2021-08-24:
I'm sorry. It's my fault. The previous board may be damaged. I replace it with a new one and the driver works successfully.

alexforencich, 2021-08-24:
Still, I think it deserves some investigation.  It's possible that the FPGA on the second board has slightly different timing performance vs. the first one, and if the design didn't close timing (negative WNS/TNS) then that would cause the hardware to misbehave.  If you have a chance, it would be great to see the output of mqnic-dump just to see what's going on with the queue pointers in the hardware.  I will also do some testing soon with 8 GB of RAM to see if this is a repeatable issue.

BearRiding, 2021-08-25:
Where could I find the mqnic-dump log file?

alexforencich, 2021-08-25:
when you're getting that error from the driver, run the mqnic-dump utility like so: `mqnic-dump -d /dev/mqnic0 > dump.log`