corundum/corundum #68: Missing ethtool definitions in mqnic module for new kernel 5.11

nightseas, 2021-08-25:
Recently had my server upgraded to Ubuntu 20.04 with new kernel 5.11 and failed to compile the mqnic kernel module.
```bash
$ uname -a
Linux T630-Test-Server 5.11.0-27-generic #29~20.04.1-Ubuntu SMP Wed Aug 11 15:58:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

make[1]: Entering directory '/usr/src/linux-headers-5.11.0-27-generic'
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_main.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_dev.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_netdev.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_port.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_ptp.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_i2c.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_board.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_tx.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_rx.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_cq.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_eq.o
  CC [M]  /home/ubuntu/corundum/modules/mqnic/mqnic_ethtool.o

/home/ubuntu/corundum/modules/mqnic/mqnic_ethtool.c:41:63: warning: ‘struct ethtool_drvinfo’ declared inside parameter list will not be visible outside of this definition or declaration
   41 | static void mqnic_get_drvinfo(struct net_device *ndev, struct ethtool_drvinfo *drvinfo)
      |                                                               ^~~~~~~~~~~~~~~
/home/ubuntu/corundum/modules/mqnic/mqnic_ethtool.c: In function ‘mqnic_get_drvinfo’:
/home/ubuntu/corundum/modules/mqnic/mqnic_ethtool.c:46:20: error: dereferencing pointer to incomplete type ‘struct ethtool_drvinfo’
   46 |     strlcpy(drvinfo->driver, DRIVER_NAME, sizeof(drvinfo->driver));
      |                    ^~
/home/ubuntu/corundum/modules/mqnic/mqnic_ethtool.c: At top level:
/home/ubuntu/corundum/modules/mqnic/mqnic_ethtool.c:53:62: warning: ‘struct ethtool_ts_info’ declared inside parameter list will not be visible outside of this definition or declaration
   53 | static int mqnic_get_ts_info(struct net_device *ndev, struct ethtool_ts_info *info)
      |                                                              ^~~~~~~~~~~~~~~
/home/ubuntu/corundum/modules/mqnic/mqnic_ethtool.c: In function ‘mqnic_get_ts_info’:
/home/ubuntu/corundum/modules/mqnic/mqnic_ethtool.c:59:11: error: implicit declaration of function ‘ethtool_op_get_ts_info’ [-Werror=implicit-function-declaration]
   59 |     ret = ethtool_op_get_ts_info(ndev, info);
      |           ^~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/corundum/modules/mqnic/mqnic_ethtool.c:63:9: error: dereferencing pointer to incomplete type ‘struct ethtool_ts_info’
   63 |     info->so_timestamping |=

```

Looks like header missing of ethtool definitions due to kernel updates.
Adding `ethtool.h` will solve the issue.

```c
index 78770859..a784d0eb 100644
--- a/modules/mqnic/mqnic.h
+++ b/modules/mqnic/mqnic.h
@@ -39,6 +39,7 @@ either expressed or implied, of The Regents of the University of California.
 #include <linux/miscdevice.h>
 #include <linux/netdevice.h>
 #include <linux/etherdevice.h>
+#include <linux/ethtool.h>
 #include <linux/net_tstamp.h>
 #include <linux/ptp_clock_kernel.h>
```

alexforencich, 2021-08-25:
I actually just discovered that, and there was a second one that I was also missing:

https://github.com/alexforencich/corundum/commit/8bf38e20c78cfbee5d0b9b8349fb2a7b3eab4dbb

That change will make it upstream relatively soon.

alexforencich, 2021-09-14:
This should be fixed in the current version of the driver.

closed 2021-09-14 by alexforencich