corundum/corundum #69: Does queue_manager ignore the back pressure on s_axil_bready?

jeehoonkang, 2021-09-05:
I'm reading Corundum's queue_manager, and I'm wondering whether it correctly considers the back pressure on s_axil_bready.

The particular line I'm concerned about is [L390](https://github.com/corundum/corundum/blob/master/fpga/common/rtl/queue_manager.v#L390). According to the condition, a new AXIL write work item (say A) can be inserted into the pipeline even if the pipeline's item (say B) at the index 1 (= PIPELINE-1) is also an AXIL write. Concretely, suppose the current cycle is t. Then:

- At t+1, B is placed on s_axil_bvalid but maybe s_axil_bready is not asserted for a while; and A is placed on the pipeline at the index 0.
- At t+2, A is placed on the pipeline at the index 1 = PIPELINE-1.
- At t+3, B is incorrectly overwritten by A for the s_axil_bvalid channel.

May I ask if my understanding is correct and queue_manager may indeed incorrectly overwrite a value on the AXIL channel? If I'm wrong, (1) how to consider the back pressure of s_axil_bready? or (2) is it the case that the overwriting is okay?

Just FYI, it seems the same concern exists for AXIL read ([L403](https://github.com/corundum/corundum/blob/master/fpga/common/rtl/queue_manager.v#L403)), but dequeue request ([L422](https://github.com/corundum/corundum/blob/master/fpga/common/rtl/queue_manager.v#L422)) correctly considers its output's back pressure.

Thank you for reading my not-so-short question! After understanding queue_manager's behavior, I'll make a PR that adds documentation comments to the code that explains the behavior so that the other people will not be confused with this.

alexforencich, 2021-09-05:
Both modules should only allow one AXI lite operation in the pipeline at any time exactly due to this backpressure issue.  However, it's possible there is a minor issue somewhere, I will take a look.  

At any rate, don't bother with comments for these modules, they're both going to be replaced relatively soon as part of the new descriptor handling logic for variable-length descriptors.  The design of the current queue manager modules is left over from when the descriptor fetch was implemented in the per-port datapath.  That's not the case anymore, and for variable-length descriptors the queue pointers need to be managed more directly by the descriptor management logic, so these modules are either going to be replaced or significantly rewritten and the whole pipeline structure is completely going away.

alexforencich, 2021-09-05:
Yeah, it looks like both AXI lite implementations are wrong.  Previously, I used the AXI lite interconnect module for the AXI lite operations, and this only allowed a single AXI lite operation to be active at any given time.  I since switched over to a different module (AXI lite crossbar) which does not have this restriction, so I should also make sure that the queue manager modules only allow a single operation in the pipeline since the pipeline itself doesn't support backpressure (which was an intentional design decision).  However, it looks like the enqueue/dequeue request implementation is correct as it checks to make sure there is no other operation in the pipeline before starting a new operation.  This is not really ideal from a throughput perspective....

Anyway, like I said all of this is going away sometime in the near future, to be replaced by something much more powerful and flexible.  So I will fix potential hanging bugs, but I don't want to waste time on any other enhancements.

jeehoonkang, 2021-09-05:
@alexforencich Thank you for your detailed answer. I've heard that the queue manager module is going to be replaced soon, but I'd like to suggest a possible simple fix as a transient measure. I'd like to ask if you could have a look at the following...

I think may it's okay to set PIPELINE = 1? The assertion at L177 says PIPELINE must be >= 2, but it doesn't look like this minimal depth requirement is required/used in any form. Furthermore, (1) it will fix the backpressure problem; and (2) it would effectively improve throughput by reducing stalls.

alexforencich, 2021-09-06:
No, minimal pipeline depth for using URAMs is at least 3, and it needs to be bigger for more queues.  I think 8192 queues (the default for most targets) requires pipeline set to 4 to close timing.  Possibly even higher.  The whole point of having the pipeline is because the RAM blocks on the FPGA require it when assembling large arrays, and the clock-to-output delays of BRAM and URAM are bad and terrible, respectively.  I would have to check on Fmax, I don't think it's an issue at 250 MHz but the Fmax does fall if you're not using all of the pipeline registers.

jeehoonkang, 2021-09-30:
Thank you for your detailed answer. I see the issue is resolved in the recent master branch.

I'd like to ask a very basic question about your last comment. (I'm asking here because I alone couldn't find out the answer from the Internet...) Now I see the necessity of pipelines for URAM at the high level, but I don't see how it is addressed in Corundum. For instance, it seems URAM is used for `queue_ram` in `queue_manager.v`, and reading from `queue_ram` is pipelined as follows: https://github.com/corundum/corundum/blob/master/fpga/common/rtl/queue_manager.v#L677-L681 But if reading from `queue_ram` at L677 has bad timing, isn't it the case that `queue_ram_read_data_reg` and subsequent pipeline registers will be assigned bad values at the next cycles? How do they help in addressing the bad timing of `queue_ram`?

(One more minor thing: can we "rename" `queue_ram_read_data_reg` into `queue_ram_read_data_pipeline_reg[0]`? It seems more consistent with the naming convention of the other pipeline registers.)

alexforencich, 2021-09-30:
Take a look at UG573, figures 2-3 and 2-5.  For 8192 entries, two URAM blocks are required in a cascade chain.  The address is provided to the address input.  In this case, I don't think I am using the IREG_PRE/IREG_CAS registers; it goes straight to the memory array.  The memory array is synchronous, so the read takes place into queue_ram_read_data_reg with a one-cycle delay (total delay so far: 1 cycle).  Then, the data is stored for one cycle in OREG (total delay so far: 2 cycles).  Then it bypasses OREG_ECC, leaves the first URAM via CAS_OUT_DOUT, enters the second URAM via CAS_IN_DOUT, and is stored in OREG_CAS (total delay so far: 3 cycles).  Then, the data is available to the fabric logic.  But, the Tcko is rather high, so you can't hang much logic directly on the output of the URAM, so I run it through a fabric register (total delay so far: 4 cycles) before it's used for anything.  

I think that register cannot be renamed, as then Vivado will not infer the URAM correctly.  It can be a bit picky when it comes to inferring RAMs.  Looking at how that code is written, I think I may have tried what you're recommending at one point, and Vivado was not happy, so I had to add `queue_ram_read_data_reg` and use that instead of `queue_ram_read_data_pipeline_reg[0]`.

alexforencich, 2021-10-01:
Also, if you take a look at the implemented design for any of the US+ designs, you'll see that only the `queue_ram_read_data_pipeline_reg[3]` instances appear in the leaf cells list for the queue management modules.  This is because the other stages are inside of the URAM blocks.  If you look at the URAM blocks, you can see that they all have `IREG_PRE_A/B = FALSE` and `OREG_A/B = TRUE`.  And if you look at the cell pins on the URAM blocks, you can see how the cascade chain connections are set up, with pairs of URAM blocks forming a cascade chain - on one, the `ADDR_A/B`, `DIN_B`, etc pins are connected along with `CAS_OUT`, then on the next `CAS_IN`, `DOUT_A`, etc. are connected.

jeehoonkang, 2021-11-19:
Alex, thank you so much for your detailed explanations. Honestly, I didn't understand anything you've said last month, but now I can see (at least) your point about timing and URAM clearly. Your replies help me much more than Xilinx UG!

Before closing this issue, I wonder if you could have a look at `tx_scheduler_rr`. I think it has the same (potential) bug as `queue_manger` and `cpl_queue_manager`, at https://github.com/corundum/corundum/blob/master/fpga/common/rtl/tx_scheduler_rr.v#L514 and https://github.com/corundum/corundum/blob/master/fpga/common/rtl/tx_scheduler_rr.v#L526 . I wonder what do you think about it.

alexforencich, 2021-11-20:
Ah yep, that's also a potential problem.  I'll take a look at it.  The AXI lite interface on the scheduler is not really intended to be poked at very often, so I think making the same fix here would be reasonable.