corundum/corundum #70: `AU200/fpga_100g` cocotb test is broken

lomotos10, 2021-09-14:
`pytest fpga/mqnic/AU200/fpga_100g` results in the following error message:

```
INFO     cocotb:simulator.py:267      0.00ns ERROR    cocotb.regression                         regression.py:399  in _score_test                     Test Failed: run_test_nic (result was TypeError)
INFO     cocotb:simulator.py:267                                                                                                                      Traceback (most recent call last):
INFO     cocotb:simulator.py:267                                                                                                                        File "/kaist-cp-home/sungsoo.han/WD/play-corundum/corundum/fpga/mqnic/AU200/fpga_100g/tb/fpga_core/test_fpga_core.py", line 368, in run_test_nic
INFO     cocotb:simulator.py:267                                                                                                                          tb = TB(dut)
INFO     cocotb:simulator.py:267                                                                                                                        File "/kaist-cp-home/sungsoo.han/WD/play-corundum/corundum/fpga/mqnic/AU200/fpga_100g/tb/fpga_core/test_fpga_core.py", line 275, in __init__
INFO     cocotb:simulator.py:267                                                                                                                          self.qsfp0_mac = EthMac(
INFO     cocotb:simulator.py:267                                                                                                                        File "/kaist-cp-home/sungsoo.han/.local/lib/python3.8/site-packages/cocotbext/eth/eth_mac.py", line 539, in __init__
INFO     cocotb:simulator.py:267                                                                                                                          super().__init__(*args, **kwargs)
INFO     cocotb:simulator.py:267                                                                                                                      TypeError: object.__init__() takes exactly one argument (the instance to initialize)
INFO     cocotb:simulator.py:267      0.00ns ERROR    cocotb.regression                         regression.py:484  in _log_test_summary               Failed 1 out of 1 tests (0 skipped)
INFO     cocotb:simulator.py:267      0.00ns INFO     cocotb.regression                         regression.py:557  in _log_test_summary               *************************************************************************************
INFO     cocotb:simulator.py:267                                                                                                                      ** TEST                         PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
INFO     cocotb:simulator.py:267                                                                                                                      *************************************************************************************
INFO     cocotb:simulator.py:267                                                                                                                      ** test_fpga_core.run_test_nic    FAIL            0.00          0.04         0.03  **
INFO     cocotb:simulator.py:267                                                                                                                      *************************************************************************************
INFO     cocotb:simulator.py:267 
INFO     cocotb:simulator.py:267      0.00ns INFO     cocotb.regression                         regression.py:574  in _log_sim_summary                *************************************************************************************
INFO     cocotb:simulator.py:267                                                                                                                      **                                 ERRORS : 1                                      **
INFO     cocotb:simulator.py:267                                                                                                                      *************************************************************************************
INFO     cocotb:simulator.py:267                                                                                                                      **                               SIM TIME : 0.00 NS                                **
INFO     cocotb:simulator.py:267                                                                                                                      **                              REAL TIME : 0.43 S                                 **
INFO     cocotb:simulator.py:267                                                                                                                      **                        SIM / REAL TIME : 0.00 NS/S                              **
INFO     cocotb:simulator.py:267                                                                                                                      *************************************************************************************
INFO     cocotb:simulator.py:267 
INFO     cocotb:simulator.py:267      0.00ns INFO     cocotb.regression                         regression.py:259  in tear_down                       Shutting down...
INFO     cocotb:simulator.py:267 Addressing configuration for axil_crossbar_addr instance fpga_core.core_inst.core_pcie_inst.core_inst.iface[0].interface_inst.axil_crossbar_inst.axil_crossbar_wr_inst.s_ifaces[0].addr_inst
INFO     cocotb:simulator.py:267  0 ( 0): 000000 / 18 -- 000000-03ffff
INFO     cocotb:simulator.py:267  1 ( 0): 040000 / 18 -- 040000-07ffff
INFO     cocotb:simulator.py:267  2 ( 0): 080000 / 19 -- 080000-0fffff
INFO     cocotb:simulator.py:267  3 ( 0): 100000 / 20 -- 100000-1fffff
INFO     cocotb:simulator.py:267  4 ( 0): 200000 / 20 -- 200000-2fffff
INFO     cocotb:simulator.py:267  5 ( 0): 300000 / 19 -- 300000-37ffff
INFO     cocotb:simulator.py:267  6 ( 0): 380000 / 19 -- 380000-3fffff
INFO     cocotb:simulator.py:267  7 ( 0): 400000 / 20 -- 400000-4fffff
INFO     cocotb:simulator.py:267 Addressing configuration for axil_crossbar_addr instance fpga_core.core_inst.core_pcie_inst.core_inst.iface[0].interface_inst.axil_crossbar_inst.axil_crossbar_rd_inst.s_ifaces[0].addr_inst
INFO     cocotb:simulator.py:267  0 ( 0): 000000 / 18 -- 000000-03ffff
INFO     cocotb:simulator.py:267  1 ( 0): 040000 / 18 -- 040000-07ffff
INFO     cocotb:simulator.py:267  2 ( 0): 080000 / 19 -- 080000-0fffff
INFO     cocotb:simulator.py:267  3 ( 0): 100000 / 20 -- 100000-1fffff
INFO     cocotb:simulator.py:267  4 ( 0): 200000 / 20 -- 200000-2fffff
INFO     cocotb:simulator.py:267  5 ( 0): 300000 / 19 -- 300000-37ffff
INFO     cocotb:simulator.py:267  6 ( 0): 380000 / 19 -- 380000-3fffff
INFO     cocotb:simulator.py:267  7 ( 0): 400000 / 20 -- 400000-4fffff
INFO     cocotb:simulator.py:267 Addressing configuration for axil_crossbar_addr instance fpga_core.core_inst.core_pcie_inst.core_inst.iface[1].interface_inst.axil_crossbar_inst.axil_crossbar_wr_inst.s_ifaces[0].addr_inst
INFO     cocotb:simulator.py:267  0 ( 0): 000000 / 18 -- 000000-03ffff
INFO     cocotb:simulator.py:267  1 ( 0): 040000 / 18 -- 040000-07ffff
INFO     cocotb:simulator.py:267  2 ( 0): 080000 / 19 -- 080000-0fffff
INFO     cocotb:simulator.py:267  3 ( 0): 100000 / 20 -- 100000-1fffff
INFO     cocotb:simulator.py:267  4 ( 0): 200000 / 20 -- 200000-2fffff
INFO     cocotb:simulator.py:267  5 ( 0): 300000 / 19 -- 300000-37ffff
INFO     cocotb:simulator.py:267  6 ( 0): 380000 / 19 -- 380000-3fffff
INFO     cocotb:simulator.py:267  7 ( 0): 400000 / 20 -- 400000-4fffff
INFO     cocotb:simulator.py:267 Addressing configuration for axil_crossbar_addr instance fpga_core.core_inst.core_pcie_inst.core_inst.iface[1].interface_inst.axil_crossbar_inst.axil_crossbar_rd_inst.s_ifaces[0].addr_inst
INFO     cocotb:simulator.py:267  0 ( 0): 000000 / 18 -- 000000-03ffff
INFO     cocotb:simulator.py:267  1 ( 0): 040000 / 18 -- 040000-07ffff
INFO     cocotb:simulator.py:267  2 ( 0): 080000 / 19 -- 080000-0fffff
INFO     cocotb:simulator.py:267  3 ( 0): 100000 / 20 -- 100000-1fffff
INFO     cocotb:simulator.py:267  4 ( 0): 200000 / 20 -- 200000-2fffff
INFO     cocotb:simulator.py:267  5 ( 0): 300000 / 19 -- 300000-37ffff
INFO     cocotb:simulator.py:267  6 ( 0): 380000 / 19 -- 380000-3fffff
[... cut ...]
INFO     cocotb:simulator.py:267  1 ( 0): 10000 / 16 -- 10000-1ffff
=========================== short test summary info ============================
FAILED fpga/mqnic/AU200/fpga_100g/tb/fpga_core/test_fpga_core.py::test_fpga_core
============================== 1 failed in 1.27s ===============================
```

Additionally, this error was reproduced in `AU250/fpga_100g`, but no errors were present in `AU200/fpga_10g` nor `AU250/fpga_10g`

It seems that `fpga_100g` test is broken while `fpga_10g` is fine, but this might need some more experimentation.

alexforencich, 2021-09-14:
Update cocotbext-eth.

minseongg, 2021-09-14:
I solved this problem by `pip3 install cocotbext-eth --upgrade` .

closed 2021-09-14 by lomotos10