corundum/corundum #73: Wrong BAR registers

noborienten, 2021-11-01:
Corundum was installed on ADM-PCIE-9V3, and mqnic drivers were installed.
And when I printed out the dmesg, the following results came out. I added some output codes. 
As can be seen, the values of register values including FW ID and Board ID are incorrect.
[image]

I don't know what the problem is.
Can you help me solve the problem?

alexforencich, 2021-11-01:
Well, that's very odd.  You're using git master from here with no modifications, correct?  Can you try with my development fork here: https://github.com/alexforencich/corundum/ ?

Does it make any difference what order you read the registers?  Do you get the same value if you read the same register several times?

alexforencich, 2021-11-01:
Also, what version of Vivado are you using, which variant of the ADM-PCIE-9V3 did you build, and what did Vivado report for WNS and TNS?

noborienten, 2021-11-01:
I didn't make any modifications.
The register value has not changed, whether it is reversed or read several times.
Vivado version is 2021.02, and ADM-PCIE-9V3 VU3P-2 is used (I'm not sure if the answer you wanted).
And Vivado reported WNS as 0.044 and TNS as 0.000.

Also, I tried with https://github.com/alexforencich/corundum/ , but i got the same result.

noborienten, 2021-11-01:
I am sorry. It is my fault
I don't install corundum on fpga 😅

closed 2021-11-01 by noborienten

alexforencich, 2021-11-01:
Hmmm. so if the design was not loaded on the FPGA, why was the driver trying to attach to the FPGA in the first place?

noborienten, 2021-11-02:
Actually, I also add device id of ADM-PCIE-9V3 (4144:0901) to the mqnic_pci_id_table ...
I think it is the reason.

alexforencich, 2021-11-02:
Ah yep, that would do it.  Although the driver not automatically attaching to the card is definitely a sign that something wasn't set up correctly.  Anyway, good thing you figured that one out!

noborienten, 2021-11-02:
Thank you for your answer.
Your open source and YouTube videos are very helpful to me. Thank you for this as well.