corundum/corundum #74: How could I make Alveo U50 detective by Vivado HARDWARE MANAGER？

IskandarZhang, 2021-11-09:
In order to download fpga.bit to Alveo board, I opened fpga.prj and start HARDWARE MANAGER, but I found no cards on localhost as the figure below shows:
[image]
After using tcl command current_hw_target, WARNING: [Labtools 27-1339] No hardware targets exist on the server [Check to make sure the cable targets connected to this machine are properly connected and powered up, then use the disconnect_hw_server and connect_hw_server commands to re-register the hardware targets.] But I suggest that the Alveo U50 boards are linked to my pc correctly.
Best.

alexforencich, 2021-11-09:
For the U50, you need a separate USB JTAG programming cable, part number HW-DMB-1-G. This connects to the utility connector on the board. Do you have one of these cables wired up, or is the card only connected via PCIe?

IskandarZhang, 2021-11-10:
> For the U50, you need a separate USB JTAG programming cable, part number HW-DMB-1-G. This connects to the utility connector on the board. Do you have one of these cables wired up, or is the card only connected via PCIe?

Thanks for your prompt reply! I realized that I didn't have this cable linked and I have submitted the purchase request of this kind of cable to my university, but the purchase cycle is so long that I cannot complete the developing goal on time. May I ask is it feasible to modify the Makefile to make the generated file format .xclbin instead of .bit? If possible, I would like to ask further for some details of the modification. Sincere regards.

alexforencich, 2021-11-10:
Corundum is not compatible with the Alveo framework, unfortunately.  It must run directly on the card.  However, it may be possible to talk to the FPGA via JTAG from a different JTAG adapter.  I have the connector pinout mostly sorted out, but the one time I tried it, Vivado couldn't talk to the FPGA and there wasn't a clear reason why.  I suspect maybe the BMC microcontroller has something to do with it, and I may need to pull a different pin high or low to tell the BMC to release the JTAG interface.  Do you have any other USB JTAG cables, such as a DLC9 or DLC10?

IskandarZhang, 2021-11-10:
> Corundum is not compatible with the Alveo framework, unfortunately. It must run directly on the card, but once you get it on the card then you can update it over PCIe with mqnic-fw. However, it may be possible to talk to the FPGA via JTAG from a different JTAG adapter. I have the connector pinout mostly sorted out, but the one time I tried it, Vivado couldn't talk to the FPGA and there wasn't a clear reason why. I suspect maybe the BMC microcontroller has something to do with it, and I may need to pull a different pin high or low to tell the BMC to release the JTAG interface. Do you have any other USB JTAG cables, such as a DLC9 or DLC10?

In the last few hours I searched for the JTAG cable you mentioned before, but unfortunately I didn't find any available. I had some ideas before, about how to modify the makefile to generate .xclbin file but now I also think that this is not working. May I ask you what I can do to help future development based on Corundum while waiting for the JTAG cable to be purchased? 
Best.

alexforencich, 2021-11-10:
Well, you can always poke at it in sim.  But I may be able to provide remote access to a card in one of the testbeds at UCSD.  Are you on the Corundum slack workspace?

IskandarZhang, 2021-11-10:
> Well, you can always poke at it in sim. But I may be able to provide remote access to a card in one of the testbeds at UCSD. Are you on the Corundum slack workspace?

It's very kind of you, but I don't think I'm on the slack workspace. :(

alexforencich, 2021-11-10:
The link to join is right at the top of the readme

IskandarZhang, 2021-11-10:
> The link to join is right at the top of the readme

Thanks for guiding me and I'm already in it with the usr name zhangyuxiang. What could I do next?

alexforencich, 2021-11-11:
For potentially setting up remote access, I will follow up on slack or via email.