corundum/corundum #75: A switch fabric inside Corundum

Winters123, 2021-11-14:
I'm thinking about adding a switch fabric inside Corundum. The basic purpose is to support hardware-accelerated virtual switch. Specifically, we could have an arbiter in the front that merges all input axis data (dma, mac...) into one and feeds it into a single pipeline, and de-mux the axi stream to different directions (dma, mac...) at the end of the pipeline. I suppose a switch fabric like this is beneficial to lots of applications like NIC-accel OVS and such. 

I guess the easist way to support such a fabric is to move the `dma_client_sink/source.v` out of `port.v`, connects the rx/tx from DMA and the rx/tx from MAC into an arbiter, and de-mux them into different channels at the end of the pipeline. The interconnection should be added in `mqnic_core_pcie_us.v`. 

The challenge here I guess should be the maximum performance of the pipeline. Theoritically, the pipeline is more likely to become the bottleneck as the number of ports grows. Yet, there should be some spaces that we can play with to increase the performance. 

I would love to hear your suggestions or thoughts on how to build such a mechanism inside Corundum.

alexforencich, 2021-11-14:
It's an interesting idea, and I am going to be moving a step in that direction with a shared interface datapath where basically all of the datapath components move from the port module into the interface module, with shared-memory FIFOs used to interface with the MACs.  However, major concerns are head of line blocking and throughput, so I am going to keep the RX and TX paths separate and implement credit-based flow control on the TX path.  

If you merge both the RX and TX path together, then I think some more careful thought would be required to deal with these issues.  Simply widening the pipeline is not a complete solution, because you can still get stuff backing up from downstream FIFOs, creating issues with head-of-line blocking.  And also beyond 512 bits (and even at 512 bits to some extent) you start running in to issues with packet rate, since a 512 bit interface moves a complete minimum length frame every clock cycle.  The 100G designs already use a 512 bit interface, so even if you're merging the RX and TX for a single port, you might want to move to a segmented interface that can move multiple packets per clock cycle.  And this problem only becomes worse with more ports.

Winters123, 2021-11-14:
I think moving the datapath components to `interface.v` would be a nice starting point to support switch-like applications. I guess it makes more sense to see all the RX datapaths merge at the beginning of the pipeline (i.e. the switch fabric). We could add a de-mux module at the end of it to redirect packets to different TX paths. By then Corundum will be able to support packets being redirected to all destinations (physical ports, OS).

Totally agree with the potential head-of-line blocking issue, which is why I would love to hear your suggestions on this :). The idea of a segmented interface sounds cool but I'm not sure about how it can be beneficial to performance. Could you explain that a bit more, please? 

If you are moving to that direction, are there any means that we could contribute to it (code, doc, discussion) ?

alexforencich, 2021-11-15:
So, I am keeping the RX and TX paths separate to alleviate some head of line blocking issues.  On RX, it's not really a problem as there is no major bandwidth bottleneck - everything gets with-converted, FIFOed, merged, and then sent to the RX datapath.  If the RX datapath can't keep up, well, stuff gets dropped at the RX MACs.  On TX, the problem is that each port is going to be limited to line rate, and the datapath has to be at least N*line rate.  So if you send more than the TX FIFO size, the TX path gets all jammed up.  So, what I'm going to do is limit the amount of data that can exist in the TX path at any one time, and the schedulers cannot issue TX requests if there isn't space available in the TX FIFOs.  Possibly you can use the same strategy if you're also merging in data from the RX path, and if things get jammed up then you'll just drop frames on the RX side.  But it's certainly a lot easier to exert backpressure against scheduler transmit requests than it is against an RX FIFO.  

The other aspect of this new setup is it will be able to support traffic classes and form a path to implement priority flow control.  

The reason you need a segmented interface is because with the current interface, all packets start in lane 0.  Let's say you have a 512 bit interface that can transfer 64 bytes in each clock cycle and you're running at 250 MHz.  The stat of each packet must fall in lane 0, so if your packets are 64 bytes or less, you can transfer 250 Mpps.  But if you packets are 65 bytes, that one extra byte must be sent in a transfer cycle all by itself, so your packet rate falls to 125 Mpps.  However, if you have a segmented interface, then you can start a packet, say, in lane 32.  In this case, I think you end up being able to send 2 packets every 3 cycles, or 167 Mpps.  And this issue gets worse with wider interfaces.  For instance, it doesn't matter how wide your interface is, you will never be able to send more than one packet per clock cycle if it's not segmented.  So if you implement some crazy wide interface, say 2048 bits, you'll still only be able to send 250 Mpps.  And what's even worse, if you send 64 byte packets, you can only utilize 1/4 of the bus width, so the data rate will only be 128 Gbps instead of 512 Gbps.  And this also isn't just an issue for 64 byte packets, but really any packet size that is slightly larger than an even multiple of the bus width, since it will leave a lot of empty byte lanes.  

Now, for Corundum, TBH I don't particularly care about raw packet rate.  It's impossible even to hit the max packet rate at 100 Gbps over PCIe gen 3 x16, so I can cut corners with the internal interfaces and not have any major issues since the bottleneck is PCIe.  But for switching applications, the calculus is completely different since most traffic may never cross the PCIe bus.

Winters123, 2021-11-15:
Thanks for the detailed explanation. That's the reason you implement `dma_ram_demux` on the DMA path, right? 

I'm not worried about the performance bottleneck for port2port switching applications as Corundum is still a NIC after all. But with PCIe 3x16 and the multi-queue DMA engine, I would expect Corundum to be able to work at high-rate for virtual switch applications (with DPDK implemented). This could attract a bunch of guys who are interested in virtual switch acceleration and such.

btw, is there already a timetable for adding such a mechanism? Is there a way to get involved in the development process (discussion, code...)?

alexforencich, 2021-11-15:
For `dma_ram_demux`, that's just so that the host-facing DMA interface module can be shared across multiple internal components.  But yes, that is a segmented interface, and I made that interface segmented for similar reasons.  From the standpoint of the DMA engine, it's about proving full throughput for unaligned operations as it supports accesses to adjacent memory locations on the same cycle, which would otherwise have to take place on separate clock cycles without the segmented interface.

Let's talk about it at the developer meeting tomorrow.

nightseas, 2021-12-02:
Just personal view, will this be a more flexible and general way to implement switching logic in NIC? I've seen several commercial or special purpose designs similar to this structure.

[image]

alexforencich, 2021-12-02:
General and flexible, sure.  But it will consume a lot of resources and potentially introduce flow control problems.  I'm definitely open to considering something like that as an option that can be enabled via verilog parameter and perhaps configured via switchdev or something, and disabled when not needed to save on logic resources.