corundum/corundum #76: errors by using pytest when test corundum

su-33, 2021-11-18:
I'm using pytest to run the simulation of corundum, and there are issues occured. The information is:

---
        [root@localhost fpga_core]# pwd
        ~/corundum-master/fpga/mqnic/AU200/fpga_10g/tb/fpga_core
        [root@localhost fpga_core]# python3 -m pytest -n auto
        Test session starts (platform: linux, Python 3.9.2, pytest 6.2.5, pytest-sugar 0.9.4)
        rootdir: ~/corundum-master, configfile: tox.ini
        plugins: forked-1.3.0, xdist-2.4.0, sugar-0.9.4, cocotb-test-0.2.1
        
        ......
        
        test_fpga_core.py:753: 
        _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
        /usr/local/python3/lib/python3.9/site-packages/cocotb_test/simulator.py:941: in run
            return sim.run()
        /usr/local/python3/lib/python3.9/site-packages/cocotb_test/simulator.py:217: in run
            self.execute(cmds)
        /usr/local/python3/lib/python3.9/site-packages/cocotb_test/simulator.py:258: in execute
            with subprocess.Popen(
        /usr/local/python3/lib/python3.9/subprocess.py:951: in __init__
            self._execute_child(args, executable, preexec_fn, close_fds,
        /usr/local/python3/lib/python3.9/subprocess.py:1743: in _execute_child
            env_list.append(k + b'=' + os.fsencode(v))
        _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
        filename = None
        
            def fsencode(filename):
                """Encode filename (an os.PathLike, bytes, or str) to the filesystem
                encoding with 'surrogateescape' error handler, return bytes unchanged.
                On Windows, use 'strict' error handler if the file system encoding is
                'mbcs' (which is the default encoding).
                """
        >       filename = fspath(filename)  # Does type-checking of `filename`.
        E       TypeError: expected str, bytes or os.PathLike object, not NoneType
        
        /usr/local/python3/lib/python3.9/os.py:810: TypeError
---
It seems that there is something wrong with the variate of the file path. But I was completely following the steps in [getting started](https://github.com/corundum/corundum/wiki/Getting-Started). What can I do to run the simulation successfully?

su-33, 2021-11-18:
Similarly, running tox in /path/to/corundum/ will get 54 errors, each of which is as same as the issue above.

alexforencich, 2021-11-18:
That's strange, seems like cocotb-test is having trouble running something.  Do you have Icarus Verilog installed?  Also, why are you running as root?

su-33, 2021-11-18:
Thanks for replying.
Yes, the iverilog had been installed successfully on my server:

---
    [root@localhost ~]# iverilog -V
    Icarus Verilog version 10.2 (stable) ()
---

As for running as root, I tried the same option by using another general user account, and the issue happened, too. Should I uninstall all the dependencing packages and reinstall them through another general user account?

czrsirud, 2023-08-01:
Hi, I have the same problem, will this be blamed on Centos?

alexforencich, 2023-08-01:
Centos/RHEL only affects the device driver, it should not affect the tests.

I'm wondering if the problem may be related to symbolic links.  The directory "corundum-master" leads me to believe that this is NOT a clone of the repo, instead a zip file was downloaded from github and extracted.  Since zip files do not support symbolic links, all of the links would have to be recreated for all of the tests and build scripts to work.  You would probably run in to a similar issue if you use WSL and have the files on a Windows filesystem (which also does not support symbolic links).

czrsirud, 2023-08-02:
I tried your suggestion, but I still get the same error, and I get the same error message when I try to use $tox in the Corundum root directory.

czrsirud, 2023-08-02:
> Centos/RHEL only affects the device driver, it should not affect the tests.
> 
> I'm wondering if the problem may be related to symbolic links. The directory "corundum-master" leads me to believe that this is NOT a clone of the repo, instead an archive of some sort was downloaded from github and extracted. If the archive was not extracted correctly, then the symbolic links may be broken. I suspect that maybe WSL was involved here at some point - if the archive is extracted on a Windows filesystem which does not support symbolic links, then all of the links will be broken. If you're using WSL, you'll have to extract the archive within WSL on a linux-native filesystem where the symlinks will work.
> 
> So, I recommend trying a clean clone of the repository (`git clone https://github.com/corundum/corundum.git`) on a Linux native filesystem. And if you must download an archive for whatever reason, I recommend using a .tar.gz or similar Linux native archive instead of zip as I think it is more likely for the symlinks to be extracted correctly.

I tried your suggestion, but I still get the same error, and I get the same error message when I try to use $tox in the Corundum root directory.

canchen0720, 2023-11-17:
Hi,
Have you solved the problem? I have met the same one but I can't solve it.

canchen0720, 2023-11-17:
> Hi, I have the same problem, will this be blamed on Centos?

Did you solve the problem? I met the same problem too.

constant007, 2024-12-02:
sane issue， any one can help？

canchen0720, 2024-12-02:
> sane issue， any one can help？
I met this problem when I chose Ubuntu16.04. Now I have changed to Ubuntu20.04，there is no problems.