corundum/corundum #82: Enabling app will break axis_async_fifo.tcl contraints

likewise, 2021-12-13:
For the case where the (default template) app block is enabled, timing closure fails. Some debugging shows the "get_pins" result is empty for "set write_clk [get_clocks -of_objects [get_pins $fifo_inst/wr_ptr_reg_reg[0]/C]]" in when "axis_async_fifo.tcl" runs.

This might have to do with a recent rename in axis_async_fifo.v in the pointer synchronization section, although I have difficulty understanding the naming convention used in the registers passing the clock domains.

The above gives an error in Vivado, but the build proceeds, and the timing closure is not met.

Reproduce for Xilinx U50. 

fpga/mqnic/AU50/fpga_100g/fpga/Makefile
+# Add template application block
+SYN_FILES += app/template/rtl/mqnic_app_block.v
+SYN_FILES += lib/axi/rtl/axil_ram.v

fpga/mqnic/AU50/fpga_100g/fpga/config.tcl
 # Application block configuration
-dict set params APP_ENABLE "0"
+dict set params APP_ENABLE "1"

cd fpga/mqnic/AU50/fpga_100g/fpga/
make

likewise, 2021-12-13:
More specifically it is the RX async FIFO that fails to get pins:
get_pins core_inst/core_inst/core_pcie_inst/core_inst/iface[0].port[0].rx_async_fifo_inst/fifo_inst/wr_ptr_reg_reg[0]/C

Adding an asterix gives a pin:

get_pins core_inst/core_inst/core_pcie_inst/core_inst/iface[0].port[0].rx_async_fifo_inst/fifo_inst/wr_ptr*reg_reg[0]/C
core_inst/core_inst/core_pcie_inst/core_inst/iface[0].port[0].rx_async_fifo_inst/fifo_inst/wr_ptr_gray_sync1_reg_reg[0]/C

alexforencich, 2021-12-13:
That's strange, I'll definitely take a look at that in the next day or two.  Usually what gets the timing constraint TCL scripts screwed up is when vivado removes/merges things that the script is looking for.  But for extracting the clocks, I thought I picked stuff that wouldn't/couldn't get merged so that it could extract that reliably.  Could be some sort of a connection issue, perhaps.  The CI should cover the template app block instantiation though.

likewise, 2021-12-13:
My Vivado is `Vivado v2020.2 (64-bit) SW Build: 3064766 on Wed Nov 18 09:12:47 MST 2020 IP Build: 3064653 on Wed Nov 18 14:17:31 MST 2020`

Adding a "keep" attribute to the specific register worked-around the problem.

I added some extra TCL puts to see what the script picks up. With this fix, the script finds the register and does its job.

```
diff --git a/fpga/lib/eth/lib/axis/rtl/axis_async_fifo.v b/fpga/lib/eth/lib/axis/rtl/axis_async_fifo.v
index 2d89f2da..d19131d3 100644
--- a/fpga/lib/eth/lib/axis/rtl/axis_async_fifo.v
+++ b/fpga/lib/eth/lib/axis/rtl/axis_async_fifo.v
@@ -161,7 +161,7 @@ localparam DEST_OFFSET = ID_OFFSET   + (ID_ENABLE   ? ID_WIDTH   : 0);
 localparam USER_OFFSET = DEST_OFFSET + (DEST_ENABLE ? DEST_WIDTH : 0);
 localparam WIDTH       = USER_OFFSET + (USER_ENABLE ? USER_WIDTH : 0);

-reg [ADDR_WIDTH:0] wr_ptr_reg = {ADDR_WIDTH+1{1'b0}};
+(* keep = "true" *) reg [ADDR_WIDTH:0] wr_ptr_reg = {ADDR_WIDTH+1{1'b0}};
```

```
Inserting timing constraints for axis_async_fifo instance core_inst/core_inst/core_pcie_inst/core_inst/iface[0].port[0].ptp.tx_ptp_ts_fifo_inst
DEBUG core_inst/core_inst/core_pcie_inst/core_inst/iface[0].port[0].ptp.tx_ptp_ts_fifo_inst/wr_ptr_reg_reg[0]/C
set read_clk_period 4.000
set write_clk_period 3.103

Inserting timing constraints for axis_async_fifo instance core_inst/core_inst/core_pcie_inst/core_inst/iface[0].port[0].rx_async_fifo_inst/fifo_inst
DEBUG core_inst/core_inst/core_pcie_inst/core_inst/iface[0].port[0].rx_async_fifo_inst/fifo_inst/wr_ptr_reg_reg[0]/C
set read_clk_period 4.000
set write_clk_period 3.103

Inserting timing constraints for axis_async_fifo instance core_inst/core_inst/core_pcie_inst/core_inst/iface[0].port[0].tx_async_fifo_inst/fifo_inst
DEBUG core_inst/core_inst/core_pcie_inst/core_inst/iface[0].port[0].tx_async_fifo_inst/fifo_inst/wr_ptr_reg_reg[0]/C
set read_clk_period 3.103
set write_clk_period 4.000
```

alexforencich, 2021-12-13:
Strange, I built the app template design for both the U50 and the fb2CG@KU15P, and there were no errors.  Are you building off of the code here, or the code on my fork?

likewise, 2021-12-14:
My issue report is bogus. My app module was not clean after all. I had a pass-through module but had the m_axis and s_axis mixed up in the connection. This allowed Vivado to optimize out most of the FIFO signals (I assume).

Sorry for the time Alex, I am closing the issue report.

closed 2021-12-14 by likewise

alexforencich, 2021-12-14:
Oops, no worries!  I also just realized that I'm not currently doing synthesis tests with the application block enabled.  So I should probably add one to make sure that I don't end up with an issue like this going forward.  My plan was to add a DMA benchmark application module, but I need to write at least a userspace component to that otherwise it doesn't do anything useful on the hardware.  Although, I suppose it's useful from a simulation and synthesis standpoint, even if I don't yet have a userspace tool to poke at it once it's running on an FPGA.