corundum/corundum #85: BUG: unable to handle page fault for address, during mqnic_prepare_rx_desc/mqnic_refill_rx_buffers

articice, 2021-12-30:
Build: Vivado 2019.2 (with compatibility change mentioned in https://github.com/corundum/corundum/issues/5#issuecomment-661286593)
Device: SQRL BCU-1525 (clone of Xilinx VCU-1525)
Test motherboard: **only 8GB of RAM**, Ubuntu 20.04, kernel 5.4.0-91-generic
SFP: not connected (I have 2 direct attach cables but nowhere to connect yet)

I get the following page fault when inserting the module:
```
kernel: [   76.213257] BUG: unable to handle page fault for address: 00000000fee00818
```
```
kernel: [   76.213285] Call Trace:
kernel: [   76.213297]  ? mqnic_prepare_rx_desc+0x9f/0x100 [mqnic]
kernel: [   76.213301]  mqnic_refill_rx_buffers+0x4c/0x70 [mqnic]
kernel: [   76.213303]  mqnic_activate_rx_ring+0xde/0xf0 [mqnic]
kernel: [   76.213307]  mqnic_start_port+0x1a0/0x318 [mqnic]
kernel: [   76.213310]  mqnic_open+0x2e/0x50 [mqnic]
kernel: [   76.213315]  __dev_open+0xd4/0x170
kernel: [   76.213317]  __dev_change_flags+0x191/0x1f0
kernel: [   76.213319]  dev_change_flags+0x26/0x60
kernel: [   76.213323]  do_setlink+0x84a/0x9a0
kernel: [   76.213327]  ? cpumask_next+0x1b/0x20
kernel: [   76.213331]  ? __snmp6_fill_stats64.isra.0+0xe7/0x110
kernel: [   76.213338]  ? __nla_validate_parse+0x42/0x140
kernel: [   76.213340]  __rtnl_newlink+0x580/0x970
kernel: [   76.213342]  ? __nla_put+0x20/0x30
kernel: [   76.213345]  ? __kmalloc_node_track_caller+0x20a/0x320
kernel: [   76.213348]  ? pskb_expand_head+0x75/0x2d0
kernel: [   76.213354]  ? __kmalloc_reserve.isra.0+0x31/0x90
kernel: [   76.213358]  ? security_sock_rcv_skb+0x2f/0x50
kernel: [   76.213359]  ? sk_filter_trim_cap+0x48/0x220
kernel: [   76.213360]  ? skb_queue_tail+0x48/0x50
kernel: [   76.213363]  ? __netlink_sendskb+0x42/0x50
kernel: [   76.213367]  ? security_capable+0x3d/0x60
kernel: [   76.213374]  ? _cond_resched+0x19/0x30
kernel: [   76.213380]  ? kmem_cache_alloc_trace+0x177/0x240
kernel: [   76.213381]  rtnl_newlink+0x49/0x70
kernel: [   76.213383]  rtnetlink_rcv_msg+0x2c0/0x380
kernel: [   76.213386]  ? vsnprintf+0x35d/0x4e0
kernel: [   76.213388]  ? rtnl_calcit.isra.0+0x100/0x100
kernel: [   76.213389]  netlink_rcv_skb+0x50/0x120
kernel: [   76.213391]  rtnetlink_rcv+0x15/0x20
kernel: [   76.213393]  netlink_unicast+0x187/0x220
kernel: [   76.213394]  netlink_sendmsg+0x222/0x3e0
kernel: [   76.213398]  sock_sendmsg+0x65/0x70
kernel: [   76.213400]  ____sys_sendmsg+0x212/0x280
kernel: [   76.213402]  ___sys_sendmsg+0x88/0xd0
kernel: [   76.213403]  ? rtnl_unlock+0xe/0x10
kernel: [   76.213406]  ? addrconf_sysctl_forward+0x111/0x280
kernel: [   76.213410]  ? get_max_files+0x20/0x20
kernel: [   76.213414]  ? __fget_light+0x57/0x70
kernel: [   76.213415]  __sys_sendmsg+0x5c/0xa0
kernel: [   76.213416]  __x64_sys_sendmsg+0x1f/0x30
kernel: [   76.213420]  do_syscall_64+0x57/0x190
kernel: [   76.213422]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
kernel: [   76.213423] RIP: 0033:0x7f394d6e512d
```

One weird thing I noticed is that RX queue count is 256, while TX queue count is 8192.
It looks like device gets added as eth0 and recognized by udev, but after this crash the machine's entire networking stack becomes broken (there's also motherboard's enp6s0, docker and VirtualBox).

There's no warnings about allocation, but maybe there's just to little RAM to allocate all buffers?

Logs attached:
[mqnic_crash.log](https://github.com/corundum/corundum/files/7791898/mqnic_crash.log)
[mqnic_udevadm_monitor.txt](https://github.com/corundum/corundum/files/7791899/mqnic_udevadm_monitor.txt)

alexforencich, 2021-12-30:
The queue count difference is expected, most of the US+ designs are set to 8192 TX and 256 RX queues by default.  Running out of memory is also not completely unexpected on a low-memory system.  Most probably some tweaks should be implemented to improve on this.  One of the things on my to-do list is to support changing the TX and RX queue counts at run time, which should help with this sort of thing.  Potentially this can be coupled with module parameters to set default queue count limits.  I think these should be relatively straightforward driver changes.  But screwing up the network stack is a more serious problem, I think it should be set up to somewhat gracefully handle running out of memory.  I'll take a closer look at the logs tomorrow and see if I can make any sense of what might be going on here.

articice, 2021-12-30:
@alexforencich I've implemented module params and reduced tx_queue_count & tx_cpl_queue_count and **it helped**
```
sudo insmod mqnic.ko tx_queue_count=256 tx_cpl_queue_count=256 
```
```
[  229.243258] mqnic 0000:0a:00.0: TX queue count set: 256, supported 8192
[  229.243261] mqnic 0000:0a:00.0: TX queue offset: 0x00100000
[  229.243263] mqnic 0000:0a:00.0: TX completion queue count set: 256, supported 8192
[  229.243265] mqnic 0000:0a:00.0: TX completion queue offset: 0x00200000
```

closed 2022-09-20 by articice