corundum/corundum #86: cocotb sim error with cadence simulator (with walk-around)

kaoruzhu1, 2021-12-31:
hi alex,

i run corundum cocotb-based testbench on cadence simulator (xrun).

my test env is:
conda3 with python 3.9
cocotb-1.60
corundum commit-tag: 76e18d2af8e8e97fdda4290114ee060cd15335c9

------------------------------------------------------------
the following error was caught:

 44203.20ns INFO     Memory write, address 0x00008000, length 8, BE 0xf/0xf
 44368.00ns ERROR    Exception raised by this forked coroutine
/opt/miniconda3/envs/cocotb-env/lib/python3.9/site-packages/cocotb/scheduler.py:541: FutureWarning: "Unwatched" tasks that throw exceptions will not cause the test to fail. See issue #2664 for more details.
  warnings.warn(
 44368.00ns INFO     run_test_nic failed
                     Traceback (most recent call last):
                       File "/opt/miniconda3/envs/cocotb-env/lib/python3.9/site-packages/cocotbext/pcie/xilinx/us/interface.py", line 526, in _run
                         frame.data.append((sample.tdata >> (i*32)) & 0xffffffff)
                       File "/opt/miniconda3/envs/cocotb-env/lib/python3.9/site-packages/cocotb/binary.py", line 592, in __rshift__
                         return int(self) >> int(other)
                       File "/opt/miniconda3/envs/cocotb-env/lib/python3.9/site-packages/cocotb/binary.py", line 509, in __int__
                         return self.integer
                       File "/opt/miniconda3/envs/cocotb-env/lib/python3.9/site-packages/cocotb/binary.py", line 336, in integer
                         return self._convert_from(self._str)
                       File "/opt/miniconda3/envs/cocotb-env/lib/python3.9/site-packages/cocotb/binary.py", line 231, in _convert_from_unsigned
                         return int(x.translate(_resolve_table), 2)
                       File "/opt/miniconda3/envs/cocotb-env/lib/python3.9/site-packages/cocotb/binary.py", line 77, in __missing__
                         return self.resolve_x(key)
                       File "/opt/miniconda3/envs/cocotb-env/lib/python3.9/site-packages/cocotb/binary.py", line 59, in resolve_error
                         raise ValueError("Unresolvable bit in binary string: '{}'".format(chr(key)))
                     ValueError: Unresolvable bit in binary string: 'z'

----------------------------------------------------------

as shown above, Memory write operations were always failed:

it seams that the root cause is:
cocotbext/pcie/xilinx/us/interface.py", line 526, in _run
                         frame.data.append((sample.tdata >> (i*32)) & 0xffffffff)

my walk-around is :
COCOTB_RESOLVE_X=ZEROS

alexforencich, 2021-12-31:
This sounds like it could be some sort of X propagation issue.  Can you attach a full waveform dump that gtkwave can open?  I'll see if I can track down what the root cause is.  Setting COCOTB_RESOLVE_X is more of a band-aid than a proper fix.

kaoruzhu1, 2022-01-01:
[dump.vcd.zip](https://github.com/corundum/corundum/files/7797359/dump.vcd.zip)
[sim.log](https://github.com/corundum/corundum/files/7797361/sim.log)

alexforencich, 2022-01-01:
Ah, I see what's going on.  The cadence tools are z extending instead of 0 extending when the wire is wider than the output port it is connected to.  And it looks like there are a few other cases of the same behavior.  I'll see if I can come up with a decent workaround in the HDL that isn't too messy.