corundum/corundum #88: Support of 1Gb Ethernet and cheaper FPGA families

ohault, 2022-01-18:
I would find also interesting for this project to also support 1Gb Ethernet and cheaper FPGA families / dev boards.
A stating point could be to add this idea in the porting guide.

Basseuph, 2022-01-19:
Hi ohault,

this is a partial duplicate and actually two questions in one.

There has been a discussion about supporting cheaper FPGAs, see #35. Maybe help us by defining what is cheap in your context? I would suggest to further discuss this topic in the original issue. What do you think @alexforencich?

As per the 1 GbE support, the verilog-ethernet library used to implement the Ethernet interfaces provides infrastructure for implementing 1 GbE interfaces as well.
Watch out for a pull request that introduces SGMII interface support for the SFP+ ports of the ZCU106. This will be based on a Xilinx GMII to SGMII IP Core.

Ulrich

alexforencich, 2022-01-19:
Supporting 1G is trivial - just connect a 1G MAC to the core logic, possibly with a width converter, and you're good to go.  There may be some bugs to be sorted out with respect to the datapath width...it's parametrized, but has only been tested at 64 bits and wider, so there could be issues below 32 or 16 bits in some modules.  Anyway, a bigger issue is that the control path does not scale with the datapath, so even if you shrink the data path to 8 bits, the design will still use a lot of logic resources.  On smaller FPGAs, transaction-layer PCIe support via hard IP, resource consumption, and timing closure are all going to be significant challenges.  Also, most likely smaller/slower devices will not be able to support more advanced features like SR-IOV, RDMA, etc. due to the limited available resources and timing closure issues.

alexforencich, 2022-01-20:
Anyway, closing as duplicate of #35

closed 2022-01-20 by alexforencich

ohault, 2022-01-20:
Thank you. Does it possible to document the current 1Gb Ethernet support and limitations on the Wiki/md pages of this project ?
I agree and I apology. For clarity, I should have mentioned the support of the small FPGAs only in the body of this ticket.

alexforencich, 2022-01-20:
No worries.  Yeah, mentioning this in the docs is probably a good idea, especially since a few people have asked about it.  TBH, gigabit Ethernet is not really an issue at all, you can take any existing 10G design and swap out the 10G MACs for 1G MACs and it should work fine.  Running on smaller devices is a different story, however.  Mainly this is not something I have explored much, so I don't have anything concrete to say other than enumerating what the requirements are on the PCIe side (which rules out a lot of low end parts) and noting that resource consumption and timing closure are likely to be major issues on small, slow parts.  The design is optimized for good performance on 10G, 25G, and 100G links, and for gigabit links on small FPGAs you would probably make rather different design decisions in many areas.

ohault, 2022-01-20:
It's very interesting. We can see it as two goals :
1) support entry levels Gigabit Ethernet technologies 1G, but also 2.5G and 5G (IEEE 802.3bz, NBASE-T and MGBASE-T)
2) optimized design for high-end FPGAs (somehow not so complex) and also for smaller FPGAs (more challenging)

I propose to rename the title of this issue as "Support of Gigabit Ethernet standards for 1G, 2.5G and 5G"

alexforencich, 2022-01-20:
again, IMO there isn't really much to your point 1, at least in terms of the Corundum core logic.  All you need to do is instantiate the correct MAC and PHY interface modules that correspond to the Ethernet interfaces on your target board, and connect that to the streaming interfaces presented by the core logic.  Point 2 is a different story.  I'm not sure what the best approach would be.  Possibly it would be better to have a rather separate design that's optimized for area, and if you go far enough down that path it might make more sense just to start a completely separate project.