corundum/corundum #89: Error changing RX/TX_QUEUE_INDEX_WIDTH in Simulation

andreasbraun90, 2022-01-26:
I tried to set Rx and TX_QUEUE_INDEX_WIDTH to 4 in ZCU106 simulation and got errors during elaboration.

` ../../lib/axis/rtl/axis_fifo.v:202: error: Indexed part widths must be constant and greater than zero.`
`../../lib/axis/rtl/axis_fifo.v:202:      : This part width expression violates the rule: KEEP_WIDTH`
`../../lib/axis/rtl/axis_fifo.v:202: error: Concatenation repeat may not be zero in this context.`
`../../lib/axis/rtl/axis_fifo.v:202: error: Unable to elaborate r-value: (KEEP_ENABLE)?(m_axis_pipe_reg[(PIPELINE_OUTPUT)-('sd1)][KEEP_OFFSET+:KEEP_WIDTH]):({KEEP_WIDTH{1'd1}})
`

I changed those lines in the Makefile and test_fpga_core.py:
`export PARAM_TX_QUEUE_INDEX_WIDTH ?= 4`
`export PARAM_RX_QUEUE_INDEX_WIDTH ?= 4`
and 
`parameters['TX_QUEUE_INDEX_WIDTH'] = 4`
`parameters['RX_QUEUE_INDEX_WIDTH'] = 4`

What am I missing?

alexforencich, 2022-01-26:
OK, this seems to be a parametrization issue in the FIFO module when the data width is less than 8 and the keep width is not overridden.  The default `KEEP_WITH` is `DATA_WIDTH/8`, which results in `KEEP_WIDTH=0`, which apparently causes some problems.  I will look in to the proper way to fix this, possibly by changing that default to `(DATA_WIDTH+7)/8`.

andreasbraun90, 2022-01-27:
I'm wondering why the TX_QUEUE_INDEX_WIDTH setting has any influence on the data width of the FIFO in simulation. A DATA_WIDTH of less than 8 bit seems not reasonable as well. Changing the TX_QUEUE_INDEX_WIDTH parameter in Verilog works fine. 

Regarding the AXIS FIFO module: The AXIS spec states that "the AXI4-Stream interface allows the data width to be any integer number of data bytes" so DATA_WIDTH/8 should be fine for TKEEP.

alexforencich, 2022-01-27:
I think the problem is actually RX_QUEUE_INDEX_WIDTH, as there are some FIFOs in the ingress pipeline that carry the RX queue index and hence are RX_QUEUE_INDEX_WIDTH bits wide.  TBH, the annoying thing about that error message is it doesn't include the full instance name, only the file name, so it took a bit of guess work to figure out which FIFO was causing the problem.  

Changing the parameter in the verilog doesn't matter as it is overridden by the simulator.

alexforencich, 2022-03-30:
I just updated the default KEEP_WIDTH parameter value in the AXI stream library.  Please try out this change on my fork here and let me know if you're still seeing the same issue: https://github.com/alexforencich/corundum

alexforencich, 2022-04-20:
I tested this with the current version of Corundum and it seems to work fine now.  So, I'll go ahead and close this as I think it's fixed.

closed 2022-04-20 by alexforencich