corundum/corundum #90: Simulation throws IndexError in TxRing-free_desc function for Tx packet count > 1024

andreasbraun90, 2022-01-26:
When running the ZCU106 simulation and increasing the number of packets to be transmitted to over 1024 results in an IndexError:
https://github.com/corundum/corundum/blob/0f82e0c5f3b9fa9a15e99d6e43b08b3b4dff7909/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/test_fpga_core.py#L433
```
1369148.00ns INFO     RX CC frame: UsPcieFrame(data=[0x00040010, 0x00000001, 0x00000019, 0x00000405], byte_en=[], parity=[0x0, 0x0, 0x0, 0x0], first_be=0x0, last_be=0x0, discontinue=False, seq_num=0)
1369164.09ns INFO     CPL data: (0, 1024, 1514, 1363601)
1369164.09ns INFO     Ring index: 1024
1369164.09ns ERROR    Exception raised by this forked coroutine
corundum.git/.tox/py3/lib/python3.6/site-packages/cocotb/scheduler.py:544: FutureWarning: "Unwatched" tasks that throw exceptions will not cause the test to fail. See issue #2664 for more details.
  FutureWarning
1369164.09ns INFO     run_test_nic failed
                      Traceback (most recent call last):
                        File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 1083, in _run
                          await self.handler(self.index)
                        File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 1219, in interrupt_handler
                          await eq.process()
                        File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 359, in process
                          await self.interface.process_tx_cq(cq)
                        File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 905, in process_tx_cq
                          ring.free_desc(ring_index)
                        File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 514, in free_desc
                          pkt = self.tx_info[index]
                      IndexError: list index out of range
1369164.09ns INFO     **************************************************************************************
                      ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                      **************************************************************************************
                      ** test_fpga_core.run_test_nic    FAIL     1369164.09        2711.00        505.04  **
                      **************************************************************************************
                      ** TESTS=1 PASS=0 FAIL=1 SKIP=0            1369164.09        2711.48        504.95  **
                      **************************************************************************************
```

alexforencich, 2022-01-26:
Whoops, I think this will fix that: https://github.com/alexforencich/corundum/commit/2132a8d98f844a2c9076e15ce30ebb51580d98f2

andreasbraun90, 2022-01-27:
Hi,

the IndexError is fixed, thanks! Unfortunately there seem to be 'None' packets now which trigger the assertion. (count variable set to 4096)

```
# 1411679.73ns INFO     Memory write, address 0x0b784500, length 59, BE 0xf/0x3
# 1411716.00ns INFO     RX CC frame: UsPcieFrame(data=[0x00040010, 0x00000001, 0x00000003, 0x00000443], byte_en=[], parity=[0x0, 0x0, 0x0, 0x0], first_be=0x0, last_be=0x0, discont
inue=False, seq_num=0)
# 1411732.09ns INFO     CPL data: (0, 1090, 1514, 1410340)
# 1411732.09ns INFO     Ring index: 66
# 1411732.09ns INFO     Test stopped by this forked coroutine
# 1411732.09ns INFO     run_test_nic failed
#                       Traceback (most recent call last):
#                         File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 1065, in _run
#                           await self.handler(self.index)
#                         File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 1201, in interrupt_handler
#                           await eq.process()
#                         File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 341, in process
#                           await self.interface.process_tx_cq(cq)
#                         File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 887, in process_tx_cq
#                           ring.free_desc(ring_index)
#                         File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 497, in free_desc
#                           self.driver.free_pkt(pkt)
#                         File "corundum.git/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/mqnic.py", line 1214, in free_pkt
#                           assert pkt is not None
#                       AssertionError
# 1411732.09ns INFO     **************************************************************************************
#                       ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
#                       **************************************************************************************
#                       ** test_fpga_core.run_test_nic    FAIL     1411732.09         924.02       1527.81  **
#                       **************************************************************************************
#                       ** TESTS=1 PASS=0 FAIL=1 SKIP=0            1411732.09         924.50       1527.02  **
#                       **************************************************************************************
```

alexforencich, 2022-01-28:
Aha, I do not have a check in the driver model to see if the ring is full when you call start_xmit, so it's overwriting previous packets and we're effectively getting a double free operation.  I'll look in to adding that.  Fortunately, start_xmit is async, so making that block until space is available should be doable.

alexforencich, 2022-01-28:
Try this: https://github.com/alexforencich/corundum/commit/155aa5caaee42042c9dc4b65b4b102507888a60d

andreasbraun90, 2022-01-28:
Great, it works!

closed 2022-01-28 by andreasbraun90