corundum/corundum #92: Support 10 Gbps on 100 Gbps design

mclayton1000, 2022-02-18:
Hi @alexforencich ,

thank you for your invaluable work.

I have a question about the 100 Gbps design.

Is it possible to support multiple speed on the 100 Gbps design?
For example, is it possible to support 10 Gbps speed?

Thank you, best regards.

alexforencich, 2022-02-18:
I'm assuming you're referring to run-time switching between 10G and 100G, without reloading the FPGA configuration?  Currently, this is not supported.  It requires a rather significant reconfiguration of the datapath between the core logic and the transceivers, and this requires additional logic resources as well as development time to get all of the kinks worked out.  It is in principle possible, but implementing has not been a high priority - it's an FPGA, you can just reflash the card to run at a different link rate.  We are currently working on run-time switching between 10G and 25G, but this is much simpler as it just requires updating a bunch of registers and resetting the transceivers, so the overhead is quite low.  We're adding a management soft core to handle this sort of stuff, possibly once that's operational we can consider expanding the run-time rate switching capabilities to include 100G.

alexforencich, 2022-02-18:
Also, I will note that Corundum is specifically designed so that you can run the same application logic and such on any of the design variants (different link rates, different boards) with minimal modification.  So you should be able to, for example, prototype something at 10G and later run it at 100G, possibly even on a different board.  Or you could build 10G and 100G variants of the same design, and then install the appropriate one on the card depending on the required link rate.

mclayton1000, 2022-02-18:
Hi @alexforencich ,

thank you for the reply.

Yes, I'm referring to run-time switching between 10G and 100G without reloading the FPGA.
I know that this is supported on Xilinx CMAC with the (non-free) AN/LT enabled but that should
only support 1,10,25,40,50,100 Gbps rates on copper and backplane.
As far as I understand the fiber optic standards such as 10GBASE-SR/LR are not supported
in that case.

Do you think that it would be possible to construct a runtime speeds-switchable design
which includes support for both copper and fiber standards?

Thank you, best regards.

alexforencich, 2022-02-18:
The hard CMACs on US and US+ do not support run-time switching, they can only run at 100 Gbps in the form of a 4x25G or a 10x10G link.  The AN logic is capable of negotiating a rate, but actually implementing that rate is outside of the scope of the Xilinx-provided IP core.  The CMAC does support runtime switching between 4x25G and 10x10G, but these are both 100G links.  On Versal devices, they have the MRMAC core which can switch between 10G/25G/40G/50G/100G.  

To do run-time switching, you would have to add a 10G/25G MAC alongside the CMAC, mux and demux the streaming interfaces and other host-side signals appropriately, switch the transceiver connections, and update the transceiver configuration registers and ports.  Doable, but it would take a fair amount of effort.

mclayton1000, 2022-02-18:
> 
Hi @alexforencich ,

Yes, I know about the MRMAC but actually only development boards exist with Versal FPGA.
No Alveo or other production boards.

So the AN logic just negotiate a rate and that rate should be supported with proper custom logic?
I didn't understand it. Thank you for the clarification.

Yes, without an already ready IP with run-time switch support the clean option is to have
multiple designs and deploy them as needed...

alexforencich, 2022-02-18:
There are PCIe form factor boards with Versal parts available, such as the Xilinx VCK5000, HTG HTG-VSL1, and Alpha Data ADM-PA100...

Anyway, the CMAC documentation is really unclear about exactly what the AN logic does.  But from what I understand, the AN logic is really quite separate from the actual CMAC and serves only to perform the actual exchange of information that takes place during autonegotiation, actually acting on that and reconfiguring things is "left as an exercise to the user".  Which IMO is rather lame.  Additionally, the fact that the CMAC cores don't support running at 40G is also rather annoying, as you're stuck with using soft MACs for 40G even if you're using a part with CMACs.  

If this is an important enough feature that you might want to put some time into the actual implementation, let me know, I have put some thought into this previously and could provide some guidance.

mclayton1000, 2022-02-18:
@alexforencich ,

unfortunately the Xilinx VCK5000 does not support Vivado flow.

mclayton1000, 2022-02-23:
Hi @alexforencich ,

yes, the VCK5000 board does not support Vivado flow while the FPGA mounted on it does.

About the AN/LT features I think you're right. I found the following page from Xilinx:
https://support.xilinx.com/s/article/69026?language=en_US
which says that LT algorithm is "user's responsibility".

The MRMAC is more flexible:
https://support.xilinx.com/s/article/1209580?language=en_US
and seems that it supports "seamless rate change".

closed 2022-02-23 by mclayton1000

alexforencich, 2022-02-23:
Right, the CMAC doesn't do it out of the box, additional logic would have to be implemented to support switching between 10G/25G and 100G.  Basically, a 10G/25G MAC would have to be added along side the CMAC along with some multiplexing logic, and then the transceivers and mux logic would have to be reconfigured to switch between rates.

So, the question is, is this a feature that you would like to see implemented in Corundum?  Currently, I do not have a pressing need for such a feature (swapping out the FPGA config is sufficient for the time being), so I cannot devote significant time to the implementation as there are other things that are more important.  But, if you're interested enough to help with the implementation, I can provide some guidance and support.  Also, there may be others in the Corundum community that might be able to help as well, but I cannot make any promises.

mclayton1000, 2022-04-27:
Hi @alexforencich ,

thank you for the reply and sorry for my long time response.

I think that a logic like the one you described could be very complex
and I can't imagine how can be implemented. Multiple design approach is still OK for me.
However, if you have some ideas on how to implement the multi-rate support,
please, share them.

Thank you, best regards.