corundum/corundum #95: how to make fpga.mcs

wyz1031, 2022-03-17:
Loading the design via indirect JTAG into nonvolatile memory with Vivado requires basically the same steps as loading it into volatile configuration memory, the main difference is that the configuration flash image must first be generated by running make fpga.mcs after using make to generate the bit file. Once this file is generated, connect with the hardware manager, add the configuration memory device (check the makefile for the part number), and program the flash. After the programming operation is complete, boot the FPGA from the configuration memory, either via Vivado (right click -> boot from configuration memory) or by performing a cold reboot (full shut down, then power on). When using the makefile, run make flash to generate the flash images, program the flash via indirect JTAG, and boot the FPGA from the configuration memory. Finally, reboot the host computer to re-enumerate the PCIe bus.

This is my result：

test@Yoga-14sARH-2021:~/Desktop/corundum-master/fpga/mqnic/ZCU106/fpga_pcie/fpga$ source /opt/Xilinx/Vivado/2020.2/settings64.sh
test@Yoga-14sARH-2021:~/Desktop/corundum-master/fpga/mqnic/ZCU106/fpga_pcie/fpga$ make program
echo "open_hw" > program.tcl
echo "connect_hw_server" >> program.tcl
echo "open_hw_target" >> program.tcl
echo "current_hw_device [lindex [get_hw_devices] 0]" >> program.tcl
echo "refresh_hw_device -update_hw_probes false [current_hw_device]" >> program.tcl
echo "set_property PROGRAM.FILE {fpga.bit} [current_hw_device]" >> program.tcl
echo "program_hw_devices [current_hw_device]" >> program.tcl
echo "exit" >> program.tcl
vivado -nojournal -nolog -mode batch -source program.tcl

****** Vivado v2020.2 (64-bit)
  **** SW Build 3064766 on Wed Nov 18 09:12:47 MST 2020
  **** IP Build 3064653 on Wed Nov 18 14:17:31 MST 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

source program.tcl
# open_hw
WARNING: 'open_hw' is deprecated, please use 'open_hw_manager' instead.
# connect_hw_server
INFO: [Labtools 27-2285] Connecting to hw_server url TCP:localhost:3121
INFO: [Labtools 27-2222] Launching hw_server...
INFO: [Labtools 27-2221] Launch Output:

****** Xilinx hw_server v2020.2
  **** Build date : Nov 18 2020 at 09:50:49
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [Labtools 27-3415] Connecting to cs_server url TCP:localhost:3042
INFO: [Labtools 27-3417] Launching cs_server...
INFO: [Labtools 27-2221] Launch Output:

******** Xilinx cs_server v2020.2
  ****** Build date   : Nov 04 2020-05:02:56
    **** Build number : 2020.2.1604437376
      ** Copyright 2017-2020 Xilinx, Inc. All Rights Reserved.

# open_hw_target
INFO: [Labtoolstcl 44-466] Opening hw_target localhost:3121/xilinx_tcf/Digilent/210249855365
# current_hw_device [lindex [get_hw_devices] 0]
# refresh_hw_device -update_hw_probes false [current_hw_device]
INFO: [Labtools 27-1435] Device xczu7 (JTAG device index = 0) is not programmed (DONE status = 0).
# set_property PROGRAM.FILE {fpga.bit} [current_hw_device]
# program_hw_devices [current_hw_device]
INFO: [Labtools 27-3164] End of startup status: HIGH
program_hw_devices: Time (s): cpu = 00:00:10 ; elapsed = 00:00:10 . Memory (MB): peak = 3181.180 ; gain = 1.000 ; free physical = 10751 ; free virtual = 14514
# exit
INFO: [Common 17-206] Exiting Vivado at Thu Mar 17 14:25:21 2022...
test@Yoga-14sARH-2021:~/Desktop/corundum-master/fpga/mqnic/ZCU106/fpga_pcie/fpga$ make fpga.mcs
make: *** No rule to make target 'fpga.mcs'.  Stop.
test@Yoga-14sARH-2021:~/Desktop/corundum-master/fpga/mqnic/ZCU106/fpga_pcie/fpga$

alexforencich, 2022-03-17:
Programming the flash is not currently supported for Zynq devices, unfortunately.  The zynq devices do not boot the same way as normal FPGAs, as the SoC has to boot first, which then loads the FPGA configuration.  I have not had the time to figure out the process for zynq devices when the ARM cores are not used.  I suspect Vitis may need to be involved, which would complicate things significantly.

wyz1031, 2022-03-17:
[image]
When I create boot image，there are still errors

alexforencich, 2022-03-17:
Unfortunately, I don't know what to tell you; I have never generated a flash image for a zynq device before, so I'm not sure what the process is.  You can try asking about it in the zynq channel on the Corundum slack workspace (the link to join is at the top of the readme).  There are several folks on there who are much more knowledgeable about zynq devices than I am.

wyz1031, 2022-03-17:
Thankyou all the same.

Basseuph, 2022-03-17:
As Alex pointed out, during start-up the PS loads the bitstream into the PL. To make this happen a couple of things need to be done.
Please note that the current ZCU106 project does not instantiate the PS, so some additional steps need to be done:

A - No modification to mqnic required
1. The Vitis projets needs an additional 'dummy' Vivado project to configure the PS and the SW to run on it
2. The dummy vivado project used as a basis to export the hardware description (old: HDF, new: XSA) needs to instantiate the PS
3. make sure the instantiated zynq has a valid configuration for the targeted board
4. make sure the instantiated zynq does not use ANY PL interfaces
5. build the Vivado project
6. use the results for a vitis project
7. swap the bitstream included into the results of the vitis project (SD Card boot files, etc.) from using the dummy project results  to including the original corundum bitstream
8. generate the vitis results and use them on the board

B - Including some modification of mqnic:
1. modify mqnic on ZCU106 to include a dummy instance of the PS
2. then use vitis and follow the steps 6-8 above

There is a development stream of mqnic targeting the integration of the NIC with the PS, which implements approach B, but already includes the PS-PL interfaces, so this might be a good starter for you: https://github.com/missinglinkelectronics/corundum/tree/mle/tb/zcu106-zynqmp-axi/fpga/mqnic/ZCU106/fpga_zynqmp
Please make sure to remove the PS-PL interfaces in case you want to follow approach A.

alexforencich, 2022-03-17:
Pardon my language, but why in the *&(##$*& does Xilinx have make this process so darn complicated......

Anyway, if adding a Zynq PS instance to the ZCU106 design would make the process simpler, then I see no reason not to do that.  Also, if support for the 250-SoC gets added at some point, I suspect we'll have a similar issue for that board as well.

wyz1031, 2022-03-18:
> As Alex pointed out, during start-up the PS loads the bitstream into the PL. To make this happen a couple of things need to be done. Please note that the current ZCU106 project does not instantiate the PS, so some additional steps need to be done:
> 
> A - No modification to mqnic required
> 
>     1. The Vitis projets needs an additional 'dummy' Vivado project to configure the PS and the SW to run on it
> 
>     2. The dummy vivado project used as a basis to export the hardware description (old: HDF, new: XSA) needs to instantiate the PS
> 
>     3. make sure the instantiated zynq has a valid configuration for the targeted board
> 
>     4. make sure the instantiated zynq does not use ANY PL interfaces
> 
>     5. build the Vivado project
> 
>     6. use the results for a vitis project
> 
>     7. swap the bitstream included into the results of the vitis project (SD Card boot files, etc.) from using the dummy project results  to including the original corundum bitstream
> 
>     8. generate the vitis results and use them on the board
> 
> 
> B - Including some modification of mqnic:
> 
>     1. modify mqnic on ZCU106 to include a dummy instance of the PS
> 
>     2. then use vitis and follow the steps 6-8 above
> 
> 
> There is a development stream of mqnic targeting the integration of the NIC with the PS, which implements approach B, but already includes the PS-PL interfaces, so this might be a good starter for you: https://github.com/missinglinkelectronics/corundum/tree/mle/tb/zcu106-zynqmp-axi/fpga/mqnic/ZCU106/fpga_zynqmp Please make sure to remove the PS-PL interfaces in case you want to follow approach A.
> 
> When you want to further get into a more detailed discussion, I would also suggest to join the slack channel as there are more people following the discussions with more detailed insight into the required steps.

https://github.com/missinglinkelectronics/corundum/tree/mle/tb/zcu106-zynqmp-axi/fpga/mqnic/ZCU106/fpga_zynqmp
[image]

When I run make this directory to bulid bitstream,there are errors.

alexforencich, 2022-03-18:
Sounds like make can't find some file.  If you run 'make fpga.xpr', do you get a different error?

wyz1031, 2022-03-18:
The error is “make: *** No rule to make target 'fpga.xpr'.  Stop.”

alexforencich, 2022-03-18:
I thought that might give you a more useful error message; apparently not.  You'll have to take a look at the file list in the make file and see if you can narrow down what's missing.  I would suggest a binary search - comment out half of the files, try building fpga.xpr, uncomment half of the commented-out ones, try building again, etc. until you can narrow down which file is missing.

Basseuph, 2022-03-18:
Strange, this does work for me.

Side-note: the current BD tcl requires the use of Vivado 21.1. If you need to use another Vivado version, please modify ip/zynq_bd.tcl in line 23. This might not work anymore, because Xilinx does not state anything w.r.t. BD tcl portability.

Basseuph, 2022-03-18:
I just re-checked this. It builds on my side without further changes.

Did you
1. clone the repository,
2. checkout the branch mle/tb/zcu106-zynqmp-axi
3. goto the fpga/mqnic/ZCU106/fpga_zynqmp directory
4. source the vivado env
5. exec make
?

wyz1031, 2022-03-22:
Modify ip/zynq_bd.tcl in line 23,there are errors.I am downloading and installing Vivado 2021.1.

Basseuph, 2022-03-22:
OK, please let us know if this then works for you or if you encounter issues again.

wyz1031, 2022-03-23:
[image]
source /tools/Xilinx/Vivado/2021.1/settings64.sh
I have installed the Vivado 2021.1,but the error has not changed.I think the problem is makefile,instead of the verison of Vivado.

wyz1031, 2022-03-23:
> I just re-checked this. It builds on my side without further changes.
> 
> Did you
> 
> 1. clone the repository,
> 2. checkout the branch mle/tb/zcu106-zynqmp-axi
> 3. goto the fpga/mqnic/ZCU106/fpga_zynqmp directory
> 4. source the vivado env
> 5. exec make
>    ?

The question is solved.The symlinks are broken.Now,I am executing make.Thankyou!!!

Basseuph, 2022-03-23:
Congratulations, great to see this.

closed 2022-04-20 by alexforencich