corundum/corundum #97: simulation will hang if PARAM_APP_ENABLE  is set 1 in Makefile

kaoruzhu1, 2022-03-25:
hi Alex,

The fpga design variant is ZCU106/2_fpga_pcie/tb/fpga_core.

I found the simulation will always hang if I changed 'export PARAM_APP_ENABLE ?= 1' in ZCU106/2_fpga_pcie/tb/fpga_core/Makefile.

the simulation will not proceed  @67672.00ns:
here is a peice of log info.

parity=[0x0, 0x0, 0x0, 0x0], first_be=0x0, last_be=0x0, discontinue=False, seq_num=0)
 67484.09ns INFO     CPL data: (0, 22, 60, 0, 0, 3364)
 67484.09ns INFO     Ring index: 22
 67484.09ns INFO     Packet: Packet(data=b'\x14\x15\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNO', timestamp_s=0, timestamp_ns=0, rx_checksum=0x0d24)
 67484.09ns INFO     CPL data: (0, 23, 60, 0, 0, 9275)
 67484.09ns INFO     Ring index: 23
 67484.09ns INFO     Packet: Packet(data=b'\x15\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOP', timestamp_s=0, timestamp_ns=0, rx_checksum=0x243b)
 67484.09ns INFO     CPL data: (0, 24, 60, 0, 0, 15186)
 67484.09ns INFO     Ring index: 24
 67484.09ns INFO     Packet: Packet(data=b'\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQ', timestamp_s=0, timestamp_ns=0, rx_checksum=0x3b52)
 67484.09ns INFO     CPL data: (0, 25, 60, 0, 0, 21097)
 67484.09ns INFO     Ring index: 25
 67484.09ns INFO     Packet: Packet(data=b'\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQR', timestamp_s=0, timestamp_ns=0, rx_checksum=0x5269)
 67484.09ns INFO     Packet: Packet(data=b'\x14\x15\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNO', timestamp_s=0, timestamp_ns=0, rx_checksum=0x0d24)
 67484.09ns INFO     Packet: Packet(data=b'\x15\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOP', timestamp_s=0, timestamp_ns=0, rx_checksum=0x243b)
 67484.09ns INFO     Packet: Packet(data=b'\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQ', timestamp_s=0, timestamp_ns=0, rx_checksum=0x3b52)
 67484.09ns INFO     Packet: Packet(data=b'\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQR', timestamp_s=0, timestamp_ns=0, rx_checksum=0x5269)
 67504.00ns INFO     TX CQ frame: UsPcieFrame(data=[0x00380018, 0x80000000, 0x00000801, 0x00c00000, 0x0000001a], byte_en=[0x0, 0x0, 0x0, 0x0, 0xf], parity=[0xb, 0x7, 0xc, 0xf, 0xe], first_be=0xf, last_be=0x0, discontinue=False, seq_num=0)
 67508.00ns INFO     TX CQ frame: UsPcieFrame(data=[0x00300018, 0x80000000, 0x00000001, 0x00c00013], byte_en=[0x0, 0x0, 0x0, 0x0], parity=[0xf, 0x7, 0xe, 0xe], first_be=0xf, last_be=0x0, discontinue=False, seq_num=0)
 67616.00ns INFO     RX CC frame: UsPcieFrame(data=[0x00040018, 0x00000001, 0x00000013, 0x0000001a], byte_en=[], parity=[0x0, 0x0, 0x0, 0x0], first_be=0x0, last_be=0x0, discontinue=False, seq_num=0)
 67632.09ns INFO     Interrupt handler end (IRQ 0)
 67652.00ns INFO     TX CQ frame: UsPcieFrame(data=[0x0038000c, 0x80000000, 0x00000801, 0x00c00000, 0x80000000], byte_en=[0x0, 0x0, 0x0, 0x0, 0xf], parity=[0xb, 0x7, 0xc, 0xf, 0x7], first_be=0xf, last_be=0x0, discontinue=False, seq_num=0)
 67660.00ns INFO     TX CQ frame: UsPcieFrame(data=[0x00080018, 0x80000000, 0x00000801, 0x00c00000, 0x00000007], byte_en=[0x0, 0x0, 0x0, 0x0, 0xf], parity=[0xb, 0x7, 0xc, 0xf, 0xe], first_be=0xf, last_be=0x0, discontinue=False, seq_num=0)
 67664.00ns INFO     TX CQ frame: UsPcieFrame(data=[0x0008000c, 0x80000000, 0x00000801, 0x00c00000, 0x80000000], byte_en=[0x0, 0x0, 0x0, 0x0, 0xf], parity=[0xb, 0x7, 0xc, 0xf, 0x7], first_be=0xf, last_be=0x0, discontinue=False, seq_num=0)
 67672.00ns INFO     TX CQ frame: UsPcieFrame(data=[0x0088000c, 0x80000000, 0x00000801, 0x00c00000, 0x80000000], byte_en=[0x0, 0x0, 0x0, 0x0, 0xf], parity=[0xf, 0x7, 0xc, 0xf, 0x7], first_be=0xf, last_be=0x0, discontinue=False, seq_num=0)

alexforencich, 2022-03-29:
If you set `PARAM_APP_ENABLE` to 1 and change nothing else, the simulation *should* refuse to run as `mqnic_app_block` will not be found (as in, it should not even get far enough to hang).  Did you update the file list in the makefile to include the application section?

alexforencich, 2022-04-01:
Odd, I'll take a look.  This is not related to https://github.com/corundum/corundum/issues/89, correct?

alexforencich, 2022-04-20:
I just tested the current version with this parameter change and it ran to completion.  So, I'm not sure what issue you could be running in to here.  I suspect that maybe there is a simulator-related issue, as you were running in to some issues with VCS previously (#86).  The template application is tested as part of the CI, although not specifically with the ZCU106, so any serious integration issues with the application section should be flagged.