corundum/corundum #99: Support for link state detection

joft-mle, 2022-04-22:
Hi Alex,

as mentioned in the last Corundum Developer Meeting, we would like to have some form of link state detection. So, we would like to start a discussion by providing a very first work-in-progress (WIP) implementation/proposal/sketch, see our branch [mle/eth-link-state](https://github.com/missinglinkelectronics/corundum/tree/mle/eth-link-state).

On this branch we

- first add a per-port CSR module and
- then introduce a "link state" register there, which for now just exports at its bit 0 the RX Block Lock signal coming from the associated PHY as a _first possible_ indication of "link up" or "link down". This register could, in the future, get more fields to inform software about other aspects of a link's state.

The last commit implements the top-level logic/wiring for the `ZCU106/fpga_pcie` design, which we used for testing. (Note that this design currently fails timing, due to missing constraints from the PHY's `rx_block_lock` signal to the first `sync_signal/sync_reg`.) All commits marked "WIP" need more work of course  - to not break all the design other than `ZCU106/fpga_pcie`.

If you are ok with the basic structure of our proposal, we would then go on and work on the missing pieces. For non-ZCU106 designs, we would tie the new mqnic_core top-level signal to 1 (as if the RX Block Lock was always "up"), as can be seen in the 4th commit. Furthermore, we would work on a driver patch, to report the RX Block Lock bit as the associated netdev's carrier state. This would also involve a background task, which periodically polls the bit, for now, since there is such thing as a "link state" event/interrupt, which could be added in the future.

joft-mle, 2022-05-20:
This issue can be closed. [Commits](https://github.com/corundum/corundum/commit/36e202352b8a98fdd0b324a570bcf4bbc7b20208) pushed to master recently provide desired functionality. Pull request #101 is a follow-up, adding basic support for link status monitoring in the mqnic driver.

alexforencich, 2022-05-23:
Thanks for the pull request!

closed 2022-05-23 by alexforencich