corundum/corundum #103: Apparently bash specific syntax in testbench Makefiles?

joft-mle, 2022-05-31:
Hi @alexforencich, I recently came across a minor issue within the testbench Makefiles. The shell syntax `$((16#00000000))`, e.g. [here](https://github.com/corundum/corundum/blob/dd2853bf4002b318bd1d05203dab846380f17be7/fpga/mqnic/ZCU106/fpga_pcie/tb/fpga_core/Makefile#L199), to interpret a hexadecimal number seems to be `bash` specific. And at least Debian and Ubuntu have used in the past and still use, by default, the simpler (POSIX) shell implementation `dash`, for `/bin/sh`. And at least on older distros, for some reason, `make` even seems to always use `dash` (independent of where `/bin/sh` points to; unclear why).

I'm usually using `$((0x00000000 + 0))` for hex. interpretation in shell scripts, since `dash` does support that syntax.

alexforencich, 2022-05-31:
Huh.  I switched over to the `16#deadbeef` syntax due to https://github.com/alexforencich/verilog-pcie/issues/3 as I thought it was more compatible.  Apparently it seems that using hex in shell scripts/makefiles might be a bad idea in general as support for it seems to be terrible.

alexforencich, 2022-05-31:
And after re-reading that issue, apparently it was not a syntax issue of `0xdeadbeef` vs `16#deadbeef`, but instead an issue with quotes.  So maybe the `0xdeadbeef` syntax is fine after all.  If you switch that over to `0x00000000` does it work?  And is the extra `+ 0` necessary for some reason?

joft-mle, 2022-06-02:
Oh, yes, you are right, LOL, I didn't realize that before. The `<expr>` in `$((<expr>))` can indeed just be an `<operand>` (in a supported number format) _without_ any `<operator>` _(Maybe in a distant past, that was not the case? ;-)  )_. And yes, `dash` just does not support `<base>#<number>` as an `<operand>` format, but does support `0x<number>` for hex. So, yes, there is no need for ` + 0` ...

alexforencich, 2022-08-12:
This is fixed in the mainline repo now.

closed 2022-08-12 by alexforencich