corundum/corundum #105: Clock for PTP modules

nightseas, 2022-06-20:
Hi,

Some questions about the clock for PTP modules, taking example of ADM-PCIe-9V3.
- Any consideration of using 300MHz memory interface clock as ptp_sample_clk instead of the same GTY ref clock for ptp_clk?
- Let's say if the GTY clock comes from Synchronized Ethernet which has <1ppb accuracy and the 300MHz ptp_sample_clock comes from a local oscillator that is 30ppm, will that bring in any extra time error on PTP sync?
- For 10G/25G design, is gtwiz_userclk_rx_usrclk2_out/phy_rx_clk the correct one to extract as GTY RX recovery clock?

It seems ADM 9V3 has the HW capability of supporting SyncE. I'm trying to do some modifications to make it happen.

Thanks & BR.

alexforencich, 2022-06-20:
- ptp_sample_clk should already be sourced from the 300 MHz system clock, via the MMCM in fpga.v.
- The point of the sample clock is effectively to compensate for synchronizer delays.  In the PTP CDC module, there is digital PLL that attempts to lock a divided source clock (fixed) to a divided destination clock (variable, fractional).  ptp_sample_clk is used to clock the phase and frequency detector logic for this PLL.  If this clock is not used, then the PFD is clocked by the dest clock domain.  The idea is that if you have a clock that is asynchronous to the source and dest clocks, then you can get a better measurement of the relative phase of the two clocks.  So I don't think the accuracy or stability of this clock is particularly important, the important part is that it is asynchronous to both the source and dest clock domains.  
- Yes, that should be the correct clock

And yeah, getting syncE working on any card would be very cool.  I'm thinking that we'll have to get the risc-v soft core in the loop in terms of tuning the transceiver reference clock source so the driver doesn't need any board-specific code.  But this can be retrofitted later on once the soft core is integrated.  Not sure what the procedure is for adjusting the PLL on the ADM-PCIE-9V3, but I think you might have to go through the BMC microcontroller.  We'll probably also have to tweak the driver a bit, as when you're doing SyncE, the PTP clock period registers should be set to their nominal value, and the frequency should be tuned via the PLL.  

It also seems like a number of FPGA boards could potentially support SyncE, including the U50, U200, U250, U280, VCU1525, VCU108, VCU118, ZCU106, possibly the fb2CG@KU15P (if Silicom makes some modifications to the BMC FW), and possibly a few others.  Basically, anything that has a way to make fractional adjustments to the GT ref clk.  Although some configuration changes may be required as many of these designs don't currently use the required oscillator.  

I have also been doing some head-scratching over the DDMTD technique used in white rabbit, specifically how to generate small frequency offsets with the MMCMs/PLLs.  I have put together a small python script for computing PLL settings for the US/US+ PLLs and MMCMs.  Not sure if 1 MMCM would be sufficient, or if cascading two would be necessary to get a sufficiently small offset.

leehongming, 2022-06-20:
hi alex, I'm happy to see that you are talking something related to my previous work about DDMTD in White Rabbit.
You may be interested in this file, which talking about how to generate DMTD helper clock with inner cascaded PLL in FPGA.
https://ohwr.org/project/white-rabbit/uploads/a5b9eaa8397883e30953b4eb1cd97831/DMTD_clock_generation_using_FPGA_internal_PLLs.pdf
As far as I know, peter(https://ohwr.org/peterj)  has implemented this kind of DDMTD in artix-7/kintex-7. You may find some code with https://ohwr.org/project/wr-cores

alexforencich, 2022-06-21:
Oh nice, I did not realize that you were involved with White Rabbit at all.  So it seems like a ratio of 4096/4095 (offset of about 250 ppm) is sufficient?  It looks like it should be possible to get an offset of around 1000 ppm with a single US+ MMCM, which is only out by about a factor of 4.  But two cascaded MMCMs or PLLs would presumably be able to produce a much smaller offset, and could potentially perform better by not using any fractional dividers.  Anyway, it would be very cool to get to a point where Corundum could work as a white rabbit node.

nightseas, 2022-06-21:
I did some test on ADM-PCIE-9V3.
So the idea is to use the Si5328 to generate free-run MGT reference clocks and to extract SyncE recovery clock from phy_rx_clk and send to Si5328 as reference. Once locked, the GTY transceiver and PTP clock shall work with high precision SyncE clock.
[image]

Not sure if this is the correct way but what I did was dividing the qsfp_0_rx_clk_0_int to 25MHz with MMCM, and send out to Si5328 via OBUFDS. The I2C of Si5328 is directly connected to FPGA, so that the configuration can be downloaded from host software.

 After configuration of PLL and the reset on FPGA, free-run clock works fine on a 10GE design. However switching to recovery clock seemed to cause glitch on clock and link down eventually, which is not good. Could be a problem on the reference switching function of Si5328.

leehongming, 2022-06-25:
Yes, two cascaded MMCMs or PLLs could get 4095/4096, even 16385/16384 with fractional dividers.
In current WR, this clock generation circuit could be used to generate the helper clock, which samples both the local ref clock and recovery clock from PHY to get their phase difference. Then the DMTD works as PD to build main PLL with the help of external vco, making the local ref clock track with recovery clock, so that sync-E could be achieved. 
DMTD has some limitation. It is only possible to measure the phase difference between two syntonization clock, which means that it is not possible to split DDMTD with main PLL. I have no idea how to achieve the low jitter main PLL in corundum without the help of external VCO. Maybe we could control the phase shift of PLL in FPGA rapidly.

alexforencich, 2022-06-25:
My plan was to adjust the divider on the reference clock source. For many of the silicon labs parts, it has a resolution in the ppb. Would that not work?

leehongming, 2022-06-25:
I think the hard part to use DDMTD is how to syntonized ref clock(PLL output) and recovery clock at the beginning as DMTD has limited track region.
Maybe FPGA TDC is more preferable. It could also measure the trace difference with two clocks in ps precision, without the limitation like DMTD.

alexforencich, 2022-06-25:
Ah yeah, I was wondering about the tracking range during the initial locking process. What about using both DDMTD and a more normal phase detector, switching over to DDMTD once the PLL is effectively locked?

leehongming, 2022-06-25:
Sounds to be reasonable.
If control loop is achieved in software(Real-Time CPU) , then switch-over could be implemented easily. WR uses embedded software in LM32 CPU (62.5MHz) , whose performance is enough.

nightseas, 2023-06-17:
> Could be, but my thought was to close the loop on the FPGA instead of in the PLL chip. These chips usually have rather high resolution fractional dividers, so the idea would be to put the phase detector on the FPGA, and then the PLL dividers can be periodically adjusted to bring the ref clock into alignment with the RX clock. This would minimize disruption to the reference clock as it would only be adjusted by a few ppm one way or the other, with no source switching or any other major disruptions.

Hey Alex,

Just some updates of SyncE/PTP evaluation.
I have implemented a similar structure like this one but with Renesas PLL and high-precision oscillator, by designing a customized FPGA board.
[image]

The result is quite promising. PTP noise generation of a boundary clock is reduced to as low as 2ns time error and sub nano sec cTE with or without SyncE clock.
[image]

You can find the porting and implementations in my fork. I guess this issue is closed.

Next steps that I'm planning are GNSS, EXTTS, NIC cascading, and PTM. The white rabbit is also very interesting to dig into.

BR.

alexforencich, 2023-06-17:
How are you planning on doing PTM?  Soft PCIe IP core of some sort?

nightseas, 2023-06-18:
I'm now working on GNSS and EXTTS. PTM is not yet well studied.

My initial thought, which could be totally wrong, was to pass through PTM messages (TLPs) to logic and implement cross-timestamping with PHC.
Do you think if it's possible with Xilinx PCIe HIP?

And an interesting thing is that [OCP claims WIP of PTM on Corundum](https://www.opencompute.org/w/index.php?title=PTM_Readiness). Not sure if you are aware of that.

alexforencich, 2023-06-18:
I suspect PTM is not possible with the Xilinx US/US+ HIP. There are two things that are needed: handling of the PTM messages, and PTP-like timestamping of said messages. The timestamping is not possible with the HIP, but it might be possible to implement enough of the RX side of the PCIe PHY in soft logic to capture the timestamps. Basically, pick off all of the GT data lines for both RX and TX so the PTM messages can be detected and timestamped. But, I'm also not sure if the Xilinx HIP can pass PTM messages through, or if they'll get dropped as unknown messages.

And yes, I am aware of the OCP thing, I just spoke with Ahmad last week. They want to get PTM and white rabbit working, but it's unclear what the path is going to be at this point.