corundum/corundum #106: simulation will hang if PARAM_RX_HASH_ENABLE and PARAM_RX_CHECKSUM_ENABLE both set 0 in Makefile

zhouandy1985, 2022-07-01:
I am evaluating corundum now, when I Set both PARAM_RX_HASH_ENABLE and PARAM_RX_CHECKSUM_ENABLE 0 , the simulation (test_mqnic_core_pcie_s10.py), will hang, the problem comes from  just bypass in mqnic_ingress.v module at this mode, 
the m_axis_tready will never goes high and rx_engine module will never successful receive a "s_axis_rx_req_*", they come into a deadlock situation, the receive path will hang. 
After add a axis_fifo ，i fixed the problem, but  this should be probelm of mqnic_interface_rx module , and i believe there is a more decent solution.

alexforencich, 2022-07-09:
I reworked the RX request generation code to fix this problem: https://github.com/alexforencich/corundum/commit/6b0df7f33fb42fe13a9f9c0e446935a1fa6a5e8b

This change will end up in this repo relatively soon once I get a couple of other things squared away.

zhouandy1985, 2022-09-15:
well done

closed 2022-09-15 by zhouandy1985