corundum/corundum #109: How to see what "parameter" and "localparamter" values are in the cocotb simulation?

zaidtahirbutt, 2022-07-18:
Is there a way to display the value of a particular parameter or localparameter when the simulation of an example fpga in Corundum is done using "make WAVES=1"?  (many parameters are in the form of equations, so it would be easier if I can just display their values in the simulation).

alexforencich, 2022-07-21:
Depends on the simulator, and the parameter.  Some simulators will include parameters in the waveform dumps, so you can just look at the values.  If this isn't the case, then there are a couple of options.  If the parameter drives a signal width, then you can simply look at the signal width in the waveform dump.  If this isn't the case, then a simple thing you can do is add an initial block with `$display` statements to print out the parameter values you're interested in.

zaidtahirbutt, 2022-07-21:
Thanks. Using $display would be an easy and quick solution.

closed 2022-07-21 by zaidtahirbutt